espressif/esp-idf #5092: Support for the LAN8742a.

vanBassum, 2020-04-10:
Hello,

We are currently using IDF version 3.3 and the LAN8742a chip for ethernet. We created our own driver witch works for us. Now with the upgrade to 4.0 the library has changed significantly and I can't find any information how to write a driver using the new framework.

I tried to implement my own 'esp_eth_phy_new_lan8742a' function based upon the LAN8720, but I couldn't get it to work.

Are there any plans to support this chip?
Or is there some documentation to help my write the driver myself?

suda-morris, 2020-04-10:
Sorry about that, yes, Ethernet driver has upgraded to support more kinds of Ethenret (even 3rd party modules). Thanks for letting me know your confusion, we will add a chapter to guide users how to add his own PHY driver.
In your case, I think it's a good option to add new phy driver based on lan8720.
1. copy  `esp_eth_phy_lan87820.c` and rename it to `esp_eth_phy_lan8742.c`, the new file doesn't have to be inside esp-idf, it can resides in your application folder, so that you can keep updated to upstream.
2. then, you should read lan8742 datesheet, to find the difference between lan8720 (especially to find the register that tells you the link status (speed, duplex) and auto-negotiation result

vanBassum, 2020-04-14:
The only thing I changed was the version check.
`PHY_CHECK(id1.oui_msb == 0x7 && id2.oui_lsb == 0x30 && id2.vendor_model == 0x13, "wrong chip ID", err);`

This seems to be working, I get an IP address through DHCP and I can send and receive TCP packages. I don't know for sure, but I think further testing is required to ensure that everything is compatible.

closed 2020-04-14 by vanBassum

Sejny, 2021-04-13:
LAN8720 driver has issue: (cause problems only when you use PHY CLK from ESP32)
https://github.com/espressif/esp-idf/issues/6821

Symptom is random incorrect initialisation and ethernet not working.