espressif/esp-idf #5094: Problem of "implicit declaration of function" in ESP-IDF - v4.1-dev-1086-g93a8603c5

ROYMURLI, 2020-04-10:
Hi,
I was using IDF version - **v3.3-beta1-136-g97eecfa1b** for my project and developed my whole project in it.
But when I downloaded IDF version- **v4.1-dev-1086-g93a8603c5** , I am getting compilation error:-

 implicit declaration of function 'esp_wifi_set_auto_connect' [-Werror=implicit-function-declaration]
   esp_wifi_set_auto_connect(false);

 implicit declaration of function 'system_init' [-Werror=implicit-function-declaration]
  system_init();

implicit declaration of function 'system_efuse_read_mac' [-Werror=implicit-function-declaration]
      system_efuse_read_mac(base_mac_addr);

Kindly tell me is there any change in "component" of IDF for these functionality and how can I fix that.

liuzfesp, 2020-04-13:
Hi All these APIs are deprecated and are removed in v4.1 and later IDF releases. 

For `esp_wifi_set_auto_connect` and `system_init`, you can just delete them because the are empty function in v3.3.

The `system_efuse_read_mac` is replaced by `esp_efuse_mac_get_default()` in IDF v4.1 now, so you need to change it accordingly.

ROYMURLI, 2020-04-13:
Thanks @liuzfesp . Its working.

closed 2020-04-13 by ROYMURLI