espressif/esp-idf #5095: bzero is deprecated function

petrkr, 2020-04-10:
Consider replace `bzero` by `memset` as long bzero is deprecated and could lead to compile problems with newer gcc

I already tested it here and it works.
https://github.com/espressif/esp-idf/blob/master/components/heap/heap_caps.c#L409

http://man7.org/linux/man-pages/man3/bzero.3.html
https://www.quora.com/In-C-what-is-the-difference-between-bzero-and-memset

igrr, 2020-04-20:
Hi @petrkr, newlib (which is the C library we use) currently doesn't mark `bzero` as deprecated. Which compile issues did you encounter with bzero and newer GCC?

petrkr, 2020-04-20:
It's only for make you notice about that It can be removed in future versions. That is why I mark it only as "feature request" or something like not important in today. Just keep in mind it could be problem later.

closed 2023-05-25 by zikalino