espressif/esp-idf #5098: maximum data of notify event in gatt client

sowmimalu, 2020-04-11:
hai ,
   i m using gatt client example .my application is to connected a central with multiple periperal .here my central is esp32 and pheriperal is NRF52840. i can able to connect esp gatt client with periperal.able to send data through gatt write event .my pheriperal needs to send back a 30 bytes of data through notification to client (esp32). i changed in peripheral side for send a 30 bytes of data and check with app.then tried to send a data to client .but esp gatt client is not getting a data in notification more than 20 bytes..how to solve this ??

chegewara, 2020-04-11:
MTU - 3 bytes. Default MTU is 23 bytes.

sowmimalu, 2020-04-11:
is it possible to get more than 20 bytes through notification.
	case ESP_GATTC_NOTIFY_EVT:

		if (p_data->notify.is_notify) {

			ESP_LOGI(GATTC_TAG, "ESP_GATTC_NOTIFY_EVT, Receive notify value:");
			printf("notifyy\n");
			esp_log_buffer_hex(GATTC_TAG, p_data->notify.value,
					p_data->notify.value_len);
}
i have set the local mtu size of 500.
esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
	if (local_mtu_ret) {
		ESP_LOGE(GATTC_TAG, "set local  MTU failed, error code = %x",
				local_mtu_ret);
	}
	else
	{
		ESP_LOGE(GATTC_TAG,"SUCCES IN REG MTU");
	}
	return 0;
}
then when gatt is opening it shows always 23 of mtu size.

chegewara, 2020-04-11:
It is possible, but both, central and peripheral, has to accept MTU bigger than 23.
Also you have to request MTU after establishing connection with `esp_ble_gattc_send_mtu_req`.

sowmimalu, 2020-04-11:
here is my log i got from client side.

I (11290) GATTC: d0 3f 9d 21 28 4b
I (11290) GATTC: Searched Adv Data Len 22, Scan Response Len 4
I (11290) GATTC: Searched Device Name Len 13
I (11290) GATTC: Nordic_Blinky

comparing string
I (11300) GATTC: Searched device Nordic_Blinky
I (11310) GATTC: Stop scan successfully
gattc open for 1st device
I (11490) GATTC: ESP_GATTC_OPEN_EVT conn_id 0, if 2, status 0, mtu 23
I (11490) GATTC: REMOTE BDA:
I (11490) GATTC: d0 3f 9d 21 28 4b
E (11500) GATTC: Received MAC: d0:3f:9d:21:28:4b
mtu ok
esp_ble_gattc_search_service
I (12340) GATTC_COASTER: Status 0, MTU 200, conn_id 0
I (12340) GATTC_COASTER: SEARCH RES: conn_id = 0 is primary service 1
I (12350) GATTC: start handle 11 end handle 65535 current handle value 11
uuid searchI (12350) GATTC: UUID16: ee
I (12360) GATTC: Get service information from remote device
I (12370) GATTC: ESP_GATTC_SEARCH_CMPL_EVT
ESP_GATTC_SEARCH_CMPL_EVT
count 1
I (12420) GATTC: write descr success

after connecting my client with periperal. i am sending a sending a mtu req
	esp_err_t mtu_ret = esp_ble_gattc_send_mtu_req(gattc_if,  //me
					p_data->open.conn_id);                            //me

		if (mtu_ret) {
			ESP_LOGE(GATTC_TAG, "config MTU error, error code = %x", mtu_ret);
		}
		else
		{
			printf("mtu ok\n");
		}
i am sending a mtu request on client side.its getting request ok..which is in log.
seted local mtu in esp client side as 200 and periperal as NRF_SDH_BLE_GATT_MAX_MTU_SIZE as 247 changed .but in client always showing mtu of 23 after the  gatt connection with periperal. is there any further changes to do ??

chegewara, 2020-04-11:
This looks good:
`I (12340) GATTC_COASTER: Status 0, MTU 200, conn_id 0`
Now you can try notifications.

sowmimalu, 2020-04-11:
Now its working fine. changed default size of mtu on peripheral side. i can able to get more than 20 bytes through notification .thanks for your support.

Alvin1Zhang, 2020-04-12:
@chegewara Thanks for the notes. @sowmimalu  Thanks for reporting, and glad that your issue is resolved. Feel free to reopen if the issue still happens. Thanks.

closed 2020-04-12 by Alvin1Zhang