espressif/esp-idf #5099: 3.1SDK boot无法启动3.3.1SDK编译的固件

shenyong-iot01, 2020-04-11:
您好，最近SDK从3.2.2升级到3.3.1SDK发现一批老设备升级固件（3.1SDK）后无法启动，请帮忙确认！
同一硬件：
3.1SDK  bootloader可以正常启动3.3.2SDK编译的固件
3.2.2SDK bootloader可以正常启动3.3.1SDK编译的固件
3.1SDK无法正常启动3.3.1SDK编译的固件

【2020-04-10 17:48:29:386】 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5752
load:0x40078000,len:9020
load:0x40080000,len:6064
entry 0x40080330
[0;33mW (29) boot: PRO CPU has been reset by WDT.[0m
[0;33mW (29) boot: WDT reset info: PRO CPU PC=0x4008334f[0m
[0;33mW (29) boot: WDT reset info: APP CPU PC=0x400621b6[0m
[0;32mI (35) boot: ESP-IDF v3.1-dirty 2nd stage bootloader[0m
[0;32mI (41) boot: compile time 16:32:12[0m
[0;32mI (45) boot: Enabling RNG early entropy source...[0m
[0;32mI (50) boot: SPI Speed      : 40MHz[0m
[0;32mI (54) boot: SPI Mode       : DIO[0m
[0;32mI (58) boot: SPI Flash Size : 4MB[0m
[0;32mI (62) boot: Partition Table:[0m
[0;32mI (66) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (73) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (81) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (88) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (96) boot:  3 ota_0            OTA app          00 10 00010000 00180000[0m
[0;32mI (103) boot:  4 ota_1            OTA app          00 11 00190000 00180000[0m
[0;32mI (111) boot: End of partition table[0m
[0;32mI (115) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2ca14 (182804) map[0m
[0;32mI (161) esp_image: segment 1: paddr=0x0003ca3c vaddr=0x3ffb0000 size=0x03488 ( 13448) load[0m
[0;32mI (165) esp_image: segment 2: paddr=0x0003fecc vaddr=0x40080000 size=0x00144 (   324) load[0m
[0;32mI (168) esp_image: segment 3: paddr=0x00040018 vaddr=0x400d0018 size=0xcd1cc (840140) map[0m

【2020-04-10 17:48:29:600】[0;32mI (349) esp_image: segment 4: paddr=0x0010d1ec vaddr=0x40080144 size=0x002bc (   700) load[0m
[0;32mI (350) esp_image: segment 5: paddr=0x0010d4b0 vaddr=0x40080400 size=0x10a3c ( 68156) load[0m
[0;32mI (385) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (385) boot: Disabling RNG early entropy source...[0m

Alvin1Zhang, 2020-04-12:
@shenyong-iot01 您好，感谢反馈，我们会尽快查看并提供反馈，谢谢。

shenyong-iot01, 2020-04-14:
@Alvin1Zhang 麻烦帮忙确认，因为有一批老设备需尽快出来处理方案! 谢谢！

shenyong-iot01, 2020-04-14:
3.2SDK bootloader存在同样问题  boot: ESP-IDF v3.2-dirty 2nd stage bootloade

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6212
load:0x40078000,len:10180
ho 0 tail 12 room 4
load:0x40080400,len:6640
entry 0x40080760
[0;32mI (31) boot: ESP-IDF v3.2-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 15:23:16[0m
[0;32mI (31) boot: Enabling RNG early entropy source...[0m
[0;32mI (36) boot: SPI Speed      : 40MHz[0m
[0;32mI (40) boot: SPI Mode       : DIO[0m
[0;32mI (44) boot: SPI Flash Size : 4MB[0m
[0;32mI (48) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (66) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (74) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (81) boot:  3 ota_0            OTA app          00 10 00010000 00180000[0m
[0;32mI (89) boot:  4 ota_1            OTA app          00 11 00190000 00180000[0m
[0;32mI (96) boot: End of partition table[0m
[0;32mI (100) boot: No factory image, trying OTA 0[0m
[0;32mI (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2d114 (184596) map[0m
[0;32mI (152) esp_image: segment 1: paddr=0x0003d13c vaddr=0x3ffb0000 size=0x02ed4 ( 11988) load[0m
[0;32mI (155) esp_image: segment 2: paddr=0x00040018 vaddr=0x400d0018 size=0xd0018 (851992) map[0m
【2020-04-14 16:14:25:049】[0;32mI (333) esp_image: segment 3: paddr=0x00110038 vaddr=0x3ffb2ed4 size=0x007ac (  1964) load[0m
[0;32mI (334) esp_image: segment 4: paddr=0x001107ec vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (340) esp_image: segment 5: paddr=0x00110bf4 vaddr=0x40080400 size=0x10b34 ( 68404) load[0m
[0;32mI (378) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (378) boot: Disabling RNG early entropy source...[0m

shenyong-iot01, 2020-04-14:
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/4474142/sdkconfig.zip)
sdkconfig.zip 附件为我3.2.2SDK以及3.3.1SDK的配置文件，3.2.2SDK编译出的固件可以被老版本boot正常启动，3.3.1SDK编译出的固件不能被老版本boot启动。

shenyong-iot01, 2020-04-14:
@Alvin1Zhang 经验证进一步定位老bootloader 配置SPI：
�[0;32mI (36) boot: SPI Speed : 40MHz�[0m
�[0;32mI (40) boot: SPI Mode : DIO�[0m

使用上条信息附件sdkconfig.zip编译出的固件  SPI Speed 80M  SPI Mode : QIO
但3.2.2SDK的可以正常启动   3.3.1的无法正常启动。SPI FLASH配置相同。
3.3.2bootlodaer 40MHZ  DIO 可以正常启动80MHZ  QIO 3.2.2SDK编译出的固件
3.3.2bootlodaer 40MHZ  DIO 可以无法启动80MHZ  QIO 3.3.1SDK编译出的固件
3.3.2bootlodaer 40MHZ  DIO 可以正常启动40MHZ  DIO 3.3.1SDK编译出的固件
请帮忙确认！

costaud, 2020-04-15:
@shenyong-iot01 能否提供所使用的3.1和3.3.1的commit id

shenyong-iot01, 2020-04-15:
不用关注3.1SDK，直接使用3.2.2SDK即可！
[image]
编译时更改spi flash配置项目
3.3.2sdk bootlodaer 40MHZ DIO can boot 3.2.2SDK 80MHZ QIO firmware
3.3.2sdk bootlodaer 40MHZ DIO can not boot 3.3.1SDK 80MHZ QIO firmware
3.3.2sdk bootlodaer 40MHZ DIO can boot 3.3.1SDK 40MHZ DIO firmware

3.2.2SDK 
git show
commit 055943e29346e77c50589c61a8a26101a8b35d7b
Merge: b34a087 0d7f6a5
Author: Angus Gratton <angus@espressif.com>
Date:   Mon Jun 24 09:12:59 2019 +0800

    Merge branch 'bugfix/docs_rtd_failure_on_missing_submodule_v3_2_bis' into 'release/v3.2'
    
    docs: Add mqtt submodule to RTD build configuration to correct file and delete...
    
    See merge request idf/esp-idf!5310
3.3.1 SDK
git show
commit 143d26aa49df524e10fb8e41a71d12e731b9b71d
Merge: 0407ab4 e1eabe6
Author: Ivan Grokhotkov <ivan@espressif.com>
Date:   Wed Dec 11 14:23:07 2019 +0800

    Merge branch 'bugfix/doc_package_versions_v3.3' into 'release/v3.3'
    
    doc: Limit sphinxcontrib versions to <2.0.0 as we use Sphinx 1.8.5 (v3.3)
    
    See merge request espressif/esp-idf!6975

costaud, 2020-04-15:
@shenyong-iot01  

> 3.3.2sdk bootlodaer 40MHZ DIO can boot 3.2.2SDK 80MHZ QIO firmware
3.3.2sdk bootlodaer 40MHZ DIO can not boot 3.3.1SDK 80MHZ QIO firmware
3.3.2sdk bootlodaer 40MHZ DIO can boot 3.3.1SDK 40MHZ DIO firmware

所以这里前半句提到的 应该是3.2.2 不是3.3.2 是吗 ？

shenyong-iot01, 2020-04-15:
是3.2.2不好意思，boot_app内是我这边可以测试的boot  和 app 请验证！
[boot_app.zip](https://github.com/espressif/esp-idf/files/4480731/boot_app.zip)

shenyong-iot01, 2020-04-16:
[boot.log](https://github.com/espressif/esp-idf/files/4484543/boot.log)
打开cup_start 和 heap_init的打印启动log

shenyong-iot01, 2020-04-16:
[image]
按照附件内修改，编译出的固件可正常启动！多谢支持！

closed 2020-04-24 by espressif-bot