espressif/esp-idf #5101: Issues with reading UART FIFO when SMP is enabled

rojer, 2020-04-11:
ESP32 R0, ESP-IDF 3.3

After enabling SMP reads from `UART_FIFO_AHB_REG` become unreliable.
Notice that I'm using the AHB variant here, so the issue I am describing is _not_ the issue described in ECO and Workarounds for Bugs section 3.3 - or the AHB access workaround is insufficient.

Observed behavior is that when SMP mode is enabled (but never in UP mode) sometimes reading `UART_FIFO_AHB_REG` returns previous byte.
`UART_MEM_RX_STATUS_REG.UART_MEM_RX_RD_ADDR` remains unchanged but strangely enough, `UART_STATUS_REG.UART_RXFIFO_CNT` is updated (decremented).
The data remains in FIFO and can be retrieved by repeated reads but `UART_RXFIFO_CNT` will remain out of sync until FIFO count goes to 0.
I repeat: this is only observed when SMP is on, never in uni-processor mode.
The workaround I found is to:

1) When reading from FIFO, make a note of `UART_MEM_RX_RD_ADDR` before the read and compare to the value after the read. If it hasn't changed, discard data and repeat.

2) Use `UART_MEM_RX_STATUS_REG` to calculate FIFO length and ignore `UART_RX_MEM_CNT`+`UART_RXFIFO_CNT`.

I do not have repro code since I'm not using ESP-IDF code structure, but I am pretty sure this is a hardware issue.

It may be the underlying cause of #2388 (cc @qt1)

cc @projectgus

rojer, 2020-04-11:
the commit above is the workaround i implemented in Mongoose OS.

negativekelvin, 2020-04-13:
Dport workaround should be for writes only?  

Also similar to https://github.com/espressif/esp-idf/issues/2204

koobest, 2020-04-13:
Hi, @rojer

Please try to use DPORT address for reading.  

```
 __asm__ __volatile__("memw");
rd_byte = READ_PERI_REG(UART_FIFO_REG(uart_num));

```
We will update the ECO document. 

thanks !!

rojer, 2020-04-13:
@negativekelvin the DPORT -> AHB access for FIFO registers is specifically for reads. i've never encountered any problems with writing them.

@koobest i tried using DPORT_READ_PERI_REG on both DPORT and AHB version of the UART_FIFO_REG, it doesn't make a difference.
([DPORT_READ_PERI_REG](https://github.com/espressif/esp-idf/blob/c1d0daf36d0dca81c23c226001560edfa51c30ea/components/soc/soc/esp32/include/soc/dport_access.h#L167) is the macro introduced to work around ECO 3.10 about simultaneous address space access, it disabled interrupts in addition to memw)
so, in short, no - there does really seem to be something _in addition_ to ECO 3.3 and 3.10 here.

rojer, 2020-04-14:
to be clear: even DPORT_READ_PERI_REG of UART_FIFO_AHB_REG, which should be as reliable as can be, is not enough.

igrr, 2020-04-14:
I think it should be READ_PERI_REG(UART_FIFO_REG(uart_num)), i.e. using regular read over the DPORT (0x3ff.....) address. However this will work on ECO1 and later. On the original chip revision, this is not reliable because of issue 3.3.

> there does really seem to be something in addition to ECO 3.3 and 3.10 here.

That's right, that's what @koobest meant, ECO document will be updated to describe the issue.

rojer, 2020-04-14:
@igrr i see, ok. unfortunately, there are plenty of R0s in the field.
so, on a related note - DPORT_* macros are provided to deal with 3.10 and they use a special function. however, the memory-mapped structs ([uart_dev_t](https://github.com/espressif/esp-idf/blob/2e14149bff63fe211d6d30ff0cd84c90b5fface9/components/soc/soc/esp32/include/soc/uart_struct.h#L23), [i2c_dev_t](https://github.com/espressif/esp-idf/blob/2e14149bff63fe211d6d30ff0cd84c90b5fface9/components/soc/soc/esp32/include/soc/i2c_struct.h#L23), etc) still [point to DPORT addresses](https://github.com/espressif/esp-idf/blob/2e14149bff63fe211d6d30ff0cd84c90b5fface9/components/esp32/ld/esp32.peripherals.ld).
those are marked volatile, so we have `memw`, but not the full workaround provided by [esp_dport_access_reg_read](https://github.com/espressif/esp-idf/blob/0b8f17e6183c401ea6322be384da8a32a75fc76f/components/esp32/dport_access.c#L240). 
does that mean that accessing peripherals in SMP mode via these structs on ESP32 < R3 is not safe?

igrr, 2020-04-14:
3.10 doesn't apply to UART, I2C. I see there is a mistake in the description of 3.10, the address ranges should be the other way around:

> Running in dual-core CPU mode, when one CPU bus reads address space A
(0x3FF4_0000 ~ 0x3FF7_FFFF), while the other CPU bus reads address space B
(0x3FF0_0000 ~ 0x3FF1_EFFF), an incorrect read can be generated on the CPU reading
address space A. 

"Address space A" should be "0x3FF0_0000 ~ 0x3FF1_EFFF", which includes crypto peripherals and the MMU configuration:
[image]

rojer, 2020-04-14:
> 3.10 doesn't apply to UART, I2C.

this is good news!

> "Address space A" should be "0x3FF0_0000 ~ 0x3FF1_EFFF", which includes crypto peripherals and the MMU configuration

hm... then address space A becomes the same as B, so the problem happens if both CPUs access 0x3FF0_0000 ~ 0x3FF1_EFFF?

igrr, 2020-04-14:
Sorry, I meant that the address ranges seem to be swapped around in the description. "A" should be 0x3FF0_0000 ~ 0x3FF1_EFFF, "B" should be 0x3FF4_0000 ~ 0x3FF7_FFFF. I'll follow up with the documentation colleagues to check if my understanding is correct and if the documentation needs to be updated.

negativekelvin, 2020-04-14:
You can see how idf does reads and writes. Anyway I don't think it is related to this issue.

https://github.com/espressif/esp-idf/blob/2e14149bff63fe211d6d30ff0cd84c90b5fface9/components/soc/src/esp32/include/hal/uart_ll.h#L167-L174

https://github.com/espressif/esp-idf/blob/2e14149bff63fe211d6d30ff0cd84c90b5fface9/components/soc/src/esp32/include/hal/uart_ll.h#L185-L192

rojer, 2020-04-14:
> buf[i] = READ_PERI_REG(fifo_addr);

this wouldn't work on R0 due to 3.3, would it?

anyway, i'll let you guys figure it out and update ECO. i have good repro setup for this, if you'd like me to test something in particular, please let me know.

negativekelvin, 2020-04-14:
3.3 says 

> When written via DPORT, consecutive **writes** to the same address may be lost.

Alvin1Zhang, 2020-06-01:
@rojer Thanks for reporting, feel free to reopen if any updates.

closed 2020-06-01 by Alvin1Zhang

projectgus, 2020-06-01:
(For anyone missing additional context, a new ECO document V2.0 was released 2019-05-08 and fixes the accidentally swapped ECO 3.10 address space ranges.)

negativekelvin, 2020-06-01:
https://esp32.com/viewtopic.php?f=2&t=15476

rojer, 2020-06-01:
@Alvin1Zhang @projectgus so what are the concrete steps to deal wit the UART issue? in my testing none of the address spacing / memory barriers seemed to help - AHB or DPORT, volatile or not, some reads from the FIFO register just seem to "stutter" every now and then.

koobest, 2020-06-01:
HI, @rojer 
Can you provide a simple code(with sdk config file) that can reproduce this issue?

rojer, 2020-06-01:
@koobest i don't work with IDF directly and the app i'm using for testing is a mongoose-os app, for which the source is closed so i cannot release it. but the essence is: pump data between UART and TCP connection, with some test apps to generate traffic.
i don't have the time right now to write an IDF-based test app. i have done some more testing and have additional information to add: DPORT behavior for reads is different for SMP and UP mode.
here are combinations i tried:
1. UP mode
  a. (`CONFIG_FREERTOS_UNICORE=y`), AHB read (`READ_PERI_REG(UART_FIFO_AHB_REG(uart_no))`) - ok
  b. DPORT read (`DPORT_READ_PERI_REG(UART_FIFO_REG(uart_no))`) - NOT ok, fails quickly
  c. ([MEM_RX_ADDR workaround](https://github.com/cesanta/mongoose-os/commit/726446d016e9316f128fa066e9f1d146c733914d)) - ok
2. SMP mode
  a. AHB read - NOT ok
  b. DPORT read - ok
  c. MEM_RX wa - ok

So, depending on UP/SMP mode either DPORT or AHB read leads to failure and only [MEM_RX workaround](https://github.com/cesanta/mongoose-os/commit/726446d016e9316f128fa066e9f1d146c733914d) works in both modes.
Hope this helps.

I should add that in SMP mode DPORT_READ_PERI_REG is more than just a volatile read from a register, it also disables interrupts. Simple volatile read in SMP mode (`*((volatile uint32_t *) UART_FIFO_REG(uart_no))`) also eventually fails.

koobest, 2020-06-02:
Hi,
Do you really try the fix the same as  in IDF?
https://github.com/cesanta/mongoose-os/blob/726446d016e9316f128fa066e9f1d146c733914d/platforms/esp32/src/esp32_uart.c#L88
```
byte = UART_FIFO_REG(UART_FIFO_REG(uart_no));
```

https://github.com/cesanta/mongoose-os/blob/726446d016e9316f128fa066e9f1d146c733914d/platforms/esp32/src/esp32_uart.c#L63

Generally, ` esp32_uart_rx_fifo_len(int uart_no)` can return `rx_fifo_cnt` correctly, but when the data length is equal to `FIFO_SIZE`, it will return 0 and this result is wrong.

thanks !!

rojer, 2020-06-02:
> Hi,
> Do you really try the fix the same as in IDF?

as far as i can tell, IDF uses DPORT barrier read - via the volatile struct member [here](https://github.com/espressif/esp-idf/blob/d7e6eecd66c63b5d4f725060603bd13b43b29b3c/components/driver/uart.c#L875). (v3.3 branch)

as mentioned, i tried it, and it is not enough on its own.
in my code i'm using AHB read which works in UP mode but not with SMP. but as i also describe above, there is no one mode that suits both, so i just chose AHB read.

> https://github.com/cesanta/mongoose-os/blob/726446d016e9316f128fa066e9f1d146c733914d/platforms/esp32/src/esp32_uart.c#L63
>
> Generally, ` esp32_uart_rx_fifo_len(int uart_no)` can return `rx_fifo_cnt` [...]

funny that you mentioned it, i just spent some time debugging exactly this. i submitted a [fix](https://github.com/cesanta/mongoose-os/commit/2bed8d58598f8712c480d67dbbe07fa86fd9bcac) that uses fifo_cnt to distinguish between the two cases.

> 
> thanks !!

ginkgm, 2020-06-03:
Hi all, 

This issue is supposed to be fixed in commit eea38d76986f366d07fa781d19e28e7fda5d4ca8, which will be synced from internal server later.

rojer, 2020-06-15:
@ginkgm the comiit has been published and i see that you now use rd/wr_addr to calculate queue length.
this is part of the fix but not all of it.
what i observed is when the "stutter" happens, fifo_cnt is updated while rd_addr and wr_addr are not.
it's almost as if the read does not propagate to the FIFO circuit and returns old data that was stored in the bus register but gets far enough to update fifo_cnt (hence why i call it stuttering - it returns a byte we had already read before). this is why the most important part of my fix is where we read the byte from the fifo: after the read we check if rd_addr has been updated, meaning the byte that was returned actually comes from the FIFO and not a stale one.

koobest, 2020-06-16:
Hi, @rojer 
Another way to workaround this issue is to use critical section.  but I don't know if critical section is available in Mongoose OS.
```
portENTER_CRITICAL
    read_fifo();
portEXIT_CRITICAL
```
By the way, I would suggest you use APB address to read fifo.

thanks !!