espressif/esp-idf #5102: ERROR: tool openocd-esp32 has no installed versions

Austin299792, 2020-04-11:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): V4.2-dev-1097-g2e14149
- Build System:  Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): ? My gcc version is 4.8.5 20150623
- Operating System: CentOS 7 arm
- Using an IDE?: No
- Power Supply: external 5V

## Problem Description

I’m am following the instructions [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#get-started-get-esp-idf) to install ESP-IDF. I get to the part where I source the export.sh and get the following error:

> Adding ESP-IDF tools to PATH...
ERROR: tool openocd-esp32 has no installed versions. Please run ‘/opt/esp-idf/install.sh’ to install it

When I run install.sh, the script completes successfully. I can see the application in ~/.espressif/tools.

Based on similar bug report, I checked the path in the tools.json file and the path to scripts was not the same. My file system:

~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200406/openocd-esp32/share/openocd/scripts

Configured in tools.json:

~/.espressif/tools/openocd-esp32/share/openocd/scripts

Moving my directory to match the path in tools.json does not fix the problem. 

Any advice to help resolve the problem?

igrr, 2020-04-13:
can you please try installing libusb package?

Austin299792, 2020-04-13:
Hello. Related to that request, I tried compiling openocd-esp32 myself. One of the requirements was for libusb-1.0, but I discovered that CentOS calls their Libyan package libusbx now. This prevented me from compiling openocd-esp32 and I went back to troubleshooting the installation from install.sh

Do you think libusbx and no libusb-1.0 is related to this problem?

antmak, 2020-05-19:
@Austin299792 Hi! Right `libusbx` is the same as `libusb1.0` in CentOS 7. Maybe it needs some little changes in openocd-esp32 build process. Have you managed to solve your problem?

Alvin1Zhang, 2020-08-18:
@Austin299792 Thanks for reporting, would you please help share if any updates for this issue? Thanks.

CalinRadoni, 2020-10-12:
I've got a very similar error on a Ubuntu Focal container and solved it by installing `python3-libusb1`

hj91, 2020-12-17:
using apt-get install libhidapi-libusb0 on debian 10 solves the problem. This is for everyone trying it on debian.

Patti4832, 2020-12-28:
> 
> 
> I've got a very similar error on a Ubuntu Focal container and solved it by installing `python3-libusb1`

Thanks for the hint. Worked for me

closed 2022-03-10 by dobairoland