espressif/esp-idf #5103: timer_disable_intr doesn't work

francis2tm, 2020-04-12:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Linux
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
Hey,
So I followed the implementation of timer interrupt (https://github.com/espressif/esp-idf/bl ... ple_main.c ). It worked perfectly but now I want to disable and then re enable the interuption caused by Group 0 Timer 1 's alarm, and I can't seem to disable the interrupt.

This is the timer's config and init:
```cpp
    void timerGrp0Init(int timer_idx, double timer_interval_sec){
        /* Select and initialize basic parameters of the timer */
        timer_config_t config;
        config.divider = TIMER_DIVIDER;
        config.counter_dir = TIMER_COUNT_UP;
        config.counter_en = TIMER_PAUSE;
        config.alarm_en = TIMER_ALARM_EN;
        config.intr_type = TIMER_INTR_LEVEL;
        config.auto_reload = 1;
    #ifdef TIMER_GROUP_SUPPORTS_XTAL_CLOCK
        config.clk_src = TIMER_SRC_CLK_APB;
    #endif
        timer_init(TIMER_GROUP_0, timer_idx, &config);
     
        /* Timer's counter will initially start from value below.
           Also, if auto_reload is set, this value will be automatically reload on alarm */
        timer_set_counter_value(TIMER_GROUP_0, timer_idx, 0x00000000ULL);
     
        /* Configure the alarm value and the interrupt on alarm. */
        timer_set_alarm_value(TIMER_GROUP_0, timer_idx, timer_interval_sec * TIMER_SCALE);
        timer_enable_intr(TIMER_GROUP_0, timer_idx);
        timer_isr_register(TIMER_GROUP_0, timer_idx, timerGrp0Isr, (void*)timer_idx, ESP_INTR_FLAG_IRAM, NULL);
     
        timer_start(TIMER_GROUP_0, timer_idx);
    }

```

 This is the ISR:
```cpp
    void IRAM_ATTR timerGrp0Isr(void *para){                            //This interrupt is handled by CPU1
        int timer_idx = (int)para;
        timer_spinlock_take(TIMER_GROUP_0);
     
        //Wake acqAdc2 in order to start ADC readings from adc2. CPU0 will start immediatly acquiring
        vTaskNotifyGiveFromISR(acquiring_2_task, (BaseType_t*)NULL);
     
        //Wake acqAdc1 in order to start ADC readings form adc1. This will only start when this handler is terminated.
        vTaskNotifyGiveFromISR(acquiring_1_task, (BaseType_t*)NULL);
     
        //Clear the interrupt
        timer_group_clr_intr_status_in_isr(TIMER_GROUP_0, timer_idx);
     
        //After the alarm has been triggered we need enable it again, so it is triggered the next time
        timer_group_enable_alarm_in_isr(TIMER_GROUP_0, timer_idx);
     
        timer_spinlock_give(TIMER_GROUP_0);
    }

```
 And then, when acquiring_1_task receives the notification, it immediately disables the interrupt like so: 
```cpp
    void acqAdc1(){
        //Init Timer 0_1 (timer 1 from group 0) and register it's interupt handler
        timerGrp0Init(TIMER_1, TIMER1_INTERVAL_SEC);
     
        //Config all possible adc channels
        configSixAdcChannels(ADC_RESOLUTION);
     
        while(1){
            if(ulTaskNotifyTake(pdTRUE, ONE_HOUR_MS/portTICK_PERIOD_MS ) == 1){     //THIS IS WHERE THE TASK RECIEVES THE NOTIFICATION
               
     
                if(timer_disable_intr(TIMER_GROUP_0, 1) == ESP_OK){
                    printf("Disable ok\n");
                }
     
                //TODO clear acq_config.adc1_num_channels
            }else{
                DEBUG_PRINT(W, "acqAdc1", "ulTaskNotifyTake timed out!");
            }
        }
    }
```

Alvin1Zhang, 2020-04-12:
@francis2tm Thanks for reporting, we have noted this issue and will look into.

francis2tm, 2020-04-12:
Thanks! When you have some news please post here

closed 2020-04-27 by espressif-bot