espressif/esp-idf #5104: Make flash fails after upgrading to 4.2-dev

tuupola, 2020-04-12:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-PICO-D4]
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-1097-g2e14149bf
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     [macOS]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

After updating to  4.2-dev `make flash` fails with the following error.

```
$ make flash
...
serial.serialutil.SerialException: [Errno 2] could not open port "/dev/cu.usbserial-C54ABFCA45": [Errno 2] No such file or directory: '"/dev/cu.usbserial-C54ABFCA45"'
make: *** [flash] Error 1
```

Culprit is this commit: https://github.com/espressif/esp-idf/commit/862fa815ff1ff79481ef113a9af7d9092e6f3583  which causes hyphens to be part of the port name. Reverting the commit makes problem go away.

fpgamaster, 2020-04-12:
Same here. It seems idf.py monitor to work.

Alvin1Zhang, 2020-04-12:
@tuupola Thanks for reporting and the investigation, we will look into. Thanks.

projectgus, 2020-04-15:
Hi @tuupola ,

Thanks for reporting this. I've just replied to issue #5124 and then realised this is the same root cause, but I'll link to the comment on the other issue rather than copy-pasting it:
https://github.com/espressif/esp-idf/issues/5124#issuecomment-613828807

A fix is in review and both issues will be updated when it's merged.

Angus

closed 2020-04-24 by espressif-bot