espressif/esp-idf #5107: "esp_restart" cause Core  0 panic'ed (Cache disabled but cached memory region accessed)

xiruilin, 2020-04-12:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit): [v3]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version:    v4.2-dev-1012-ga1f586642-dirty
- Build System:         [idf.py]
- Compiler version:    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

When ``CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y``,
call ``esp_restart`` on core 1, got the issue:
```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
```

### Code to reproduce this issue

```cpp
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

void restart_task(void *param)
{
    esp_restart();
    vTaskDelete(NULL);
}

void app_main(void)
{
    vTaskDelay(5000 / portTICK_PERIOD_MS);
    StackType_t *task_stack = (StackType_t *) heap_caps_calloc(1, 1024 * 3, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
    static StaticTask_t task_buf;
    xTaskCreateStaticPinnedToCore(restart_task, "restarttask", (1024 * 3),
        NULL, 5, task_stack, &task_buf, 1);
}
```

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:1416
load:0x40078000,len:12704
ho 0 tail 12 room 4
load:0x40080400,len:3856
0x40080400: _init at ??:?

entry 0x400805f8
I (105) psram: This chip is ESP32-D0WD
I (106) spiram: Found 64MBit SPI RAM device
I (106) spiram: SPI RAM mode: flash 80m sram 80m
I (108) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (116) cpu_start: Pro cpu up.
I (119) cpu_start: Application information:
I (124) cpu_start: Project name:     Test_Restart
I (130) cpu_start: App version:      1
I (134) cpu_start: Compile time:     Apr 13 2020 00:02:53
I (140) cpu_start: ELF file SHA256:  e666f5b60df6796b...
I (146) cpu_start: ESP-IDF:          v4.2-dev-1012-ga1f586642-dirty
I (153) cpu_start: Starting app cpu, entry point is 0x40081478
0x40081478: call_start_cpu1 at E:/esp-idf/components/esp32/cpu_start.c:277

I (0) cpu_start: App cpu up.
I (653) spiram: SPI SRAM memory test OK
I (654) heap_init: Initializing. RAM available for dynamic allocation:
I (654) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (660) heap_init: At 3FFB2CB0 len 0002D350 (180 KiB): DRAM
I (666) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (673) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (679) heap_init: At 4008DBE8 len 00012418 (73 KiB): IRAM
I (685) cpu_start: Pro cpu start user code
I (690) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (710) spi_flash: detected chip: gd
I (710) spi_flash: flash io: qio
W (710) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (721) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (732) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:1416
load:0x40078000,len:12704
ho 0 tail 12 room 4
load:0x40080400,len:3856
0x40080400: _init at ??:?

entry 0x400805f8
I (105) psram: This chip is ESP32-D0WD
I (105) spiram: Found 64MBit SPI RAM device
I (106) spiram: SPI RAM mode: flash 80m sram 80m
I (108) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (115) cpu_start: Pro cpu up.
I (119) cpu_start: Application information:
I (124) cpu_start: Project name:     Test_Restart
I (129) cpu_start: App version:      1
I (134) cpu_start: Compile time:     Apr 13 2020 00:02:53
I (140) cpu_start: ELF file SHA256:  e666f5b60df6796b...
I (146) cpu_start: ESP-IDF:          v4.2-dev-1012-ga1f586642-dirty
I (153) cpu_start: Starting app cpu, entry point is 0x40081478
0x40081478: call_start_cpu1 at E:/esp-idf/components/esp32/cpu_start.c:277

I (144) cpu_start: App cpu up.
I (654) spiram: SPI SRAM memory test OK
I (654) heap_init: Initializing. RAM available for dynamic allocation:
I (654) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (660) heap_init: At 3FFB2CB0 len 0002D350 (180 KiB): DRAM
I (666) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (673) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (679) heap_init: At 4008DBE8 len 00012418 (73 KiB): IRAM
I (685) cpu_start: Pro cpu start user code
I (690) spiram: Adding pool of 4096K of external SPI memory to heap allocator
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

Core  0 register dump:
PC      : 0x4008c265  PS      : 0x00060d34  A0      : 0x800d1402  A1      : 0x3ffe3bf0
0x4008c265: xt_ints_on at E:/esp-idf/components/freertos/xtensa/xtensa_intr_asm.S:164

A2      : 0x0300000c  A3      : 0x0100000c  A4      : 0x00000019  A5      : 0x383fc000
A6      : 0x00000016  A7      : 0x00000000  A8      : 0x15122500  A9      : 0x00060d20
A10     : 0x3ffb2468  A11     : 0x3ffb2470  A12     : 0x3ffb2468  A13     : 0x00060d20
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4008ce60  LEND    : 0x4008ce7c  LCOUNT  : 0xffffffff
0x4008ce60: memcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:175

0x4008ce7c: memcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:197

Backtrace:0x4008c262:0x3ffe3bf0 0x400d13ff:0x3ffe3c00 0x400813d1:0x3ffe3c20 0x400816bd:0x3ffe3c50 0x40078fc9:0x3ffe3c90 |<-CORRUPTED
0x4008c262: xt_ints_on at E:/esp-idf/components/freertos/xtensa/xtensa_intr_asm.S:163

0x400d13ff: esp_cache_err_int_init at E:/esp-idf/components/esp32/cache_err_int.c:70

0x400813d1: start_cpu0_default at E:/esp-idf/components/esp32/cpu_start.c:397

0x400816bd: call_start_cpu0 at E:/esp-idf/components/esp32/cpu_start.c:265

ELF file SHA256: e666f5b60df6796b

Rebooting...
```

Spritetm, 2020-04-14:
Do not use psram stack for that task. Psram stack can only be used if you do not use any ROM functions in it, either directly or indirectly. esp_restart() calls a rom function.

xiruilin, 2020-04-15:
@Spritetm 
I see, but if only change call ``esp_restart`` on core 0, this error will not happen.

xiruilin, 2020-04-16:
@Spritetm 
Hmm, ``esp_restart()`` is run in IRAM.
``esp-idf/components/esp_system/system_api.c`` lines 37:
```cpp
void IRAM_ATTR esp_restart(void)
{
    for (int i = SHUTDOWN_HANDLERS_NO - 1; i >= 0; i--) {
        if (shutdown_handlers[i]) {
            shutdown_handlers[i]();
        }
    }

    // Disable scheduler on this core.
    vTaskSuspendAll();

    esp_restart_noos();
}
```

xiruilin, 2020-04-16:
If change ``esp_restart`` to ``*((int *) 0) = 0``, the issue still exist.
```cpp
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

void panic_task(void *param)
{
    *((int *) 0) = 0;
    vTaskDelete(NULL);
}

void app_main(void)
{
    vTaskDelay(5000 / portTICK_PERIOD_MS);
    StackType_t *task_stack = (StackType_t *) heap_caps_calloc(1, 1024 * 3, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
    static StaticTask_t task_buf;
    xTaskCreateStaticPinnedToCore(panic_task, "panictask", (1024 * 3),
        NULL, 5, task_stack, &task_buf, 1);
}
```

Log:
```
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7356
load:0x40078000,len:14832
load:0x40080400,len:4740
0x40080400: _init at ??:?

entry 0x400806a4
I (29) boot: ESP-IDF v4.2-dev-1097-g2e14149bf-dirty 2nd stage bootloader
I (29) boot: compile time 15:30:36
I (30) boot: chip revision: 1
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32: SPI Speed      : 80MHz
I (44) boot.esp32: SPI Mode       : QIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 002a0000
I (91) boot:  3 eeprom           Unknown data     01 99 002b0000 00001000
I (99) boot:  4 flash_tone       Unknown data     01 ff 002b1000 0014f000
I (106) boot: End of partition table
I (111) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x09e48 ( 40520) map
I (133) esp_image: segment 1: paddr=0x00019e70 vaddr=0x3ffb0000 size=0x022cc (  8908) load
I (137) esp_image: segment 2: paddr=0x0001c144 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at E:/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (140) esp_image: segment 3: paddr=0x0001c550 vaddr=0x40080404 size=0x03ac8 ( 15048) load
I (155) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x16b64 ( 93028) map
0x400d0020: _stext at ??:?

I (189) esp_image: segment 5: paddr=0x00036b8c vaddr=0x40083ecc size=0x09d08 ( 40200) load
0x40083ecc: _xt_syscall_exc at E:/esp-idf/components/freertos/xtensa/xtensa_vectors.S:787

I (212) boot: Loaded app from partition at offset 0x10000
I (212) boot: Disabling RNG early entropy source...
I (213) psram: This chip is ESP32-D0WD
I (217) spiram: Found 64MBit SPI RAM device
I (222) spiram: SPI RAM mode: flash 80m sram 80m
I (227) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
[... 116 lines cut ...]
PC      : 0x4008c251  PS      : 0x00060d34  A0      : 0x800d140a  A1      : 0x3ffe3be0
0x4008c251: xt_ints_on at E:/esp-idf/components/freertos/xtensa/xtensa_intr_asm.S:164

A2      : 0x0300000c  A3      : 0x0100000c  A4      : 0x00000019  A5      : 0x383fc000
A6      : 0x00000016  A7      : 0x00000000  A8      : 0x15122500  A9      : 0x00060d20
A10     : 0x3ffb24e0  A11     : 0x3ffb24e8  A12     : 0x3ffb24e0  A13     : 0x00060d20
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4008d94c  LEND    : 0x4008d968  LCOUNT  : 0xffffffff
0x4008d94c: memcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:175

0x4008d968: memcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:197

Backtrace:0x4008c24e:0x3ffe3be0 0x400d1407:0x3ffe3bf0 0x400813cd:0x3ffe3c10 0x400816b9:0x3ffe3c40 0x40079292:0x3ffe3c80 |<-CORRUPTED
0x4008c24e: xt_ints_on at E:/esp-idf/components/freertos/xtensa/xtensa_intr_asm.S:163

0x400d1407: esp_cache_err_int_init at E:/esp-idf/components/esp32/cache_err_int.c:70

0x400813cd: start_cpu0_default at E:/esp-idf/components/esp32/cpu_start.c:397

0x400816b9: call_start_cpu0 at E:/esp-idf/components/esp32/cpu_start.c:265

ELF file SHA256: bc49aa3daf9ed211

Rebooting...
```

Spritetm, 2020-04-16:
Aaaah, the fact that this happens on the next boot makes this a different story indeed. We'll investigate this.

igrr, 2020-04-16:
@xiruilin I think we have found the reason, will update this thread when the fix is available.

closed 2020-06-12 by igrr