espressif/esp-idf #5110: "https_request" got "tcp_receive: ooseq tcplen > rcv_wnd" issue

xiruilin, 2020-04-13:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit): [v3]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version:    v4.2-dev-1012-ga1f586642-dirty
- Build System:         [idf.py]
- Compiler version:    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

``/esp-idf/examples/protocols/https_request``

Run ``https_request`` example some loops, an assertion has been made
``assertion "tcp_receive: ooseq tcplen > rcv_wnd"``

### Steps to reproduce

1. Modify ``WEB_SERVER`` and ``WEB_URL`` in ``https_request_example_main.c``
2. Add ``alpn_protos`` on lines 64
3. Build and flash

### Code to reproduce this issue

```cpp
...

/* Constants that aren't configurable in menuconfig */
#define WEB_SERVER "avs-alexa-na.amazon.com"
#define WEB_PORT "443"
#define WEB_URL "https://avs-alexa-na.amazon.com"

static const char *TAG = "example";

static const char *REQUEST = "GET " WEB_URL " HTTP/1.0\r\n"
    "Host: "WEB_SERVER"\r\n"
    "User-Agent: esp-idf/1.0 esp32\r\n"
    "\r\n";

static void https_get_task(void *pvParameters)
{
    char buf[512];
    int ret, len;

    while(1) {
        const char *alpn_protos[] = { "h2", NULL };
        esp_tls_cfg_t cfg = {
            .alpn_protos = alpn_protos,
            .timeout_ms = 10 * 1000,
            .crt_bundle_attach = esp_crt_bundle_attach,
        };
...

```

## Debug Logs

```
I (137192) example: Completed 9 requests
I (137202) example: 10...
I (138202) example: 9...
I (139202) example: 8...
I (140202) example: 7...
I (141202) example: 6...
I (142202) example: 5...
I (143202) example: 4...
I (144202) example: 3...
I (145202) example: 2...
I (146202) example: 1...
I (147202) example: 0...
I (148202) example: Starting again!
I (149572) esp-x509-crt-bundle: Certificate validated
I (150932) example: Connection established...
I (150932) example: 110 bytes written
I (150932) example: Reading HTTP response...
              I            Unexpected HTTP/1.x request: GET https://avs-alexa-na.amazon.com
I (151222) example: connection closed

I (151222) example: Completed 10 requests
I (151232) example: 10...
I (152232) example: 9...
I (153232) example: 8...
I (154232) example: 7...
I (155232) example: 6...
I (156232) example: 5...
I (157232) example: 4...
I (158232) example: 3...
I (159232) example: 2...
I (160232) example: 1...
I (161232) example: 0...
I (162232) example: Starting again!
I (163592) esp-x509-crt-bundle: Certificate validated
I (165062) example: Connection established...
I (165062) example: 110 bytes written
I (165062) example: Reading HTTP response...
              I            Unexpected HTTP/1.x request: GET https://avs-alexa-na.amazon.com
I (165332) example: connection closed

I (165342) example: Completed 11 requests
I (165342) example: 10...
I (166342) example: 9...
I (167342) example: 8...
I (168342) example: 7...
I (169342) example: 6...
I (170342) example: 5...
I (171342) example: 4...
I (172342) example: 3...
I (173342) example: 2...
I (174342) example: 1...
I (175342) example: 0...
I (176342) example: Starting again!
assertion "tcp_receive: ooseq tcplen > rcv_wnd
" failed: file "E:/esp-idf/components/lwip/lwip/src/core/tcp_in.c", line 1581, function: tcp_receive

abort() was called at PC 0x401461a3 on core 1
0x401461a3: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

Backtrace:0x40087943:0x3ffbe630 0x40087139:0x3ffbe650 0x4009572a:0x3ffbe670 0x401461a3:0x3ffbe6e0 0x400e52c5:0x3ffbe710 0x400e5a33:0x3ffbe730 0x400e63e9:0x3ffbe760 0x400eba26:0x3ffbe790 0x400f141a:0x3ffbe7b0 0x400df81d:0x3ffbe7d0 0x400df89c:0x3ffbe7f0 0x400925a5:0x3ffbe820
0x40087943: panic_abort at E:/esp-idf/components/esp_system/panic.c:330

0x40087139: esp_system_abort at E:/esp-idf/components/esp_system/system_api.c:68

0x4009572a: abort at E:/esp-idf/components/newlib/abort.c:46

0x401461a3: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x400e52c5: tcp_receive at E:/esp-idf/components/lwip/lwip/src/core/tcp_in.c:1580 (discriminator 5)

0x400e5a33: tcp_process at E:/esp-idf/components/lwip/lwip/src/core/tcp_in.c:983

0x400e63e9: tcp_input at E:/esp-idf/components/lwip/lwip/src/core/tcp_in.c:438

0x400eba26: ip4_input at E:/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:760

0x400f141a: ethernet_input at E:/esp-idf/components/lwip/lwip/src/netif/ethernet.c:186

0x400df81d: tcpip_thread_handle_msg at E:/esp-idf/components/lwip/lwip/src/api/tcpip.c:180

0x400df89c: tcpip_thread at E:/esp-idf/components/lwip/lwip/src/api/tcpip.c:154

0x400925a5: vPortTaskWrapper at E:/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: d8a25290fec5cff2

Rebooting...
```

ESP-Marius, 2020-04-13:
Hi @xiruilin,

This is probably related to this [issue](https://github.com/espressif/esp-idf/issues/4983) which has been fixed on master.

Could you try pulling the latest changes and then try again?

xiruilin, 2020-04-13:
@ESP-Marius 
Thank you for your answer.
I update to latest commit 2e14149, the error still exists.

ESP-Marius, 2020-04-14:
@xiruilin Hmm, I haven't been able to reproduce the error. Did you change anything else in the example/menuconfig?

xiruilin, 2020-04-15:
@ESP-Marius 
If set ``CONFIG_LWIP_TCP_QUEUE_OOSEQ=n`` in the ``menuconfig``,
this error will not appear again

ESP-Marius, 2020-04-15:
Did you check how much heap space you have available while your software is running? Could be that it fails here due to you running out of heap space. You can use `esp_get_free_heap_size()` to check.

xiruilin, 2020-04-15:
@ESP-Marius 
Add ``esp_get_free_heap_size()`` to get free heap: 225460

ESP-Marius, 2020-04-16:
Try increasing the window size (`LWIP_TCP_WND_DEFAULT` in menuconfig) and see if that helps.

This problem doesn't seem to be related to IDF, might be an issue in lwip.

ESP-Marius, 2020-08-07:
@xiruilin Any updates? Did this issue resolve itself?

ESP-Marius, 2020-09-17:
Closing issue due to inactivity, feel free to reopen if this is still an ongoing problem.

closed 2020-09-17 by ESP-Marius