espressif/esp-idf #5111: idf.py menuconfig failed "cmake failed with exit code 3221225725"

javfernandez, 2020-04-13:
Hi everybody I have a problem while executing the idf.py menuconfig:
Operation system:Windows 10.
Python Version: Python  3.7.2
The command is: >idf.py --port COM10 -C ./hello_world -v menuconfig
And I get 
Checking Python dependencies...
Python requirements from C:\Users\USUARIO\Desktop\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory c:\users\usuario\desktop\esp-idf\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\users\usuario\desktop\esp-idf\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v4.0-dirty
-- Building ESP-IDF components for target esp32
cmake failed with exit code 3221225725

What Im doing wrong
Thanks

Alvin1Zhang, 2020-04-14:
@javfernandez Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, and etc. would help us debug further. Thanks.

renzbagaporo, 2020-04-14:
Hi @javfernandez, I cannot seem to reproduce this issue on my Windows machine. What version of CMake do you have?

Also, can you test executing the command on the actual example directory (minus `-C ./hello_world ` of course)

dobairoland, 2020-04-14:
Neither could I reproduce. Even the `-C` option works for me as it should be in the v4.0 branch.

This is suspicious:
> Project version: v4.0-dirty

Please examine the output of `git status`, clean uncommitted changes and update the submodules.

javfernandez, 2020-04-14:
I executed from de example directory as @renzbagaporo suggested. I get the following output:
C:\Users\USUARIO\Desktop\esp-idf\hello_world>idf.py --port COM10 -v menuconfig
Checking Python dependencies...
Python requirements from C:\Users\USUARIO\Desktop\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory c:\users\usuario\desktop\esp-idf\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\users\usuario\desktop\esp-idf\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v4.0-dirty
-- Building ESP-IDF components for target esp32
Traceback (most recent call last):
  File "C:/Users/USUARIO/Desktop/esp-idf/tools/kconfig_new/confgen.py", line 571, in <module>
    main()
  File "C:/Users/USUARIO/Desktop/esp-idf/tools/kconfig_new/confgen.py", line 282, in main
    output_function(deprecated_options, config, temp_file)
  File "C:/Users/USUARIO/Desktop/esp-idf/tools/kconfig_new/confgen.py", line 298, in write_config
    deprecated_options.append_config(config, filename)
  File "C:/Users/USUARIO/Desktop/esp-idf/tools/kconfig_new/confgen.py", line 156, in append_config
    with open(path_output, 'a') as f_o:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\USUARIO\\AppData\\Local\\Temp\\confgen_tmp5033pqq0'
CMake Error at C:/Users/USUARIO/Desktop/esp-idf/tools/cmake/kconfig.cmake:208 (message):
  Failed to run confgen.py
  (C:\Users\USUARIO\.espressif\python_env\idf4.0_py3.7_env\Scripts\python.exe;C:/Users/USUARIO/Desktop/esp-idf/tools/kconfig_new/confgen.py;--kconfig;C:/Users/USUARIO/Desktop/esp-idf/Kconfig;--sdkconfig-rename;C:/Users/USUARIO/Desktop/esp-idf/sdkconfig.rename;--config;C:/Users/USUARIO/Desktop/esp-idf/hello_world/sdkconfig;--env-file;C:/Users/USUARIO/Desktop/esp-idf/hello_world/build/config.env).
  Error 1
Call Stack (most recent call first):
  C:/Users/USUARIO/Desktop/esp-idf/tools/cmake/build.cmake:441 (__kconfig_generate_config)
  C:/Users/USUARIO/Desktop/esp-idf/tools/cmake/project.cmake:337 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/USUARIO/Desktop/esp-idf/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

cd

renzbagaporo, 2020-04-20:
This seems to be a different error entirely @javfernandez. I'm inclined to be suspicious of the possible modifications as @dobairoland mentioned:

```
Project version: v4.0-dirty
```

I second his suggestion of:

```
Please examine the output of git status, clean uncommitted changes and update the submodules.
```

and trying again...

renzbagaporo, 2020-04-22:
Any updates @javfernandez ?

javfernandez, 2020-04-22:
Hi, Im gonna try what @renzbagaporo suggestedd. I see there is a lot of uncommited changes in the repository and I want to try to update the repository

javfernandez, 2020-04-24:
Well, I made it work, I had to dischard all changes and updates all submodules
Thanks for all!

renzbagaporo, 2020-04-26:
Thanks for the update @javfernandez. Please close the issue if this is no longer an issue. Thanks!

closed 2020-04-26 by javfernandez

rkashish75, 2022-09-23:
https://github.com/renzbagaporo how to do the following?
Please examine the output of git status, clean uncommitted changes and update the submodules.

rkashish75, 2022-09-23:
I get the following when i install esp-at

PS C:\Espressif\frameworks\esp-idf-v4.4\esp\esp-at> python build.py install
Ready to install ESP-IDF prerequisites..
Windows Installer Download has already installed all prerequisites.
Ready to install ESP-AT prerequisites..
Requirement already satisfied: setuptools>=21 in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from -r requirements.txt (line 4)) (51.3.3)
Requirement already satisfied: cryptography<35,>=2.1.4 in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from -r requirements.txt (line 9)) (3.4.8)
Requirement already satisfied: pyyaml in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from -r requirements.txt (line 10)) (6.0)
Requirement already satisfied: xlrd in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from -r requirements.txt (line 11)) (2.0.1)
Collecting pycryptodome==3.15.0
  Downloading pycryptodome-3.15.0-cp35-abi3-win_amd64.whl (1.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.9/1.9 MB 965.8 kB/s eta 0:00:00
Collecting idf-component-manager==1.1.4
  Downloading idf_component_manager-1.1.4-py3-none-any.whl (95 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 95.6/95.6 kB 1.4 MB/s eta 0:00:00
Requirement already satisfied: contextlib2>0.6.0 in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager==1.1.4->-r requirements.txt (line 13)) (21.6.0)
Requirement already satisfied: schema in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager==1.1.4->-r requirements.txt (line 13)) (0.7.5)
Requirement already satisfied: requests-toolbelt in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager==1.1.4->-r requirements.txt (line 13)) (0.9.1)
Requirement already satisfied: requests<3 in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager==1.1.4->-r requirements.txt (line 13)) (2.27.1)
Requirement already satisfied: future in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager==1.1.4->-r requirements.txt (line 13)) (0.18.2)
Requirement already satisfied: click>=8.0.0 in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager==1.1.4->-r requirements.txt (line 13)) (8.0.4)
Requirement already satisfied: colorama in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager==1.1.4->-r requirements.txt (line 13)) (0.4.4)
Requirement already satisfied: tqdm<5 in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager==1.1.4->-r requirements.txt (line 13)) (4.63.0)
Requirement already satisfied: six in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager==1.1.4->-r requirements.txt (line 13)) (1.16.0)
Requirement already satisfied: cffi>=1.12 in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from cryptography<35,>=2.1.4->-r requirements.txt (line 9)) (1.15.0)
Requirement already satisfied: pycparser in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from cffi>=1.12->cryptography<35,>=2.1.4->-r requirements.txt (line 9)) (2.21)
Requirement already satisfied: charset-normalizer~=2.0.0 in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from requests<3->idf-component-manager==1.1.4->-r requirements.txt (line 13)) (2.0.12)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from requests<3->idf-component-manager==1.1.4->-r requirements.txt (line 13)) (1.26.9)
Requirement already satisfied: idna<4,>=2.5 in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from requests<3->idf-component-manager==1.1.4->-r requirements.txt (line 13)) (3.3)
Requirement already satisfied: certifi>=2017.4.17 in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from requests<3->idf-component-manager==1.1.4->-r requirements.txt (line 13)) (2021.10.8)
Installing collected packages: pycryptodome, idf-component-manager
  Attempting uninstall: idf-component-manager
    Found existing installation: idf-component-manager 1.0.1
    Uninstalling idf-component-manager-1.0.1:
      Successfully uninstalled idf-component-manager-1.0.1
Successfully installed idf-component-manager-1.1.4 pycryptodome-3.15.0

[notice] A new release of pip available: 22.1.2 -> 22.2.2
[notice] To update, run: python.exe -m pip install --upgrade pip
Platform name:
1. PLATFORM_ESP32
2. PLATFORM_ESP32C3
3. PLATFORM_ESP32C2
choose(range[1,3]):1

Module name:
1. WROOM-32
2. WROVER-32
3. PICO-D4
4. SOLO-1
5. MINI-1 (description: ESP32-U4WDH chip inside)
6. ESP32-SDIO
7. ESP32-D2WD (description: 2MB flash, No OTA)
8. ESP32_QCLOUD (description: QCLOUD TX:17 RX:16)
choose(range[1,8]):1

Enable silence mode to remove some logs and reduce the firmware size?
0. No
1. Yes
choose(range[0,1]):1
Platform name:ESP32     Module name:WROOM-32    Silence:1
Cloning into submodule:"esp-idf" from "https://github.com/espressif/esp-idf.git" (This may take time)..
Cloning into 'esp-idf'...
remote: Enumerating objects: 378620, done.
remote: Counting objects: 100% (3897/3897), done.
remote: Compressing objects: 100% (1623/1623), done.
remote: Total 378620 (delta 2291), reused 3679 (delta 2130), pack-reused 374723
Receiving objects: 100% (378620/378620), 206.50 MiB | 1.76 MiB/s, done.
Resolving deltas: 100% (279073/279073), done.
[... 120 lines cut ...]
[... cut ...]
Requirement already satisfied: charset-normalizer~=2.0.0 in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from requests<3->idf-component-manager==1.1.4->-r requirements.txt (line 13)) (2.0.12)

[notice] A new release of pip available: 22.1.2 -> 22.2.2
[notice] To update, run: python.exe -m pip install --upgrade pip
Platform name:ESP32     Module name:WROOM-32    Silence:1
submodules check completed for updates.
Ready to set up ESP-IDF tools..
ERROR: This script was called from a virtual environment, can not create a virtual environment again
A fatal error occurred: set up ESP-IDF python-env failed

rkashish75, 2022-09-30:
@renzbagaporo Pls see if you can help with the issues I'm facing with esp32

dobairoland, 2022-09-30:
@rkashish75 You can report issues for ESP-AT at https://github.com/espressif/esp-at.