espressif/esp-idf #5112: no memory barrier before 8 or 16 bit loads

szmodz, 2020-04-13:
Xtensa ABI requires that at least one MEMW should be executed in between every load or
store to a volatile variable.

Looks like this is not done for 8 or 16 bit loads:

```
uint32_t load32(volatile uint32_t *s) {
    return *s;
}

uint16_t load16(volatile uint16_t *s) {
    return *s;
}

uint8_t load8(volatile uint8_t *s) {
    return *s;
}
```

> xtensa-esp32-elf-gcc -O2 -std=gnu11 -S test.c

```
	.align	4
	.global	load32
	.type	load32, @function
load32:
	entry	sp, 32
	memw
	l32i.n	a2, a2, 0
	retw.n
	.size	load32, .-load32
	.align	4
	.global	load16
	.type	load16, @function
load16:
	entry	sp, 32
	l16ui	a2, a2, 0
	retw.n
	.size	load16, .-load16
	.align	4
	.global	load8
	.type	load8, @function
load8:
	entry	sp, 32
	l8ui	a2, a2, 0
	retw.n
	.size	load8, .-load8
```

Note there's a memw before l32i, but not before l16ui or l8ui.

igrr, 2020-04-13:
Thanks for the report. We've also bumped into recently while investigating another issue.

(also cc @jcmvbkbc)

jcmvbkbc, 2020-04-13:
> Xtensa ABI requires that at least one MEMW should be executed in between every load or
store to a volatile variable.

I don't see this requirement in the ABI, only in the "Memory Access Ordering" chapter of the ISA book and memw/extw descriptions. Otherwise I can confirm this issue with any optimization levels above O0. At O0 memw is properly inserted before all volatile accesses.
This happens because with optimization loads from uint16_t and uint8_t are done with zero_extendhisi2 and zero_extendqisi2 patterns that ignore volatile property of the accessed memory. Proposed fix:

```diff --git a/gcc/config/xtensa/xtensa.md b/gcc/config/xtensa/xtensa.md
index 5b803d3cbe65..749fe477d562 100644
--- a/gcc/config/xtensa/xtensa.md
+++ b/gcc/config/xtensa/xtensa.md
@@ -538,7 +538,7 @@
   ""
   "@
    extui\t%0, %1, 0, 16
-   l16ui\t%0, %1"
+   %v1l16ui\t%0, %1"
   [(set_attr "type"    "arith,load")
    (set_attr "mode"    "SI")
    (set_attr "length"  "3,3")])
@@ -549,7 +549,7 @@
   ""
   "@
    extui\t%0, %1, 0, 8
-   l8ui\t%0, %1"
+   %v1l8ui\t%0, %1"
   [(set_attr "type"    "arith,load")
    (set_attr "mode"    "SI")
    (set_attr "length"  "3,3")])
@@ -575,7 +575,7 @@
   ""
   "@
    sext\t%0, %1, 15
-   l16si\t%0, %1"
+   %v1l16si\t%0, %1"
   [(set_attr "type"    "arith,load")
    (set_attr "mode"    "SI")
    (set_attr "length"  "3,3")])
```

I've given it a quick test and it seems to do the right thing. I'll run full regression and submit it.

szmodz, 2020-04-13:
@jcmvbkbc sorry for the inaccurate description.

szmodz, 2020-04-14:
seems to work, thanks.

jcmvbkbc, 2020-04-15:
Fixed in the gcc mainline:
https://gcc.gnu.org/git/gitweb.cgi?p=gcc.git;h=a288e202c5e50704968685fc2922d159335be2cb

closed 2020-06-12 by igrr