espressif/esp-idf #5113: Way of opting out of downloading ESP32S2 tools for the install scripts

KaeLL, 2020-04-13:
Title. ~750MiB of disk space is highly valuable to me, especially when they will not be used at all.

The default can install whatever you want, like it does now, no change there. But an binary parameter like  
`--no-esp32s2`  
or a list of targets like  
`--enable-targets=esp32,esp32s2,...`  
like GCC's would be ideal.

igrr, 2020-04-13:
This is actually pretty easy to implement for the installation part, and partially done (at least in the underlying idf_tools.py). However the `export` part is somewhat dumb at the moment and will complain if some tools aren't installed. Will fix that.

The good news is that in the next toolchain release the targets are going to get merged. Same is going to happen for the ULP binutils. So the "full" installation size is going to be somewhat reduced.

KaeLL, 2021-03-17:
@igrr any news on this feature?
```
/tools $ du -BM -d 1
21M     ./esp32s2ulp-elf
21M     ./esp32ulp-elf
275M    ./xtensa-esp32s3-elf
273M    ./xtensa-esp32-elf
550M    ./riscv32-esp-elf
274M    ./xtensa-esp32s2-elf
7M      ./openocd-esp32
1419M   .
```
This is not right.

```
$ idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - 1.24.0.123_64eb9ff-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.16.4 (recommended)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended)
```
Of the `recommended` above, only `xtensa-esp32-elf` and `esp32ulp-elf`, which amount to 294MiB, are useful to me, the rest (1125MiB) is just pointless waste of precious disk space.
The ideal way of solving this problem would be borrowing the same thing that already exists on `esptool.py`, in which I'd call `./install.sh --chip esp32` and it installs only vanilla esp32 tools (`xtensa-esp32-elf`, `esp32ulp-elf`, and `openocd-esp32`), and that's it.
An option to delete the `tools/dist` cache after the installation is a plus but not necessary.

closed 2021-07-25 by espressif-bot