espressif/esp-idf #5114: SD card SPI reader not working. Hardware or software issue? It works with Arduino Due though.

ashtorak, 2020-04-13:
## Environment

- Development Kit:    ESP32-DevKitC v4
- Module or chip used:  ESP32-WROOM-32
https://www.az-delivery.de/en/products/esp32-developmentboard?_pos=2&_sid=9b0935b0b&_ss=r
- PlatformIO in VSCode with latest commit [2e14149](https://github.com/espressif/esp-idf/commit/2e14149bff63fe211d6d30ff0cd84c90b5fface9)
- SPI reader: https://www.az-delivery.de/en/products/copy-of-spi-reader-micro-speicherkartenmodul-fur-arduino?_pos=2&_sid=636bd39db&_ss=r
- SD card: Verbatim microSDHC 16GB Premium

## Problem Description

Hi, I am new to ESP32. I wanted to migrate from Arduino Due. The SD card and reader are working fine there, but I just cannot get it to run here. I have tried different libs in ArduinoIDE and then tried the PlatformIO default ESP32 framework for az-delivery-devkit-v4 as well as the latest github commit, always using _sd_card_example_main.c_ from the examples folder with `#define USE_SPI_MODE` uncommented.
But I don't get the card to initialize and get similar outputs like below, where I used the latest commit [2e14149](https://github.com/espressif/esp-idf/commit/2e14149bff63fe211d6d30ff0cd84c90b5fface9). However, for this commit the IDE was complaining about some functions like this:

`esp_vfs_fat_sdcard_unmount(mount_point, card);`

It suggested to use _sdmmc_ instead of _sdcard_. From the function's comment I figured, it should work for SPI, so I just changed it. Not sure, if this fucked up something. 
The example from the commit in the PIO default install compiled out of the box as it should.

I also tried `host.max_freq_khz = SDMMC_FREQ_PROBING;` , but no luck.

[image]
The card reader is connected to separate 5 V supply.

## Terminal Output

0_drv:0x00,hj5
I (54) boot: SPI Flash Size :        4MB
I (58) boot: Partition Table:
I (62) boo       t: ## Label            Usage          Type ST Offset   Length
  0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 0       0 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: seg       ment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x09ab8 ( 39608) map
I (126) esp_image: segment 1: paddr=0x00019ae       0 vaddr=0x3ffb0000 size=0x0214c (  8524) load
I (130) esp_image: segment 2: paddr=0x0001bc34 vaddr=0x40080000 size=0x00400 (  1024) load      
I (133) esp_image: segment 3: paddr=0x0001c03c vaddr=0x40080400 size=0x03fd4 ( 16340) load      
I (149) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x1cb7c (117628) map       
I (193) esp_image: segment 5: paddr=0x0003cb9c vaddr=0x400843d4 size=0x06570 ( 25968) load      
I (211) boot: Loaded app from partition at offset 0       x10000
I (211) boot: Disabling RNG early entropy source...
I (211) cpu_start: Pro cpu up.         0m
I (215) cpu_start: Application information:
I        (220) cpu_start: Project name:     test
I (224) cpu_start: App version:      1
I (229) cpu_start: Compile time:     Apr 13 20       20 17:07:53
I (235) cpu_start: ELF file SHA256:  0566429c46eae0ac...
I (241) cpu_start: ESP-IDF:          GIT-NOTFOUND
I (246) cpu_start: Starting app cpu, entry point is 0       x400816c4
I (0) cpu_start: App cpu up.
I (257       ) heap_init: Initializing. RAM available for dynamic allocation:
I (263) heap_in       it: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (270) heap_init: At 3FFB31F0 len 0002CE10 (179 KiB): DRAM
I (276) heap_init: At 3FFE0440 len 00003AE0 (1       4 KiB): D/IRAM
I (282) heap_init: At 3FFE4350 len 0001BCB       0 (111 KiB): D/IRAM
I (289) heap_init: At 4008A944 len 000156BC (85 KiB): IRAM
I (295) cpu_start: Pro c       pu start user code
I (3       13) spi_flash: detected chip: generic
I (314) spi_flash: flash io: dio
I (314) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (322) example: Initializing SD card
I (322) example: Usin       g SPI peripheral
E (332) spi: SPI2 already claimed by spi master.
E (332) spi_master: spi_bus_initialize(231): host        already in use
E (342) example: Failed to initialize the card (ESP_ERR_INVALID_       STATE). Make sure SD card lines have pull-up resistors in place.

ginkgm, 2020-04-14:
Hi @ashtorak ,
I'm not sure why your compiler complains about `esp_vfs_fat_sdcard_unmount`, which is the suggested way to unregister a sd card. The `esp_vfs_fat_sdmmc_unmount` is deprecated.

Similar to this, `esp_vfs_fat_sdspi_mount` is the suggested way to attach an SD card to an SPI bus. The interface `esp_vfs_fat_sdmmc_mount` is deprecated (only deprecated for sdspi usage), which will also initialize the bus, forbidding other devices from using that bus. The bus is already initialized before in this example, so there is an error `E (332) spi_master: spi_bus_initialize(231): host already in use`.

I suggest you revert your changes and try again, it should work (I've just tested it). If there are any  error/warning, please post it here.

ashtorak, 2020-04-14:
I deleted the contents of the framework-espidf folder in Windows Explorer and just copied the github folders there. But it seems that this is not how it works with PIO :D 
Now, I went back to the PIO default espidf framework, where it gives the following output when using the example from platformio\platforms\espressif32\examples\espidf-storage-sdcard:

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7008
ho 0 tail 12 room 4
load:0x40078000,len:14076
load:0x40080400,len:4304
entry 0x400806e8
I (74) boot: Chip Revision:        1
I (75) boot_comm: chip revision: 1, min. bootloader chip        revision: 0
I (41) boot: ESP-IDF HE       AD-HASH-NOTFOUND 2nd stage bootloader
I (41) boot: compile time 15:17:06
I (41) boot: Enabling RNG early entropy source...
I (47) boot: S       PI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 4MB         0m
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I        (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 0       1 0000f000 00001000
I (85) boot:  2 factory          fact       ory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min.        application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x09a88 ( 39560) map
 [0;32mI (127) esp_image: segment 1: paddr=0x00019ab0 vaddr=0x3ffb0000 size=0x0214c (  8524) load
I (130) esp_image:        segment 2: paddr=0x0001bc04 vaddr=0x40080000 size=0x00400 (  1024) load
I (134) esp_image: segment 3: paddr=0x0001c00c vaddr=0x40080400 size=0x04004 ( 16388) load
I (149) esp_image:        segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x1cb18 (117528) map
[0;32mI (193) esp_image: segment 5: paddr=0x0003cb38 vaddr=0x40084404 size=0x06540 ( 25920) load
I (211) boot: Loaded app from partition at offset 0x10000
I (211) boot: Disabling RNG early entropy source...
I (212) cpu_start: Pro cpu        up.
I (215) cpu_start: Application information:
    32mI (220) cpu_start: Project name:     test
I (225) cpu_start: App version:      1
I (229) cpu_start: Compile time       :     Apr 14 2020 15:16:31
I (235) cpu_start: ELF file SHA256:  d1b2c0abfc508d31...
I (241) cpu_start:        ESP-IDF:          HEAD-HASH-NOTFOUND
I (247) cpu_start:        Starting app cpu, entry point is 0x400816c4
I (0) cpu_start: App cpu up.
I (258) heap_init: Initializing. RAM avail       able for dynamic allocation:
I (265) heap_init: At 3FFAE6       E0 len 00001920 (6 KiB): DRAM
I (271) heap_init: At 3FFB31F0 len 0002CE10 (179 KiB): DRAM
I (277) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRA       M
I (283) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (290) heap_init: At 4008A       944 len 000156BC (85 KiB): IRAM
I (296) cpu_start: Pro cp       u start user code
I (314) spi_flash: detected chip: generic
I (315) s       pi_flash: flash io: dio
I (315) cpu_start: Starting sched       uler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (323) example: Initializing SD card
I (323) example: Using SPI peripheral
I (333)        gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (363) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (363) gpio: GPIO[2]|        InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (363) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0      
E (373) example:        Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up 
resistors in place.

ashtorak, 2020-04-14:
I think, it is actually some hardware issue. I had some problems with restart looping when connecting the external power supply only without USB power. Then I tried a stronger power supply and suddenly it started to smell burned. Probably from the card reader as it was quite hot when touching. There must be an issue with the wiring. But shouldn't it be correct as on the picture above? I will continue looking into these hardware issues tomorrow. First I have to check if the reader is still ok now.

ashtorak, 2020-04-15:
Finally, I tried another SD card and it seems to work with a SanDisk Ultra 32 GB SDHC. Any idea why it won't work with the Verbatim? I got it from amazon: https://www.amazon.de/gp/product/B0054KHY8C

Just some additional info, if anybody is wondering: I noticed that the reader uses 3.3 V on the Arduino Due, so now I connected it also to the board power with the ESP. Turns out, that it doesn't need extra 5 V.
The SanDisk card works with the example code at 3.3 V. The Verbatim doesn't. It shouldn't depend on voltage anyway.
I always used `host.max_freq_khz = SDMMC_FREQ_PROBING;`

igrr, 2020-04-15:
@ashtorak CRC error reported in the above log (0x109) typically indicates either a signal integrity issue or incorrect voltage levels. Usually it is easier to debug if you can capture the exchange between the host and the card using a logic analyzer.

ashtorak, 2020-04-15:
Yeah. That might have been due to the external power supply on the reader. Now I only use the board power via USB and now I only get this message on the verbatim card:
`sdmmc_cmd: sdmmc_card_init: send_op_cond (1) returned 0x107`
While it works with the SanDisk card.

ashtorak, 2020-06-11:
After getting into some issues with the SanDisk card as well, I finally connected the power of the card reader to the 5 V pin of the Dev board, and it seems to work now with both cards. I am not sure why I didn't try this earlier. Probably because I measured the 5 V pin and it showed some strange voltage.

closed 2020-06-11 by ashtorak

sasukiuchiwa, 2022-04-14:
I (334) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (344) example: Initializing SD card
I (344) example: Using SPI peripheral
I (354) example: Mounting filesystem
I (354) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (404) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
E (404) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).
I (404) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (414) example: Failed to initialize the card (ESP_ERR_INVALID_RESPONSE). Make sure SD card lines have pull-up resistors in place.
still can't get this to work i used the storage sdspi example with pullups

adirsingh96, 2023-02-16:
I solved this issue by changing max frequency to 1 MHz using  host.max_freq_khz = 1000,  I used no extra pullups and supplied 3.3V to the sd card adapter.