espressif/esp-idf #5116: spi_flash interrupt disable breaks task priority inheritence

jefftharris, 2020-04-13:
## Environment

- Development Kit:      Custom board
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.0
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 3.3V

## Problem Description

If spi_flash_disable_interrupts_caches_and_other_cpu is called from a task which currently has an inherited priority from a mutex, the task loses its base priority.

For example, if I have a task with base priority 1 and inherited priority 20 which calls spi_flash_disable_interrupts_caches_and_other_cpu, the task gets elevated to 24 in the function.  However, when the priority is reset and the function returns, the task finishes with a base priority of 20 instead of 1.

Looking at vTaskPriorityInherit, the base priority is always set.  There doesn't look to be a method to preserve the base priority to restore.  

I did a quick update to add a vTaskPrioritiesSet to provide both the base and current priority.  The problem seems resolved, but is there a better method than enhancing FreeRTOS?  Or, is there another method to solve the needed raised priority in the function?

### Expected Behavior

The task would exit the function with the same base and inherited priorities

### Actual Behavior

The base priority is set to the current (inherited) priority.

Alvin1Zhang, 2020-04-14:
@jefftharris Thanks for reporting, we will look into.

jefftharris, 2020-04-14:
For reference, I've attached the patch which seems to be working.  However, I am largely unfamiliar with FreeRTOS task scheduling and have not done extensive testing.

[task-diff.txt](https://github.com/espressif/esp-idf/files/4475649/task-diff.txt)

mythbuster5, 2021-01-28:
Flash chip erasing is always the highest periority task on ESP32.

tom-borcin, 2021-05-07:
Hi @jefftharris,

we think the issue is resolved now and therefore we are closing it to keep our backlog manageable. If you don't agree with this decision, please let me know.

Thank you,

Tomas

pgrisu, 2022-07-23:
Hi @tom-borcin, can you say which commit fixed this issue?
I'm facing the same issue and checking latest idf source code it seems that no changes are included to solve this issue.
spi_flash_disable_interrupts_caches_and_other_cpu() and involved OS functions like uxTaskPriorityGet and vTaskPrioritySet are basically unchanged regarding this issue

Dazza0, 2022-07-25:
Sorry, this was closed by mistake.

AxelLin, 2022-07-25:
https://github.com/espressif/esp-idf/issues/7580

KonstantinKondrashov, 2022-09-05:
It is fixed on master aea2fe08162e1ba86ad0189178c1f1cf29238bbd. We will also backport it.

Alvin1Zhang, 2022-11-16:
* Fix on release/4.3 is at https://github.com/espressif/esp-idf/commit/96ef8f42be3691e14b548f81632bb4b3f61e7c16
* Fix on release/4.4 is at https://github.com/espressif/esp-idf/commit/d805d2ad5a4e571632a89ced76a8330368aec68f
* Fix on release/5.0 is at https://github.com/espressif/esp-idf/commit/3f9b12d62a06db1df2951b87e899a7750197d765.

Feel free to reopen.

closed 2022-11-16 by Alvin1Zhang