espressif/esp-idf #5117: esp_http_client keep-alive fails after POST request

tensor1-0, 2020-04-13:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.2-dev-1054-g7d0f1536e
- Build System: idf.py
- Compiler version: 8.2.0
- Operating System: Linux
- IDE: CLion, but compiling from shell with idf.py
- Power Supply: USB
- External peripheral: clone of Waveshare LAN8720 ETH board

## Problem Description

esp_http_client can not hold HTTPS connection alive after sending POST request to server. Next request will fail to read headers (ESP_ERR_HTTP_FETCH_HEADER). 

When I use only GET request, then everything is fully functional.

I'm using mbedtls with server and client side certificates. Both are RSA 2048, the server certificate is self-signed, the client certificate is issued by my own CA.

ESP is connected through simple D-Link Ethernet switch with server. Server is written in Rust with framework actix-web and use OpenSSL for TLS. The server has enabled keep-alive, which I tested with CURL.

I would use some server from internet like httpbin.org but I do not know some with same keep-alive settings as mine server.

### Expected Behavior

ESP send POST requests to server without error and without opening new connection, because the underlying socket is still open from previous request.

### Actual Behavior

After every POST request next request (in example GET) fails to read headers (ESP_ERR_HTTP_FETCH_HEADER). 

### Steps to reproduce

1. Run included code

Wiring is same as Frank Sautter's in this blog post: https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

```
GPIO17 - PHY_POWER   : NC - Osc. Enable - 4k7 Pulldown
GPIO22 - EMAC_TXD1   : TX1
GPIO19 - EMAC_TXD0   : TX0
GPIO21 - EMAC_TX_EN  : TX_EN
GPIO26 - EMAC_RXD1   : RX1
GPIO25 - EMAC_RXD0   : RX0
GPIO27 - EMAC_RX_DV  : CRS
GPIO00 - EMAC_TX_CLK : nINT/REFCLK (50MHz) - 4k7 Pullup
GPIO23 - SMI_MDC     : MDC
GPIO18 - SMI_MDIO    : MDIO
GND                  : GND
3V3                  : VCC

```

### Code to reproduce this issue

https://gist.github.com/tensor1-0/4a4087abfe0a8ce6a953980f69bca2e4

## Debug Logs

```
D (149947) HTTP_CLIENT: Begin connect to: https://192.168.152.1:8443
D (149947) esp-tls: host:192.168.152.1: strlen 13
D (149957) emac_esp32: receive len= 60
D (150037) esp-tls: handshake in progress...
D (150037) emac_esp32: receive len= 60
D (150047) emac_esp32: receive len= 1494
D (150047) emac_esp32: receive len= 658
D (150157) emac_esp32: receive len= 60
D (150597) emac_esp32: receive len= 60
D (151097) emac_esp32: receive len= 60
D (151097) emac_esp32: receive len= 60
D (151097) emac_esp32: receive len= 60
D (151097) emac_esp32: receive len= 1144
D (151107) HTTP_CLIENT: Write header[3]: GET / HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: 192.168.152.1
Content-Length: 0

D (151117) emac_esp32: receive len= 176
D (151117) HTTP_CLIENT: on_message_begin
D (151117) HTTP_CLIENT: HEADER=content-length:17
D (151127) HTTP_CLIENT: HEADER=date:Mon, 13 Apr 2020 18:36:04 GMT
D (151127) HTTP_CLIENT: http_on_headers_complete, status=200, offset=76, nread=76
D (151137) HTTP_CLIENT: http_on_body 17
D (151137) HTTP_CLIENT: http_on_message_complete, parser=3ffb6064
D (151147) HTTP_CLIENT: content_length = 17
I (151157) esp32-post: Connected to server: https://192.168.152.1:8443/
I (166157) esp32-post: Prepare POST to: https://192.168.152.1:8443/post
D (166157) HTTP_CLIENT: set post file length = 27
D (166157) HTTP_CLIENT: Write header[4]: POST /post HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: 192.168.152.1
Content-Length: 27
Content-Type: application/x-www-form-urlencoded

D (166177) emac_esp32: receive len= 176
D (166177) HTTP_CLIENT: on_message_begin
D (166177) HTTP_CLIENT: HEADER=content-length:17
D (166187) HTTP_CLIENT: HEADER=date:Mon, 13 Apr 2020 18:36:19 GMT
D (166187) HTTP_CLIENT: http_on_headers_complete, status=200, offset=76, nread=76
D (166197) HTTP_CLIENT: http_on_body 17
D (166197) HTTP_CLIENT: http_on_message_complete, parser=3ffb6064
D (166207) HTTP_CLIENT: content_length = 17
D (166207) HTTP_CLIENT: is_data_remain=0, is_chunked=0, content_length=17
D (166217) HTTP_CLIENT: is_data_remain=0, is_chunked=0, content_length=17
D (166217) emac_esp32: receive len= 60
I (166227) esp32-post: Response to POST from: https://192.168.152.1:8443/post
Body: '{"response":"OK"}'
I (166237) esp32-post: Prepare GET to: https://192.168.152.1:8443/
D (166247) HTTP_CLIENT: Write header[4]: GET / HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: 192.168.152.1
Content-Length: 27
Content-Type: application/x-www-form-urlencoded

D (166267) emac_esp32: receive len= 60
D (166267) emac_esp32: receive len= 60
E (171267) esp32-post: HTTP client GET error: ESP_ERR_HTTP_FETCH_HEADER
D (171267) emac_esp32: receive len= 85
D (171267) emac_esp32: receive len= 60
D (171267) emac_esp32: receive len= 0
D (171277) HTTP_CLIENT: Begin connect to: https://192.168.152.1:8443
D (171277) esp-tls: host:192.168.152.1: strlen 13
D (171287) emac_esp32: receive len= 60
D (171367) esp-tls: handshake in progress...
D (171367) emac_esp32: receive len= 60
D (171377) emac_esp32: receive len= 1494
D (171377) emac_esp32: receive len= 658
D (171487) emac_esp32: receive len= 60
D (171937) emac_esp32: receive len= 60
D (172427) emac_esp32: receive len= 60
D (172427) emac_esp32: receive len= 60
D (172437) emac_esp32: receive len= 60
D (172437) emac_esp32: receive len= 1144
D (172437) HTTP_CLIENT: Write header[3]: GET / HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: 192.168.152.1
Content-Length: 0

D (172447) emac_esp32: receive len= 176
D (172457) HTTP_CLIENT: on_message_begin
D (172457) HTTP_CLIENT: HEADER=content-length:17
D (172457) HTTP_CLIENT: HEADER=date:Mon, 13 Apr 2020 18:36:26 GMT
D (172467) HTTP_CLIENT: http_on_headers_complete, status=200, offset=76, nread=76
D (172477) HTTP_CLIENT: http_on_body 17
D (172477) HTTP_CLIENT: http_on_message_complete, parser=3ffb6064
D (172487) HTTP_CLIENT: content_length = 17
I (172487) esp32-post: Connected to server: https://192.168.152.1:8443/

```

Complete log include boot: [esp_http_client.log](https://github.com/espressif/esp-idf/files/4471640/esp_http_client.log)

Log without connection reset after error: [esp_http_client_no_recovery.log](https://github.com/espressif/esp-idf/files/4471706/esp_http_client_no_recovery.log)

## Other items if possible

[sdkconfig.tar.gz](https://github.com/espressif/esp-idf/files/4471583/sdkconfig.tar.gz)
[esp32-fw.elf.tar.gz](https://github.com/espressif/esp-idf/files/4471581/esp32-fw.elf.tar.gz)

Alvin1Zhang, 2020-04-14:
@tensor1-0 Thanks for reporting, we will look into.

shubhamkulkarni97, 2020-04-27:
@tensor1-0 Do you face this issue when connecting with WiFi as well?

tensor1-0, 2020-04-28:
> @tensor1-0 Do you face this issue when connecting with WiFi as well?

I didn't use WiFi in original project, so I never tested it.

shubhamkulkarni97, 2020-05-07:
@tensor1-0 , I was not able to reproduce your issue using WiFi. Can you please check I you face same issue when using WiFi?

tensor1-0, 2020-05-08:
> @tensor1-0 , I was not able to reproduce your issue using WiFi. Can you please check I you face same issue when using WiFi?

I can test it on WiFi, but now I need to handle some deadlines.

shubhamkulkarni97, 2020-05-26:
@tensor1-0 , Do you have any updates on the issue?

tensor1-0, 2020-06-01:
> @tensor1-0 , Do you have any updates on the issue?

I'm sorry, but I have still other workload to handle. I can look into it after mid June.

Alvin1Zhang, 2020-07-20:
@tensor1-0 Thanks for reporting, would you please help share if any updates for this issue? Thanks.

mahavirj, 2020-07-27:
Closing due to lack of response, please feel free to re-open if there are any updates.

closed 2020-07-27 by mahavirj