espressif/esp-idf #5118: 华为荣耀20 Android 10.0版本配不了网，必现

linux-liu, 2020-04-14:
手机一直发送udp包，但设备端一直收不到数据，设备端联网一直不成功。手机端超时配网失败

Alvin1Zhang, 2020-04-14:
@linux-liu Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further.

Alvin1Zhang, 2020-04-16:
Thanks for reporting @linux-liu . Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further.

Alvin1Zhang, 2020-07-02:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details, thanks.

closed 2020-07-02 by Alvin1Zhang