espressif/esp-idf #5119: To add LCP_ECHOINTERVAL/LCP_MAXECHOFAILS configuration items

AxelLin, 2020-04-14:
Hi @projectgus 

Base on the discussion on this thread:
https://www.esp32.com/viewtopic.php?f=13&t=13756

Please add below configuration items to PPP settings via menuconfig

LCP_ECHOINTERVAL
LCP_MAXECHOFAILS

Thanks,
Axel

david-cermak, 2020-04-20:
@AxelLin 
Thanks for posting this feature request, we will look into it, but this may take some time, since we'd like to test at least on general level

Meantime, might I suggest enabling this options simply on project level?

The below code in your project CMake file should apply the define to the lwip component
```
idf_component_get_property(lwip lwip COMPONENT_LIB)

target_compile_definitions(${lwip} PRIVATE "-DLCP_ECHOINTERVAL=XXX")
```

AxelLin, 2020-04-21:
> Meantime, might I suggest enabling this options simply on project level?
> 
> The below code in your project CMake file should apply the define to the lwip component
> 
> ```
> idf_component_get_property(lwip lwip COMPONENT_LIB)
> 
> target_compile_definitions(${lwip} PRIVATE "-DLCP_ECHOINTERVAL=XXX")
> ```
Hi @david-cermak 
Above change works.
Thanks.

projectgus, 2020-04-22:
@AxelLin Glad that the workaround is functioning, but we can leave this request open until we actually have this feature in ESP-IDF.

JosueSanchezEZM, 2021-02-18:
This workaround does not work for me: 
[image]

Makefile:
[image]

JosueSanchezEZM, 2021-02-25:
Is this functionality already available in esp-idf 4.3/release ?

closed 2021-03-29 by AxelLin