espressif/esp-idf #5120: WiFi never reconnecting after esp_restart()

vtunr, 2020-04-14:
## Environment

- Development Kit:      [Custom PCB]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
     `v4.0-2-g943a9b780 (Slightly modified of v4.0 to work with our code`
- Build System:         [CMake & idf.py]
- Compiler version:
    `xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0`
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt & regular CMD].
- Using an IDE?: [No]
- Power Supply:         [external 5V]

## Problem Description

I let the device chain reboot every 30-40s (using esp_restart()), and after some time (a few hours), it will never manage to reconnect to WiFi.
If i do a software reset, it's still not connecting.
I need to press on RST button to be able to have WiFi working again.
I tried on a second AP from another router, and I have the same issue.
Ethernet is enable, but no cable is plugged.

### Expected Behavior

Device can always reconnect after esp_restart.

### Steps to reproduce

1. Use `esp_restart()` every 40s if you have a working connection. 
2. Wait for the moment it will never reconnect
3. It will not reboot and you'll be stuck on WiFi not connecting

### Code to reproduce this issue

Here's the [gist](https://gist.github.com/vtunr/38d1f3a19d813c0da1d29998d860f776) of my handlers.

## Debug Logs

Find attached the WiFi logs : 
[wifi_issue.txt](https://github.com/espressif/esp-idf/files/4475342/wifi_issue.txt)

Alvin1Zhang, 2020-04-15:
@vtunr Thanks for reporting, we will look into.

liuzfesp, 2020-04-15:
Hi @vtunr, the WiFi can't reconnect the AP because it fails to scan the AP.

Could you help to do following test:

- When problem happens, use your phone to connect the AP and check whether the AP is still alive or not.
- Do the same test in WiFi only mode, maybe you can test it in WiFi example and call `esp_restart()` after WiFi successfully connected to AP.

I will try to reproduce the issue in my size and do more analysis.

vtunr, 2020-04-15:
Hi @liuzfesp, 

So : 
* For BLE, I have one service, with three characteristics, and advertising is on. No devices are connected to the ESP through BLE
* My computer is connected to the same AP, and I have no issue. 
* I will flash WiFi example with a restart, and i'll get back to you if I manage to reproduce the issue.

For the demo code, it's going to be hard to share, but if I can't reproduce with WiFi example, I'll try to find minimal code that can reproduce the issue so you can have something to work with

vtunr, 2020-04-17:
@liuzfesp I couldn't reproduce with only WiFi example. I'll try enabling other things until I can reproduce.

liuzfesp, 2020-04-19:
Hi @vtunr yes, I also can't replicate this issue with WiFi only mode.

liuzfesp, 2020-04-30:
Hi @vtunr, any update about the minimal code that can reproduce this issue?

vtunr, 2020-04-30:
Hi @liuzfesp, so I tried to reproduce it with disabling more and more component, and : 
* Disabling ETH doesn't change anything.
* Disabling BLE seems to remove the problem.

I'm still trying to have minimal code to reproduce the issue.

liuzfesp, 2020-06-04:
Hi @vtunr any update about this issue?

vtunr, 2020-06-04:
Hi @liuzfesp, I updated to SDK 4.0.1, I will run some test

Alvin1Zhang, 2020-07-06:
@vtunr Thanks for reporting, would you please help share if any further updates for this issue? Thanks.

Alvin1Zhang, 2020-10-19:
Thanks for reporting, will close due to short of updates, feel free to reopen if the issue still happens or more details. Thanks.

closed 2020-10-19 by Alvin1Zhang