espressif/esp-idf #5121: ESP SoftAP Prov HTTP request sample

25prithvi, 2020-04-14:
Can someone provide a sample HTTP request to set SSID and password via Postman software or chrome for ESP SoftAP Prov example?

Note: I don't want to use ESP SoftAP Prov Android App

shahpiyushv, 2020-04-15:
@25prithvi , the provisioning HTTP handlers use protobuf for data serialisation and so, it isn't that straight forward to be used from Postman. Please check the python script at https://github.com/espressif/esp-idf/tree/master/tools/esp_prov if Android app isn't an option for you.

Alvin1Zhang, 2020-05-12:
@25prithvi Thanks for reporting, feel free to reopen if you still have issues/questions, thanks.

closed 2020-05-12 by Alvin1Zhang

Schuck84, 2020-06-18:
Hi,
I also would like to know if there is a option to configure the WiFi credentials via a "normal" http web interface.
Could you give me some information how to proceed regarding this?
Regards