espressif/esp-idf #5122: deadlock after a wifi shutdown (esp_timer, btuT, hciT, BTC_task blocked)

pbruynbroeckQuimesis, 2020-04-14:
## Environment

- Development Kit: [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [none]
- Module or chip used: [ESP32-WROVER-B]
- IDF version (run git describe --tags to find it): v4.0-rc
- Build System: [Make]
- Compiler version (run xtensa-esp32-elf-gcc --version to find it): 8.2.0
- Operating System: [Linux]
- Power Supply: [external 5V]

## Problem Description

I have an esp32 using wifi and bluetooth:
- A lot of datas are received via the BLE.
- Then, these data are send to a server with mqtt.

All works fine, but, when I shutdown the wifi, some task on the esp are blocked like: wifi, esp_timer, btuT,  hciT, BTC_task,... and my "periodic_task".

(The periodic_task does differents things including deactivating the device connected to bluetooth when it no longer sends data. However, the device is supposed to continue sending data ...)

It seems to be a deadlock.

(The `periodic_task` and `btuT` are waiting on the same queue `0x3ffb5a6c`).

## More info

I have observed that the problem disappear if I disable the call of `esp_ble_gap_disconnect` in my `periodic_task` and the `btc_transfer_context` in `alarm.c`:

```
static void alarm_cb_handler(struct alarm_t *alarm)
{
    OSI_TRACE_DEBUG("TimerID %p\n", alarm);
    if (alarm_state != ALARM_STATE_OPEN) {
        OSI_TRACE_WARNING("%s, invalid state %d\n", __func__, alarm_state);
        return;
    }
    btc_msg_t msg;
    btc_alarm_args_t arg;
    msg.sig = BTC_SIG_API_CALL;
    msg.pid = BTC_PID_ALARM;
    arg.cb = alarm->cb;
    arg.cb_data = alarm->cb_data;
    btc_transfer_context(&msg, &arg, sizeof(btc_alarm_args_t), NULL);
}
```

## Back trace

**All thread :**
```
(gdb) i thread
  Id   Target Id         Frame 
  1    Thread 1073456008 (Name: main) 0x400982a8 in xQueueGenericReceive (xQueue=0x3f8057a0, pvBuffer=0x0, xTicksToWait=<optimized out>, 
    xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  2    Thread 1073459832 (Name: IDLE1, State: Running @CPU1) 0x401f5116 in esp_pm_impl_waiti ()
    at /home/pierre/my_project/common/esp-idf/components/esp32/pm_esp32.c:484
* 3    Thread 1073457920 (Name: IDLE0, State: Running @CPU0) 0x401f5116 in esp_pm_impl_waiti ()
    at /home/pierre/my_project/common/esp-idf/components/esp32/pm_esp32.c:484
  4    Thread 1073464304 (Name: Tmr Svc) 0x4009ae6b in prvProcessTimerOrBlockTask (xListWasEmpty=<optimized out>, xNextExpireTime=15908)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  5    Thread 1073446524 (Name: ipc1) 0x400982a8 in xQueueGenericReceive (xQueue=0x3ffb7e1c, pvBuffer=0x0, xTicksToWait=<optimized out>, 
    xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  6    Thread 1073444044 (Name: esp_timer) 0x400982a8 in xQueueGenericReceive (xQueue=0x3ffc5e38 <s_timer_semaphore_memory>, pvBuffer=0x0, 
    xTicksToWait=<optimized out>, xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  7    Thread 1073446056 (Name: ipc0) 0x400982a8 in xQueueGenericReceive (xQueue=0x3ffb7844, pvBuffer=0x0, xTicksToWait=<optimized out>, 
    xJustPeeking=0)
[... 158 lines cut ...]
    at /home/pierre/my_project//gabi-smart-care-hub/common/esp-idf/components/freertos/include/freertos/portable.h:214
214	    return id;
(gdb) bt
#0  0x400982a8 in xQueueGenericReceive (xQueue=0x3ffe37b8, pvBuffer=0x0, xTicksToWait=<optimized out>, xJustPeeking=0)
    at /home/pierre/my_project//gabi-smart-care-hub/common/esp-idf/components/freertos/include/freertos/portable.h:214
#1  0x4013c9fc in osi_sem_take (sem=0x3ffe372c, timeout=4294967295)
    at /home/pierre/my_project//gabi-smart-care-hub/common/esp-idf/components/bt/common/osi/semaphore.c:60
#2  0x40161c35 in fixed_queue_enqueue (queue=0x3ffe3728, data=0x3fff8600, timeout=4294967295)
    at /home/pierre/my_project//gabi-smart-care-hub/common/esp-idf/components/bt/common/osi/fixed_queue.c:135
#3  0x40163a8c in osi_thread_post (thread=0x3ffe370c, func=0x4016ee98 <bta_sys_event>, context=0x3f83a2e4, queue_idx=<optimized out>, 
    timeout=4294967295) at /home/pierre/my_project//gabi-smart-care-hub/common/esp-idf/components/bt/common/osi/thread.c:246
#4  0x4014e1ca in btu_task_post (sig=<optimized out>, param=0x3f83a2e4, timeout=4294967295)
    at /home/pierre/my_project//gabi-smart-care-hub/common/esp-idf/components/bt/host/bluedroid/stack/btu/btu_task.c:225
#5  0x4016ef6c in bta_sys_sendmsg (p_msg=0x3f83a2e4)
    at /home/pierre/my_project//gabi-smart-care-hub/common/esp-idf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:577
#6  0x401688bc in BTA_GATTC_SendIndConfirm (conn_id=3, handle=28)
    at /home/pierre/my_project//gabi-smart-care-hub/common/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_api.c:802
#7  0x401414a9 in btc_gattc_cb_handler (msg=0x3f83a93c)
    at /home/pierre/my_project//gabi-smart-care-hub/common/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gattc.c:881
#8  0x4013cabc in btc_thread_handler (arg=0x3f83a93c)
    at /home/pierre/my_project//gabi-smart-care-hub/common/esp-idf/components/bt/common/btc/core/btc_task.c:149
#9  0x4016382a in osi_thread_run (arg=<optimized out>)
    at /home/pierre/my_project//gabi-smart-care-hub/common/esp-idf/components/bt/common/osi/thread.c:68
(gdb) 
```

liuzfesp, 2020-04-15:
Hi @pbruynbroeckQuimesis ,

- Do you mean calling `esp_wifi_stop()` when you said `shut down wifi`?
- Why you think it's deadlock, can the system be recovered after a while? Or the system can never be recovered unless reboot the device?
- About how long time can this issue be reproduced?

pbruynbroeckQuimesis, 2020-04-15:
Hi @liuzfesp,

> Do you mean calling esp_wifi_stop() when you said shut down wifi?

I mean, the esp is connected to a wifi hot-spot and I turn off my hot-spot. ( I do nothing on the esp ).

> Why you think it's deadlock, can the system be recovered after a while? Or the system can never be recovered unless reboot the device?

The system can never be recovered unless reboot the device. 
The tasks wifi, esp_timer, btuT, hciT, BTC_task, ... are "freeze". Their `ulRunTimeCounter` (obtained with a `vTaskGetRunTimeStats`) never change.

> About how long time can this issue be reproduced?

Each time I turn off my hot-spot.

liuzfesp, 2020-04-15:
Please @wmy-espressif take a look at this issue.

pbruynbroeckQuimesis, 2020-04-28:
Hello,
I did a test with [v4.1-beta1](https://github.com/espressif/esp-idf/releases/tag/v4.1-beta1) and the [latest version](https://github.com/espressif/esp-idf/commit/4a558131eaa4d81d122edd15d4288968e5846994) but the issue is still there.

mywang-espressif, 2020-04-29:
@pbruynbroeckQuimesis , 

Thanks for the detailed logs. Here is my analysis according to your information

Task btuT is blocked in function btc_task_post, 
Task BTC_task is blocked in function btu_task_post

There is a deadlock in this.

I guess that the cause is that the workqueue for the two tasks: btuT and BTC_task are full.

To check this, you can use the following function call to check the "water-level" of these workqueues:

    `osi_thread_queue_wait_size(btu_thread, 0)`
    `osi_thread_queue_wait_size(btc_thread, 0)`

Note that these functions are "visible" can only be called inside bluetooth host.

Solution.

The default workqueue size for bluetooth Host tasks is 100,
defined in const `DEFAULT_WORK_QUEUE_CAPACITY`, in file components/bt/common/osi/thread.c,

In the case that CPU loading is high, the bluetooth host task are likely to suffer: more entries  waiting to be processed in the workqueue of each task. You can enlarge DEFAULT_WORK_QUEUE_CAPACITY and see whether the issue can be reproduced.

By the way, can you tell me the execution frequency of your periodic_task? How frequently does it call bluetooth APIs like `esp_ble_gap_disconnect`?

pbruynbroeckQuimesis, 2020-04-29:
Thanks for you response :).

Enlarge the `DEFAULT_WORK_QUEUE_CAPACITY` solves my issue :). (It would be nice to be able to modify it in the sdkconfig).

The periodic_task does different jobs (with different timers) whose check the connected bluetooth device activity every 2 sec.

When the wifi hot-spot is stopped, I notice that the BLE activity is slowed down :(. When the period_task checks the activity of the device, it sees that there has been no BLE communication for a certain time and calls esp_ble_gap_disconnect ().

pbruynbroeckQuimesis, 2020-05-08:
@mywang-espressif,

I enlarged the DEFAULT_WORK_QUEUE_CAPACITY to 200 but I had frozen tasks again.

After operating for serverl hours/days, I restart the wifi hotpost. The ESP32 disconnect but never reconnect.
The wifi and main task seem to be frozen: 
- we have no more 'wifi' log after the disconnection,
- the CLI doens't respond (but I have logs from others tasks).

**All thread**
```
Thread 1 received signal SIGINT, Interrupt.
[Switching to Thread 1073457920]
0x401f44fe in esp_pm_impl_waiti ()
    at /home/pierre/my_project/common/esp-idf/components/esp32/pm_esp32.c:484
484	in /home/pierre/my_project/common/esp-idf/components/esp32/pm_esp32.c
(gdb) i thread
  Id   Target Id         Frame 
* 1    Thread 1073457920 (Name: IDLE0, State: Running @CPU0) 0x401f44fe in esp_pm_impl_waiti ()
    at /home/pierre/my_project/common/esp-idf/components/esp32/pm_esp32.c:484
  2    Thread 1073459832 (Name: IDLE1, State: Running @CPU1) 0x401f44fe in esp_pm_impl_waiti ()
    at /home/pierre/my_project/common/esp-idf/components/esp32/pm_esp32.c:484
  3    Thread 1073597836 (Name: band_tmp_data_t) 0x400982a8 in xQueueGenericReceive (xQueue=0x3f83b768, pvBuffer=0x3ffef130, 
    xTicksToWait=<optimized out>, xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  4    Thread 1073713324 (Name: mqtt_task) xEventGroupWaitBits (
    xEventGroup=0x3fff7390, uxBitsToWaitFor=2, xClearOnExit=0, 
    xWaitForAllBits=1, xTicksToWait=<optimized out>)
    at /home/pierre/my_project/common/esp-idf/components/freertos/event_groups.c:445
  5    Thread 1073464304 (Name: Tmr Svc) 0x4009ae6b in prvProcessTimerOrBlockTask (xListWasEmpty=<optimized out>, xNextExpireTime=17161905)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  6    Thread 1073541504 (Name: tiT) 0x400982a8 in xQueueGenericReceive (xQueue=0x3f831b98, pvBuffer=0x3ffcf0e0, 
    xTicksToWait=<optimized out>, xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  7    Thread 1073446524 (Name: ipc1) 0x400982a8 in xQueueGenericReceive (xQueue=0x3ffb7e1c, pvBuffer=0x0, xTicksToWait=<optimized out>, 
    xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  8    Thread 1073598580 (Name: update_manager) 0x4000bff0 in ?? ()
  9    Thread 1073598208 (Name: dl_task) 0x400982a8 in xQueueGenericReceive (xQueue=0x3f83f970, pvBuffer=0x3f83a684, 
    xTicksToWait=<optimized out>, xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  10   Thread 1073597464 (Name: music_play_task) 0x40099f86 in vTaskSuspend (xTaskToSuspend=<optimized out>)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  11   Thread 1073597092 (Name: i2s_read_task) 0x400819d0 in esp_crosscore_int_send_yield (core_id=1)
    at /home/pierre/my_project/common/esp-idf/components/esp32/crosscore_int.c:111
  12   Thread 1073595936 (Name: httpd) 0x400982a8 in xQueueGenericReceive (xQueue=0x3ffe2f50, pvBuffer=0x0, xTicksToWait=<optimized out>, 
    xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  13   Thread 1073446056 (Name: ipc0) 0x400982a8 in xQueueGenericReceive (xQueue=0x3ffb7844, pvBuffer=0x0, xTicksToWait=<optimized out>, 
    xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  14   Thread 1073435524 (Name: btController) 0x400982a8 in xQueueGenericReceive (xQueue=0x3ffb5318, pvBuffer=0x3ffe52d0, 
    xTicksToWait=<optimized out>, xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  15   Thread 1073444044 (Name: esp_timer) 0x400982a8 in xQueueGenericReceive (xQueue=0x3ffc5e38 <s_timer_semaphore_memory>, pvBuffer=0x0, 
    xTicksToWait=<optimized out>, xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  16   Thread 1073438080 (Name: BTC_TASK) 0x400982a8 in xQueueGenericReceive (xQueue=0x3ffb5c60, pvBuffer=0x0, 
    xTicksToWait=<optimized out>, xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  17   Thread 1073596348 (Name: hciT) 0x400982a8 in xQueueGenericReceive (xQueue=0x3ffe3350, pvBuffer=0x0, xTicksToWait=<optimized out>, 
    xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  18   Thread 1073596720 (Name: btuT) 0x400982a8 in xQueueGenericReceive (xQueue=0x3ffe386c, pvBuffer=0x0, xTicksToWait=<optimized out>, 
    xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  19   Thread 1073456008 (Name: main) 0x400982a8 in xQueueGenericReceive (xQueue=0x3ffbcfcc, pvBuffer=0x0, xTicksToWait=<optimized out>, 
    xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
  20   Thread 1073700908 (Name: periodic_task) 0x400982a8 in xQueueGenericReceive (xQueue=0x3f8057a0, pvBuffer=0x0, 
[... 68 lines cut ...]
    at /home/pierre/my_project/components/cmd_manager/cmd_manager.c:147
#16 0x40107ed6 in app_main ()
    at /home/pierre/my_project/main/main.c:123
[... cut ...]
#0  0x400982a8 in xQueueGenericReceive (xQueue=0x3ffd06bc, pvBuffer=0x3ffd1ac0, 
    xTicksToWait=<optimized out>, xJustPeeking=0)
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:214
#1  0x400d2231 in queue_recv_wrapper (queue=0x3ffd06bc, item=0x3ffd1ac0, block_time_tick=4294967295)
    at /home/pierre/my_project/common/esp-idf/components/esp32/esp_adapter.c:337
#2  0x400f9989 in ppTask ()
    at /home/pierre/my_project/common/esp-idf/components/freertos/include/freertos/portable.h:235
```

TimXia, 2020-05-28:
@pbruynbroeckQuimesis Did you use esp_timer in your own tasks?

pbruynbroeckQuimesis, 2020-05-28:
@TimXia No, I always use FreeRTOS timer (and also the `ledc_timer_config`).

TimXia, 2020-05-28:
@pbruynbroeckQuimesis Did you receive wifi disconnected event?

pbruynbroeckQuimesis, 2020-05-28:
@TimXia , yes. Here the log just after to restart the wifi hotpost:

```
[2020-05-07 22:46:25] I (171269077) wifi: ap_probe_send over, resett wifi status to disassoc
[2020-05-07 22:46:25] I (171269077) wifi: state: run -> init (c800)
[2020-05-07 22:46:25] I (171269077) wifi: pm stop, total sleep time: 81871360419 us / 171249057423 us
[2020-05-07 22:46:25] 
[2020-05-07 22:46:25] I (171269087) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
[2020-05-07 22:46:25] I (171269107) WIFI-Manager: SYSTEM_EVENT_STA_DISCONNECTED (reason: 200)
[2020-05-07 22:46:25] I (171269117) WIFI-Manager: wifi was connected or connected_with_ip
[2020-05-07 22:46:25] E (171269127) MQTT_CLIENT: Error write data or timeout, written len = -1, errno=113
[2020-05-07 22:46:25] W (171269127) MQTT_CLIENT: Publish: Losing qos0 data when client not connected
[2020-05-07 22:46:25] W (171269187) BT_GATT: gatt_ind_ack_timeout send ack now
[2020-05-07 22:46:28] I (171272337) WIFI-Manager: UPDATE RSSI of  : -88
[2020-05-07 22:46:29] I (171272337) WIFI-Manager: UPDATE RSSI of  : -88
[2020-05-07 22:46:29] I (171272337) WIFI-Manager: UPDATE RSSI of  : -88
[2020-05-07 22:46:29] I (171272347) WIFI-Manager: SYSTEM_EVENT_SCAN_DONE
[2020-05-07 22:46:41] E (171285297) MQTT_CLIENT: Error transport connect
[2020-05-07 22:46:58] E (171302297) MQTT_CLIENT: Error transport connect
[2020-05-07 22:47:15] E (171319297) MQTT_CLIENT: Error transport connect
[2020-05-07 22:47:32] E (171336297) MQTT_CLIENT: Error transport connect
[2020-05-07 22:47:49] E (171353297) MQTT_CLIENT: Error transport connect
[2020-05-07 22:48:06] E (171370297) MQTT_CLIENT: Error transport connect
[2020-05-07 22:48:23] E (171387297) MQTT_CLIENT: Error transport connect
[2020-05-07 22:48:25] I (171389107) WIFI-Manager: Other event id:8
[2020-05-07 22:48:40] E (171404297) MQTT_CLIENT: Error transport connect
[2020-05-07 22:48:57] E (171421297) MQTT_CLIENT: Error transport connect
[2020-05-07 22:49:14] E (171438297) MQTT_CLIENT: Error transport connect
[2020-05-07 22:49:36] E (171460297) MQTT_CLIENT: Error transport connect
[2020-05-07 22:49:53] E (171477297) MQTT_CLIENT: Error transport connect
[2020-05-07 22:50:10] E (171494297) MQTT_CLIENT: Error transport connect
...
```

TimXia, 2020-05-28:
@pbruynbroeckQuimesis As you turned off your hot-spot, ESP32 was disconnected from the hot-spot. After receiving disconnected event, you can call esp_wifi_connect() to connect to other APs. Note that reconnection is not done by WiFi firmware, it must be done by the application.

pbruynbroeckQuimesis, 2020-05-28:
@TimXia That's already what I do.
It always works,but when the esp runs for a while, the reconnection doesn't work and the CLI doesn't respond.

When I received the SYSTEM_EVENT_STA_DISCONNECTED event:
- I call `esp_wifi_disconnect`, (to besure to disconnect. Because, it happened that we received the SYSTEM_EVENT_STA_DISCONNECTED while the STA was still connected).
- We do a wifi scan and take the known wifi with the best SSID: (call `esp_wifi_scan_start` and `esp_wifi_scan_get_ap_records`),
- We connect to this network (call `esp_wifi_set_config` and `esp_wifi_connect`)

But as I said, it's always works except sometimes when the esp has been running for several hours.

TimXia, 2020-05-29:
@pbruynbroeckQuimesis  Are you sure that the reconnection was called? From log, the WiFi task didn't receive the reconnection command. Can you add some logs to make sure you called it.

closed 2021-11-17 by Alvin1Zhang