espressif/esp-idf #5125: Detect COM port automatically or idf list-ports

keelung-yang, 2020-04-15:
For each time of flashing boards by idf, I have to open Device Manager to check COM port name, enev if there is only one board connected.

So it's very convenient, if COM port can be detected automatically, or pop a console window to select port if -p isn't specified.

dobairoland, 2020-04-16:
Hi @keelung-yang. Have you tried flashing without specifying the port? Because I think it is already working well. If that isn't working for you as you expect then please list all of your ports in your system and tell us your ESP-IDF version. Thanks!

keelung-yang, 2020-04-16:
@dobairoland I haven't, but tried, not works. I'm using latest master branch.
```shell
esptool.py --chip esp32 -p COM9 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 dmodbus.bin
Serial port COM9
Choosing default port b'COM9' (use '-p PORT' option to set a specific serial port)
```

[image]

COM4 is my ESP32-PICO_KIT board.
COM9 is used by another device:  [neoVI FIRE](https://www.intrepidcs.com/products/vehicle-network-adapters/neovi-fire-red/)

kumekay, 2020-04-17:
@keelung-yang  Thank you for description of your system,
Could you please try to run `esptool.py` directly, without passing port to it, 
like 
```esptool.py --chip esp32 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 dmodbus.bin```

At the moment `idf.py` and `esptool.py` have two different ways to select default serial port for flashing, and one in esptool.py is more sophisticated, that should handle your case correctly. However, when you run `idf.py flash` is idf.py's way to detect port is used

keelung-yang, 2020-04-17:
@dobairoland I tried _esptool.py chip_id_. It works.
```shell
components\esptool_py\esptool\esptool.py --chip esp32 --connect-attempts 1 chip_id
esptool.py v3.0-dev
Found 4 serial ports
Serial port COM9
Connecting........_____
COM9 failed to connect: Failed to connect to ESP32: Timed out waiting for packet header
Serial port COM4
Connecting.......
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, Embedded Flash, Coding Scheme None
Crystal is 40MHz
MAC: d8:a0:1d:40:8d:90
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: d8:a0:1d:40:8d:90
Hard resetting via RTS pin...
```
But _esptool.py_ is not exported by _export.bat_ on windows.
And it takes too long time to detect, and too many & too long command parameters need to input.

Sometime there are many boards to flash one by one. 
Then all what I need as a user, is determining which port is connected to a ESP32 board **_quickly_**.
Even just list ports is helpful. Then I can compare them before and after connecting my board.
Actually I've wrote a script to detect it. But after then I've to goto another console window to run _idf.py_
So it's very very convenient if port dection or list is supported by _idf.py_ **_directly_**.
Then I can finish my work in a single console window.
```python3
import re
from serial.tools import list_ports

def find_esp32_port():
    for port in list_ports.comports():
        cp210x = r'Silicon Labs CP210x USB to UART Bridge\s*\((COM\d+)\)'
        if m := re.search(cp210x, port.description):
            return m.group(1)
    return None
```

nvannote, 2020-04-20:
It would appear that esptool.py looks to the environment variables ESPTOOL_PORT and ESPTOOL_BAUD for default values, before it tries to locate a candidate port.  The tool idf.py uses the environment variables ESPPORT and ESPBAUD.

You could set those in your environment if you use the same port frequently.

Regards

keelung-yang, 2020-04-20:
@nvannote While flashing a new connected ESP32 boards one by one, the COM port number changes with +1. So environment ariables is not a good solution.

nvannote, 2020-04-20:
> @nvannote While flashing a new connected ESP32 boards one by one, the COM port number changes with +1. So environment ariables is not a good solution.

I see.  The serial driver I use does't increment unless there is more than one connected at a time.  Well, worth a mention.

Regards

closed 2021-01-15 by espressif-bot