espressif/esp-idf #5126: Separators in idf sub-commonds

keelung-yang, 2020-04-15:
'-' or '_' ? 
It's helpful to memory to use the same separator.

IMO, '_' is used in source code usually , '-' should be used in command-line UI.

```shell
Commands:
  app-flash              Flash the app only.
  bootloader             Build only bootloader.
  bootloader-flash       Flash bootloader only.
  efuse_common_table     Genereate C-source for IDF's eFuse fields.
  efuse_custom_table     Genereate C-source for user's eFuse fields.
  encrypted-app-flash    Flash the encrypted app only.
  encrypted-flash        Flash the encrypted project.
  erase_flash            Erase entire flash chip.
  erase_otadata          Erase otadata partition.
  partition_table        Build only partition table.
  partition_table-flash  Flash partition table only.
  post_debug             Utility target to read the output of async debug
  read_otadata           Read otadata partition.
  set-target             Set the chip target to build.
  show_efuse_table       Print eFuse table.
  size-components        Print per-component size information.
  size-files             Print per-source-file size information.
```
And _Commands_ should be _Sub-commands_ exactly.

kumekay, 2020-04-17:
@keelung-yang Thank you for the issue, 
most of these names look like this due to historical reasons,
but unification is a good idea! Though these aliases should be introduced not only for `idf.py`, but also for CMake targets.
BTW Many of these commands already have `-` aliases, that are just not defaults show in help.

keelung-yang, 2020-07-31:
@kumekay It also causes unnecessary troubles. There is one example here recently.
I'm building build & package scripts. We want users to download esp32 app by themself. So I need to copy app.bin, bootloader and partition table files to our product's setup directory. But always missing partition table file in released package. Guess what? `-` and `_` after debugging: partition table is saved in: `build\partition_table\partition-table.bin`. I copied file name as it's parent directory name, following bootloader's path naming way in build script.