espressif/esp-idf #5128: cJSON: per object memory allocator

dshil, 2020-04-15:
**Is your feature request related to a problem? Please describe.**

We use cJSON to send telemetry in JSON format. We do it every seconds (or often). We have tones of sub-systems that do it. As a result we have a lot of frequent dynamic allocation/de-allocation operations per second.

I explored the API and found the possibility to override the memory routines via:

```c
/* Supply malloc, realloc and free functions to cJSON */
CJSON_PUBLIC(void) cJSON_InitHooks(cJSON_Hooks* hooks);
```

I have some sub-systems that use one type of memory (e.g. big chunk of dynamically allocated memory), another sub-systems use another type of memory (static array) and other variations. I want to use the corresponding allocators with cJSON objects too.

The issue that cJSON uses global "memory manager", aka hooks 

```c
static internal_hooks global_hooks = { internal_malloc, internal_free, internal_realloc };
```
thus, all my subsystems will share this global hook. Overriding memory functions once will affect all other components.

**Describe the solution you'd like**

I would happy to have a possibility to specify custom allocator per cJSON object. We could simply store a pointer to memory functions inside the cJSON struct. It will increase the total size of cJSON object to `sizeof(pointer to memory functions)` which seems not a big issue (we can configure this by user-defined option to be able to use previous behavior).

Any ideas, suggestions?

igrr, 2020-04-15:
The solution you propose indeed looks feasible, but it doesn't seem to be a small change. I would suggest bringing this question upstream (to https://github.com/DaveGamble/cJSON/issues), as we would prefer not to deviate much from the upstream version. If this change is accepted upstream, we will of course pull it into IDF.

dshil, 2020-04-16:
Hi @igrr and thanks for reply.

I posted this issue in upstream repo. I think we can close this issue and reopen back again if they find my suggestion reasonable.

closed 2020-04-16 by dshil