espressif/esp-idf #5129: Possible bug in `esp_wifi_set_config`

atlascoder, 2020-04-15:
## Environment

- Module or chip used:  [ESP32-PICO-D4]
- IDF version: v4.1-dev-2079-g5dbabae9d-dirty
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: vscode
- Power Supply: External

## Problem Description

WiFi credentials are not being saved to NVS when [Use a partition to store PHY init data] is off in IDF configuration.

### Expected Behavior

After `esp_wifi_set_storage(WIFI_STORAGE_FLASH)` and/or `esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config)` when storage is set to flash - SSID (and probably the `wifi_config` in whole) is not stored to NVS.

### Actual Behavior

After saving, `wifi_data` filled by `esp_wifi_get_config(ESP_IF_WIFI_STA, &wifi_config)` contains empty string in `wifi_config.sta.ssid`

### Steps to reproduce

1. `make menuconfig` and set Enable NVS for WiFi, and unset [Use a partition to store PHY init data]
2. Init NVS flash and WiFi
3. Try to save credentials to flash with `esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config)`
4. Restart ESP and try to `esp_wifi_get_config(ESP_IF_WIFI_STA, &wifi_config)` OR read WiFi partition and try to find SSID inside

Alvin1Zhang, 2020-04-16:
@atlascoder Thanks for reporting and sharing the details, we will look into.

esp-HarveyRong, 2020-04-16:
Hi @atlascoder 
Did you use the API `esp_wifi_restore` to restart the device, the API will re-initialize NVS.
If not, would you mind providing relevant code to help me analyze the problem?

atlascoder, 2020-04-17:
I tried to reproduce the issue on a project started from esp-idf example, but no luck.. All works as expected, though, it seems that something strange is happening.

Here is my repo https://github.com/atlascoder/esp-idf-iss5129

As I mentioned, the issue is dependent to storing PHY params on NVS. In my project, I, must be accidentally, turned off usage NVS for PHY. And I remember that using this config the FW didn't have `phy_init` partition. Now I see the partition in boot log all the time, independently to PHY settings. It's weird.

I always used `make erase_flash && make flash`. Also, I did the full rebuild, by deletion manually `build` and doing `make clean && make build && make erase_flash && make flash`. - No matter, the `phy_init` partition always present.

For now, I think, that firstly I need to reproduce the issue.. But I am puzzled with this `phy_init` case.. Could you tell me, is this right that this partition is always being used?

atlascoder, 2020-04-17:
> Hi @atlascoder
> Did you use the API `esp_wifi_restore` to restart the device, the API will re-initialize NVS.
> If not, would you mind providing relevant code to help me analyze the problem?

When I say about restart, I mean `esp_restart`, not wiping out the settings from the flash.

atlascoder, 2020-06-14:
Hi @esp-HarveyRong and thank you for the suggestion!

It seems like my omitting of initializing `wifi_config` with `esp_wifi_get_config` was the source of my troubles. I didn't call it, instead i set all to zero by `bzero(&wifi_config, sizeof(wifi_config_t))`. What's interesting that I had no troubles to run and connect STA with such config. After that, I even got valid ssid/pwd `esp_wifi_get_config` after I had changed memory to FLASH and called `esp_wifi_get_set`, but after `esp_restart` values become empty again.

Not sure now that my founds regarding `phy_init` do matter..

esp-HarveyRong, 2020-06-15:
Hi @atlascoder
Using [esp_wifi_restore](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_wifi.h#L322) to restart the device will re-initialize NVS. In other words, the SSID you configured will be cleared.

liuzfesp, 2020-07-10:
Similar one: https://github.com/espressif/esp-idf/issues/5214

HarveyRong-Esp, 2020-10-23:
Hi @Atlascoder,
Has your issue been resolved?

Alvin1Zhang, 2020-12-09:
@atlascoder Thanks for the report and sorry for the slow turnaround, we have a fix under internal reviewing, the issue will close automatically once the fix is available on GitHub.

closed 2021-03-12 by espressif-bot