espressif/esp-idf #5130: nvs: allow to use raw partitions

dshil, 2020-04-15:
**Is your feature request related to a problem? Please describe.**

Initially, in project, all our data was saved to the FLASH using NVS API and we were happy with it. Later one developer decided to use raw API instead (`esp_partition_erase_range`, `esp_partition_erase_range` and friends). It was done because in currently used version of ESP-IDF (v3.1) it is not possible to write big blobs of data per time (blobs with size > 1984 bytes). This limitation [was removed](https://github.com/espressif/esp-idf/commit/5a27a6354165bfc97811677930d25f1596845478) in newer version. Time went by. We kept sending customers new devices that use raw API instead of NVS instead and did not have any problems.

One day everything changed. We still need to use NVS API once again because we will periodically save some data to the flash and want to be sure that we won't destroy the FLASH very soon (due to writing data to the same location on the FLASH each time). So, NVS solves this problem internally by writing data sequentially, but using raw SPI API does not.

For new customers we can simply change our partition table and specify NVS label as library requires but it is not possible for existing customers.

**Describe alternatives you've considered**

We could perform a partition migration somehow but I did not like the idea because poorly written migration may lead to critical data lose or even to some problem with devices. We have some devices deep underground and we do not have the second chance.

As a result I started to explore the NVS source code. As I mentioned, we use v3.1.

The following code is responsible for registration a new NVS partition in the internal list of partitions:

```cpp
extern "C" esp_err_t nvs_flash_init(void)
{
    return nvs_flash_init_partition(NVS_DEFAULT_PART_NAME);
}

extern "C" esp_err_t nvs_flash_init_partition(const char *part_name)
{
    Lock::init();
    Lock lock;
    nvs::Storage* mStorage;

    mStorage = lookup_storage_from_name(part_name);
    if (mStorage) {
        return ESP_OK;
    }

    const esp_partition_t* partition = esp_partition_find_first(
            ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_NVS, part_name);
    if (partition == NULL) {
        return ESP_ERR_NOT_FOUND;
    }

    return nvs_flash_init_custom(part_name, partition->address / SPI_FLASH_SEC_SIZE,
            partition->size / SPI_FLASH_SEC_SIZE);
}
```

When partition is successfully registered we can read and write data to it.

----

My solution was to introduce the possibility to work with raw partition somehow. I did the following dirty hack and extended NVS API a little bit:

```c
 * @brief Initialize NVS flash storage for the specified raw partition.
 *
 * @param[in]  partition_label   Label of the partition. Note that internally a reference to
 *                               passed value is kept and it should be accessible for future operations
 *
 * @param[in]  type              Type of the partition to be initialized.
 * @param[in]  subtype           Subtype of the partition to be initialized.
 *
 * @return
 *      - ESP_OK if storage was successfully initialized.
 *      - ESP_ERR_NVS_NO_FREE_PAGES if the NVS storage contains no empty pages
 *        (which may happen if NVS partition was truncated)
 *      - ESP_ERR_NOT_FOUND if specified partition is not found in the partition table
 *      - one of the error codes from the underlying flash storage driver
 */
esp_err_t nvs_flash_init_partition_raw(const char *partition_label,
                                       esp_partition_type_t type,
                                       esp_partition_subtype_t subtype);

/**
 * @brief Erase specified raw partition
 *
 * This function erases all contents of specified raw partition
 *
 * @param[in]  part_name    Name (label) of the partition to be erased
 * @param[in]  type         Type of the partition to be erased
 * @param[in]  subtype      Subtype of the partition to be erased
 *
 * @return
 *      - ESP_OK on success
 *      - ESP_ERR_NOT_FOUND if there is no raw partition with the specified
 *        parameters in the partition table
 */
esp_err_t nvs_flash_erase_partition_raw(const char *part_name,
                                        esp_partition_type_t type,
                                        esp_partition_subtype_t subtype);
```

----

And the implementation was just copy-pasted from the library code:

```cpp
extern "C" esp_err_t nvs_flash_init_partition_raw(const char* part_name,
                                                  esp_partition_type_t type,
                                                  esp_partition_subtype_t subtype)
{
    Lock::init();
    Lock lock;
    nvs::Storage* mStorage;

    mStorage = lookup_storage_from_name(part_name);
    if (mStorage) {
        return ESP_OK;
    }

    const esp_partition_t* partition = esp_partition_find_first(
        type, subtype, part_name);
    if (partition == NULL) {
        return ESP_ERR_NOT_FOUND;
    }

    return nvs_flash_init_custom(part_name,
                                 partition->address / SPI_FLASH_SEC_SIZE,
                                 partition->size / SPI_FLASH_SEC_SIZE);
}

extern "C" esp_err_t nvs_flash_erase_partition_raw(const char* part_name,
                                                   esp_partition_type_t type,
                                                   esp_partition_subtype_t subtype)
{
    const esp_partition_t* partition = esp_partition_find_first(
        type, subtype, part_name);
    if (partition == NULL) {
        return ESP_ERR_NOT_FOUND;
    }

    return esp_partition_erase_range(partition, 0, partition->size);
}
```

----

We tested dozen of times all seem to work fine. We even used it in the production for several customers and all data were kept.

----

Questiots to core team (@igrr):

- Is it legal? :)
- What could you suggest as a "right" solution in this scenario?

I would very appreciate any advises, ideas.

igrr, 2020-04-15:
@dshil The change you have done looks okay, i don't think it will lead to any issues as long as you keep track which partition is used for what purpose.
If you are looking for a minimal change to achieve this, then you could also modify `nvs_flash_init_partition` and `nvs_flash_erase_partition` to pass `ESP_PARTITION_SUBTYPE_ANY` instead of `ESP_PARTITION_SUBTYPE_DATA_NVS`. This will work as long as the partition labels are unique.

dshil, 2020-04-16:
Hi @igrr and thanks for reply.

I am not sure that using `ESP_PARTITION_SUBTYPE_ANY` will work due to the fact that we have the following partition table (instead foo, bar, baz real names are used, size and offset are also obfuscated for security reason):

```
nvs,      data, nvs,     0x0000, 0x0000,
otadata,  data, ota,     0x0000, 0x0000,
phy_init, data, phy,     0x0000, 0x0000,
factory,  app,  factory, 0x0000, 0x0000,
ota_0,    app,  ota_0,   0x0000, 0x0000,
ota_1,    app,  ota_1,   0x0000, 0x0000,
foo,      0x40, 0x01,    0x0000, 0x0000,
bar,      0x40, 0x02,    0x0000, 0x0000,
baz,      0x40, 0x03,    0x0000, 0x0000,
reserv_1, 0x40, 0x04,    0x0000, 0x0000,
reserv_2, 0x40, 0x05,    0x0000, 0x0000,
```

The problem is that the developer decided to use not only own **subtype** but also own **type**. Thus I still need to use my dirty solution.

Do you think this functionality can be useful for somebody who uses the upstream ESP-IDF. If yes we can think about what kind of API to provide (I can prepare a patch if you find it reasonable).

igrr, 2020-04-16:
I think the best solution would be change the argument of `nvs_flash_init_partition` and `nvs_flash_erase_partition` from string (const char*) to `const esp_partition_t*`. Then the user can simply pass the pointer to the partition structure, obtained from esp_partition_find or some other way.

Of course we can't change the argument type as that would be an incompatible change. It's a shame we aren't using C11 yet, otherwise we could add a new "overload" using [generics](https://en.cppreference.com/w/c/language/generic).

Perhaps we can create new functions `nvs_flash_init_partition_ptr` and `nvs_flash_erase_partition_ptr` which would take `const esp_partition_t*`. I don't know how widely useful this will be, though.

dshil, 2020-04-25:
Hi @igrr, I've prepare PR #5172 for this issue. I would be great if you will have time to review it.

P.S. sorry for bother you directly. I do not know anybody from core team and do not know how the PR review process is done in your project (which PR is review first).

dshil, 2020-06-15:
@igrr, I think we've done here and close the issue.

Alvin1Zhang, 2020-06-16:
Glad to hear, will close the issue, regarding backporting to v4.0.2, let us follow up in https://github.com/espressif/esp-idf/pull/5172, thanks.

closed 2020-06-16 by Alvin1Zhang

dshil, 2020-06-16:
Thanks @Alvin1Zhang