espressif/esp-idf #5131: Compilation error with latest master commit 2e14149

KaeLL, 2020-04-15:
```
/esp_idf/test_master/esp-idf/components/driver/spi_bus_lock.c: In function 'spi_bus_lock_init_main_dev':
/esp_idf/test_master/esp-idf/components/driver/spi_bus_lock.c:822:41: error: implicit declaration of function 'xSemaphoreCreateBinaryStatic'; did you mean 'xSemaphoreCreateBinary'? [-Werror=implicit-function-declaration]
     g_spi_lock_main_flash_dev->semphr = xSemaphoreCreateBinaryStatic(&main_flash_semphr);
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                         xSemaphoreCreateBinary
```

fpgamaster, 2020-04-15:
add CONFIG_FREERTOS_SUPPORT_STATIC_ALLOCATION=y to std config

igrr, 2020-04-15:
Closing as duplicate of https://github.com/espressif/esp-idf/issues/5046, please follow the link and click "Subscribe" to get notified when the fix is merged.

closed 2020-04-15 by igrr