espressif/esp-idf #5133: Keep getting asked to install python tools reqirements but they are already satisfied

noweare1, 2020-04-16:
**Using ubuntu linux and esp-idf 4.x**

**When I try and run  idf.py menuconfig     output is** 
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
The following Python requirements are not satisfied:
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui>=0.13.2.0
pygdbmi<=0.9.0.2
To install the missing packages, please run "/home/jrm/esp/esp-idf/install.sh"

**I keep getting asked to install python tools because of "missing requirements"
But when I run the install.sh command it says everything is installed. This is an endless loop.**

**so I install as directed in "getting started"** 
using command  "home/jrm/esp/esp-idf/install.sh"

**output says everything is installed and requirement already satisfied.**

Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r1-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r1-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200406 (already installed)
Installing Python environment and packages
Installing Python packages from /home/jrm/esp/esp-idf/requirements.txt
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from -r /home/jrm/esp/esp-idf/requirements.txt (line 4)) (46.1.3)
Requirement already satisfied: click>=5.0 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from -r /home/jrm/esp/esp-idf/requirements.txt (line 8)) (7.1.1)
Requirement already satisfied: pyserial>=3.0 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from -r /home/jrm/esp/esp-idf/requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from -r /home/jrm/esp/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from -r /home/jrm/esp/esp-idf/requirements.txt (line 11)) (2.9)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from -r /home/jrm/esp/esp-idf/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from -r /home/jrm/esp/esp-idf/requirements.txt (line 13)) (0.26)
Requirement already satisfied: gdbgui>=0.13.2.0 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from -r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (0.13.2.0)
Requirement already satisfied: pygdbmi<=0.9.0.2 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from -r /home/jrm/esp/esp-idf/requirements.txt (line 15)) (0.9.0.2)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from cryptography>=2.1.4->-r /home/jrm/esp/esp-idf/requirements.txt (line 11)) (1.14.0)
Requirement already satisfied: six>=1.4.1 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from cryptography>=2.1.4->-r /home/jrm/esp/esp-idf/requirements.txt (line 11)) (1.14.0)
Requirement already satisfied: Pygments<3.0,>=2.2.0 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from gdbgui>=0.13.2.0->-r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (2.6.1)
Requirement already satisfied: Flask<1.0,>=0.12.2 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from gdbgui>=0.13.2.0->-r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (0.12.5)
Requirement already satisfied: Flask-SocketIO<3.0,>=2.9 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from gdbgui>=0.13.2.0->-r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (2.9.6)
Requirement already satisfied: Flask-Compress<2.0,>=1.4.0 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from gdbgui>=0.13.2.0->-r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (1.4.0)
Requirement already satisfied: gevent<2.0,>=1.2.2 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from gdbgui>=0.13.2.0->-r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (1.4.0)
Requirement already satisfied: pycparser in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/jrm/esp/esp-idf/requirements.txt (line 11)) (2.20)
Requirement already satisfied: Jinja2>=2.4 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from Flask<1.0,>=0.12.2->gdbgui>=0.13.2.0->-r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (2.11.1)
Requirement already satisfied: itsdangerous>=0.21 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from Flask<1.0,>=0.12.2->gdbgui>=0.13.2.0->-r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (1.1.0)
Requirement already satisfied: Werkzeug<1.0,>=0.7 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from Flask<1.0,>=0.12.2->gdbgui>=0.13.2.0->-r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (0.16.1)
Requirement already satisfied: python-socketio>=1.6.1 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from Flask-SocketIO<3.0,>=2.9->gdbgui>=0.13.2.0->-r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (4.5.1)
Requirement already satisfied: greenlet>=0.4.14; platform_python_implementation == "CPython" in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from gevent<2.0,>=1.2.2->gdbgui>=0.13.2.0->-r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (0.4.15)
Requirement already satisfied: MarkupSafe>=0.23 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from Jinja2>=2.4->Flask<1.0,>=0.12.2->gdbgui>=0.13.2.0->-r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (1.1.1)
Requirement already satisfied: python-engineio>=3.9.0 in ./.espressif/python_env/idf4.2_py3.6_env/lib/python3.6/site-packages (from python-socketio>=1.6.1->Flask-SocketIO<3.0,>=2.9->gdbgui>=0.13.2.0->-r /home/jrm/esp/esp-idf/requirements.txt (line 14)) (3.12.1)

**Then I run the export.sh command
Below is the output after command   ".  /home/jrm/esp/esp-idf/export.sh"**

Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
The following Python requirements are not satisfied:
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui>=0.13.2.0
pygdbmi<=0.9.0.2
To install the missing packages, please run "/home/jrm/esp/esp-idf/install.sh"
 
**This keeps happening in an endless loop**

**Thank you for taking a look**

igrr, 2020-04-16:
Hi @noweare1, thanks for reporting, we will need to gather a bit more information to troubleshoot this.

1. Please post the output of `which -a python` (in a new terminal, before running export.sh), `git describe --tags` (in IDF directory, so that we know the exact commit ID)

2. Please add the following lines in tools/check_python_dependencies.py right after `if __name__ == "__main__":` and post the output of `. ./export.sh`:

```python
    print('{} sys.executable: {}'.format(sys.argv[0], sys.executable))
    print('{} PATH: {}'.format(sys.argv[0], os.getenv("PATH")))
    print('Environment:\n----')
    print(repr(os.environ))
    print('---\n')
```

Thanks.

davebaldwin, 2020-05-04:
I too am seeing this problem.
After installing as per the getting started instructions everything works fine as expected but if I open an new terminal window , cd to the esp-idf directory and run
. $HOME/esp/esp-idf/export.sh
cd to my project directory and run.

Responding the the request for extra information:
~ % which -a python
/usr/bin/python
/Users/dave/.espressif/python_env/idf4.2_py2.7_env/bin/python
~ % 

After inserting the code into check_python_dependencies.py the output is:

 esp-idf % . ./export.sh
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
/Users/dave/esp/esp-idf/tools/check_python_dependencies.py sys.executable: /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
/Users/dave/esp/esp-idf/tools/check_python_dependencies.py PATH: /Users/dave/.gem/ruby/2.1.7/bin:/Users/dave/.rubies/2.1.7/lib/ruby/gems/2.1.0/bin:/Users/dave/.rubies/2.1.7/bin:/Users/dave/.gem/ruby/2.7.0/bin:/Users/dave/.rubies/2.7.0/lib/ruby/gems/2.7.0/bin:/Users/dave/.rubies/2.7.0/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/Users/dave/esp/esp-idf/components/esptool_py/esptool:/Users/dave/esp/esp-idf/components/espcoredump:/Users/dave/esp/esp-idf/components/partition_table:/Users/dave/esp/esp-idf/components/app_update:/Users/dave/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin:/Users/dave/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin:/Users/dave/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/dave/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/dave/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200420/openocd-esp32/bin:/Users/dave/.espressif/python_env/idf4.2_py2.7_env/bin:/Users/dave/esp/esp-idf/tools
Environment:
----
{'GEM_ROOT': '/Users/dave/.rubies/2.1.7/lib/ruby/gems/2.1.0', 'IDF_PATH': '/Users/dave/esp/esp-idf', 'TERM_PROGRAM_VERSION': '433', 'RUBY_VERSION': '2.1.7', 'LOGNAME': 'dave', 'USER': 'dave', 'HOME': '/Users/dave', 'PATH': '/Users/dave/.gem/ruby/2.1.7/bin:/Users/dave/.rubies/2.1.7/lib/ruby/gems/2.1.0/bin:/Users/dave/.rubies/2.1.7/bin:/Users/dave/.gem/ruby/2.7.0/bin:/Users/dave/.rubies/2.7.0/lib/ruby/gems/2.7.0/bin:/Users/dave/.rubies/2.7.0/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/Users/dave/esp/esp-idf/components/esptool_py/esptool:/Users/dave/esp/esp-idf/components/espcoredump:/Users/dave/esp/esp-idf/components/partition_table:/Users/dave/esp/esp-idf/components/app_update:/Users/dave/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin:/Users/dave/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin:/Users/dave/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/dave/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/dave/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200420/openocd-esp32/bin:/Users/dave/.espressif/python_env/idf4.2_py2.7_env/bin:/Users/dave/esp/esp-idf/tools', 'RUBY_ROOT': '/Users/dave/.rubies/2.1.7', 'TERM_PROGRAM': 'Apple_Terminal', 'LANG': 'en_GB.UTF-8', 'TERM': 'xterm-256color', 'SHELL': '/bin/zsh', 'VERSIONER_PYTHON_VERSION': '2.7', 'SHLVL': '1', 'SECURITYSESSIONID': '186a6', 'XPC_FLAGS': '0x0', '_': '/usr/bin/python', 'OPENOCD_SCRIPTS': '/Users/dave/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200420/openocd-esp32/share/openocd/scripts', 'GEM_PATH': '/Users/dave/.gem/ruby/2.1.7:/Users/dave/.rubies/2.1.7/lib/ruby/gems/2.1.0', 'IDF_TOOLS_INSTALL_CMD': '/Users/dave/esp/esp-idf/install.sh', 'TERM_SESSION_ID': 'A16A76E3-1B11-40EB-9BC3-2F5DA88978DE', 'SSH_AUTH_SOCK': '/private/tmp/com.apple.launchd.DQVQ9h6Sfc/Listeners', 'XPC_SERVICE_NAME': '0', 'RUBYOPT': '', 'TMPDIR': '/var/folders/yz/25z1sq2527sf0nmw8y3_v2kw0000gn/T/', 'LaunchInstanceID': '61019497-0233-4195-8BAB-073111D0FA35', 'IDF_PYTHON_ENV_PATH': '/Users/dave/.espressif/python_env/idf4.2_py2.7_env', 'RUBY_ENGINE': 'ruby', 'OLDPWD': '/Users/dave/esp/pendant', '__CF_USER_TEXT_ENCODING': '0x1F5:0:2', 'IDF_TOOLS_EXPORT_CMD': '/Users/dave/esp/esp-idf/export.sh', 'PWD': '/Users/dave/esp/esp-idf', 'GEM_HOME': '/Users/dave/.gem/ruby/2.1.7'}
---

The following Python requirements are not satisfied:
click>=5.0
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui>=0.13.2.0
pygdbmi<=0.9.0.2
To install the missing packages, please run "/Users/dave/esp/esp-idf/install.sh"
esp-idf % 

Thanks,

Dave.

igrr, 2020-05-04:
Hi Dave, thanks for this extra information! Couple of additional questions:

1) Which shell are you using?
2) Are you adding any IDF directories to the PATH in your profile script?

What looks odd is that all IDF related directories are added at the end of the PATH variable, not at the front as they normally should be added by `export.sh`. Because of this, system Python copy is getting priority over the IDF-specific python virtual environment — this can be seen in `which -a python` output.

When you open a new terminal window and run `echo $PATH`, do you see any IDF-related paths in there?

davebaldwin, 2020-05-04:
Hi Ivan,

I am using zsh and I was appending the idf directories to my path in my .zshrc.

As an experiment I remove the directory setup in .zshrc and everything worked as expected.

I manually added the paths as including the commands to source export.sh in .zshrc (or bash) outputs a lot of stuff I didn’t want to see every time I opened a new terminal window for other non esp use.

Maybe a note in the documents that the path needs to be before the system python, etc.. or have export.sh run silently unless a problem is found?

I hadn’t suspected my additions to .zshrc as running export.sh didn’t fix things in any case (I don’t know why it didn’t fix up the path) so I was expecting the problem to be elsewhere.

Thanks for you help.

Dave.

igrr, 2020-05-04:
Right, really sorry about the instructions in the docs which tell about adding export.sh to the profile script. It is a bad idea and as you have noticed it can cause issues. We are updating the docs ASAP to replace that line with a suggestion to add an alias:

```
alias get_idf='. $HOME/esp/esp-idf/export.sh'
```

And then you can type `get_idf` in an interactive session to run export.sh without typing the full command.

igrr, 2020-05-05:
@davebaldwin the docs have been updated in https://github.com/espressif/esp-idf/commit/01b3ffa8e34d4448b2bd4f014bc05cbeac1cd97d to remove recommendation about running export.sh from the profile script.

igrr, 2020-05-05:
@noweare1 Did you have a chance to check https://github.com/espressif/esp-idf/issues/5133#issuecomment-614536173 and give us additional information to troubleshoot your issue?

chegewara, 2020-05-24:
Hi,
because my ubuntu 18 decided to crash and i couldnt recover it i decided to install ubuntu 20 this time. Yes, i encountered some problems with `. ~/esp/esp-idf/export.sh` prerequisites even if all was installed properly.
I found one very simple solution, and the way i solved it i think its more ubuntu problem not esp.
Here is solution that worked for me:
https://linuxconfig.org/ubuntu-20-04-python-version-switch-manager

yuhua0731, 2020-06-24:
@igrr sorry for bothering you. same problem occurs on my vscode. I have done what you mentioned already. IDF Onboarding is done successfully with all requirements satisfied, while it ends with error when I try to launch gui configuration tool. output is like this:

/home/huayu/esp-idf/tools/check_python_dependencies.py sys.executable: /usr/bin/python
/home/huayu/esp-idf/tools/check_python_dependencies.py PATH: /home/huayu/esp-idf/components/esptool_py/esptool:/home/huayu/esp-idf/components/espcoredump:/home/huayu/esp-idf/components/partition_table:.:/home/huayu/esp-idf/tools:.:/usr/bin:/home/huayu/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin:/home/huayu/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin:/home/huayu/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/huayu/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/huayu/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200420/openocd-esp32/bin:/home/huayu/.local/bin:/home/huayu/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/bin:/home/huayu/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin:/home/huayu/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin:/home/huayu/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/huayu/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin
Environment:
----
environ({'ELECTRON_RUN_AS_NODE': '1', 'GJS_DEBUG_TOPICS': 'JS ERROR;JS LOG', 'USER': 'huayu', 'SSH_AGENT_PID': '1430', 'XDG_SESSION_TYPE': 'x11', 'SHLVL': '0', 'HOME': '/home/huayu', 'QT4_IM_MODULE': 'ibus', 'DESKTOP_SESSION': 'ubuntu', 'GIO_LAUNCHED_DESKTOP_FILE': '/usr/share/applications/code.desktop', 'GTK_MODULES': 'gail:atk-bridge', 'GNOME_SHELL_SESSION_MODE': 'ubuntu', 'MANAGERPID': '1245', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus', 'GIO_LAUNCHED_DESKTOP_FILE_PID': '63245', 'IM_CONFIG_PHASE': '1', 'LOGNAME': 'huayu', 'GTK_IM_MODULE': 'ibus', '_': '/usr/share/code/code', 'JOURNAL_STREAM': '9:736258', 'XDG_SESSION_CLASS': 'user', 'USERNAME': 'huayu', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'WINDOWPATH': '2', 'PATH': '/home/huayu/esp-idf/components/esptool_py/esptool:/home/huayu/esp-idf/components/espcoredump:/home/huayu/esp-idf/components/partition_table:.:/home/huayu/esp-idf/tools:.:/usr/bin:/home/huayu/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin:/home/huayu/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin:/home/huayu/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/huayu/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/huayu/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200420/openocd-esp32/bin:/home/huayu/.local/bin:/home/huayu/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/bin:/home/huayu/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin:/home/huayu/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin:/home/huayu/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/huayu/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin', 'SESSION_MANAGER': 'local/ubuntu:@/tmp/.ICE-unix/1548,unix/ubuntu:/tmp/.ICE-unix/1548', 'INVOCATION_ID': '64f14a09ddcc41a088c8376883533345', 'XDG_RUNTIME_DIR': '/run/user/1000', 'XDG_MENU_PREFIX': 'gnome-', 'DISPLAY': ':0', 'LANG': 'en_US.UTF-8', 'XDG_CURRENT_DESKTOP': 'Unity', 'XAUTHORITY': '/run/user/1000/gdm/Xauthority', 'XDG_SESSION_DESKTOP': 'ubuntu', 'XMODIFIERS': '@im=ibus', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'SHELL': '/bin/bash', 'QT_ACCESSIBILITY': '1', 'GDMSESSION': 'ubuntu', 'GPG_AGENT_INFO': '/run/user/1000/gnupg/S.gpg-agent:0:1', 'GJS_DEBUG_OUTPUT': 'stderr', 'QT_IM_MODULE': 'ibus', 'PWD': '/home/huayu', 'CLUTTER_IM_MODULE': 'ibus', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/etc/xdg', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/usr/local/share/:/usr/share/', 'CHROME_DESKTOP': 'code-url-handler.desktop', 'ORIGINAL_XDG_CURRENT_DESKTOP': 'ubuntu:GNOME', 'VSCODE_NLS_CONFIG': '{"locale":"en-us","availableLanguages":{},"_languagePackSupport":true}', 'VSCODE_NODE_CACHED_DATA_DIR': '/home/huayu/.config/Code/CachedData/a5d1cc28bb5da32ec67e86cc50f84c67cc690321', 'VSCODE_LOGS': '/home/huayu/.config/Code/logs/20200624T132409', 'VSCODE_IPC_HOOK': '/run/user/1000/vscode-0f4a6180-1.46.0-main.sock', 'VSCODE_PID': '63245', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'LS_COLORS': '', 'AMD_ENTRYPOINT': 'vs/workbench/services/extensions/node/extensionHostProcess', 'PIPE_LOGGING': 'true', 'VERBOSE_LOGGING': 'true', 'VSCODE_IPC_HOOK_EXTHOST': '/tmp/vscode-ipc-794764dc-d4af-4239-8b83-1d9b791a8258.sock', 'VSCODE_HANDLES_UNCAUGHT_ERRORS': 'true', 'VSCODE_LOG_STACK': 'false', 'APPLICATION_INSIGHTS_NO_DIAGNOSTIC_CHANNEL': 'true', 'IDF_TARGET': 'esp32', 'PYTHONUNBUFFERED': '0', 'OPENOCD_SCRIPTS': '/home/huayu/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200420/openocd-esp32/share/openocd/scripts', 'IDF_PATH': '/home/huayu/esp-idf', 'PYTHON': '/usr/bin/python', 'IDF_PYTHON_ENV_PATH': '.'})
---

The following Python requirements are not satisfied:
gdbgui>=0.13.2.0
pygdbmi<=0.9.0.2
reedsolo==1.5.3
bitstring>=3.1.6
[... cut ...] elf/bin:/home/huayu/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin:/home/huayu/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/huayu/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin

ESP-IDF v4.2-dev-1660-g7d7521367-dirty

---------------------------ERROR--------------------------

Received signal: null with code: 1
-----------------------END OF ERROR-----------------------

Thanks for your help!

Ivy

igrr, 2020-06-24:
@brianignacio5 could you have a look at this case with VS Code extension above (https://github.com/espressif/esp-idf/issues/5133#issuecomment-648605984)? Looks like IDF_PYTHON_ENV_PATH, PYTHON are not set correctly, and PATH doesn't include the path to IDF python virtual environment:

`/home/huayu/esp-idf/components/esptool_py/esptool:/home/huayu/esp-idf/components/espcoredump:/home/huayu/esp-idf/components/partition_table:.:/home/huayu/esp-idf/tools:.:/usr/bin:/home/huayu/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin:/home/huayu/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin:/home/huayu/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/huayu/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/huayu/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200420/openocd-esp32/bin:/home/huayu/.local/bin:/home/huayu/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/bin:/home/huayu/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin:/home/huayu/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin:/home/huayu/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/huayu/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin`

A-Babin, 2020-06-29:
I had the same problem with the "requirements are not satisfied", "requirements already satisfied" loop on Windows 10 after following the v4.1-beta2 get started guide. I seem to have solved the issue by moving all of the espressif PATH entries to the top/to the beginning of the PATH variable. 

Since, I am trying to set up the environment with CLion (_without_ PlatformIO) and don't really want all those entries in my user PATH var, I removed them, and instead, overrode the PATH var in CLion's CMake Environment setting with only the espressif and esp-idf paths, and it is working well now. 👍

> PATH=Z:\.espressif\tools\idf-exe\1.0.1\\;Z:\.espressif\tools\ninja\1.10.0\\;Z:\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200420\openocd-esp32\bin\;Z:\.espressif\tools\ccache\3.7\\;Z:\.espressif\tools\cmake\3.16.4\bin\;Z:\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin\;Z:\esp\esp-idf\tools\;Z:\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin\;Z:\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin\;Z:\.espressif\python_env\idf4.2_py3.8_env\Scripts\

A-Babin, 2020-06-29:
I had the same problem with the "requirements are not satisfied", "requirements already satisfied" loop on Windows 10 after following the v4.1-beta2 get started guide. I seem to have resolved it by moving all of the esp PATH entries to the top/front. 

I didn't actually want all of those entries in my user environment PATH, so I removed them, and instead, overrode PATH in CLion's CMake Environment setting with only the espressif and esp-idf paths. That fixed things with CLion, but using the `export.ps1` script to load the PATH into a terminal session so I could use idf.py still didn't work. Weirdly, executing `idf.py` in a terminal or command prompt would open the idf.py script in VSCode.

I'm not sure what exactly I did to fix that, but moving Python's folder from the user environment PATH to the system PATH (basically, prepending it ahead of the user PATH entries) seems to have been necessary. There was a PATH entry for platformio's virtual environment I removed that may have been interfering, but I added it back while trying to debug and it doesn't seem to make a difference.

In any case, I'm glad to have it all working now. 👍

brianignacio5, 2020-06-30:
Hi @yuhua0731 Could you open an issue in [ESP-IDF extension for Visual Studio Code Bug Report](https://github.com/espressif/vscode-esp-idf-extension/issues/new?labels=bug-github&template=bug_report.md&title=) and give us a bit more description about your setup there ?

Would review gui menuconfig in vscode extension to see if everything is working.

Please take a look at the [Onboarding documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md) to make sure extension is configured properly.

kunjal83, 2020-07-13:
Hi,
I am facing same error post upgrading to the latest version of IDF and Python 3.7 on Ubuntu. 
I need some help getting this fixed. 

Thanks.

Totalus, 2020-09-10:
I ran into this issue as well, for the `gdbgui` package. In my case it was a python dependency issue. A warning at the end of the install script gave me a hint.

I realized that `gdbgui 0.14` [now relies](https://github.com/cs01/gdbgui/blob/853f1412d667cf14767b10dc6092091c9e5a1fe0/setup.py#L24) on `pygdbmi>=0.10.0.0b0, <0.11`.
However `pygdbmi`  is topped at 0.9.0.2 in the `requirements.txt` file.

So it ends up missing a required dependency and being incorrectly installed.

**Workaround** : before running `install.sh`, comment or remove the `pygdmi<=0.9.0.2` line from `requirements.txt`.

JAICHANGPARK, 2020-09-12:
@Totalus  Thanks it works for me

mazorvanunu, 2020-09-12:
why is it that you fix smth and then it keeps coming back to you over and over and over and over again...... have a search for let's say 
"esp idf The following Python requirements are not satisfied:"
and you'll see that the same errors are hunting you over and over and over and over and over haha wow(!), just stop! stop! re-orient yourselves and come back another way! my goodness!, by the way, this is probably since the user might have multiple python envs which the toolchain communicate with. a good tip to solve it would be to you the python executable from python_env folder (where the idf is located) and using the pip from the same folder to install stuff there (also, to whom it may concern, if you have problems with idf.py, ITS THE SAME DEAL, use the specific python in the python_env folder to initiate it (add -i for windows to see errors if any).

stop, re-think, re-implement.

jasont14, 2020-09-12:
> I ran into this issue as well, for the `gdbgui` package. In my case it was a python dependency issue. A warning at the end of the install script gave me a hint.
>
> I realized that `gdbgui 0.14` [now [...]

Thanks.  This worked for me.

crgarcia12, 2020-09-14:
@jasont14  Thank you very much! your workaround worked for me

omnipresense, 2020-09-15:
This also worked:
https://github.com/espressif/esp-idf/issues/5096#issuecomment-678818026

```python -m pip install pygdbmi==0.9.0.2 gdbgui==0.13.2.0```

gerritonagoodday, 2020-09-23:
This works for me by fully qualifying all paths, which removes all the ambiguity of the many versions of Python on my machine. 

```
$ which idf.py
/home/sweethome/git/esp-idf/tools/idf.py
$ python /home/sweethome/git/esp-idf/tools/idf.py menuconfig
```

ThomasGeor, 2020-10-09:
**alias get_idf='. $HOME/esp/esp-idf/export.sh'** worked for me

Alvin1Zhang, 2020-10-12:
Thanks for reporting and sorry for late reply, the fix is available at https://github.com/espressif/esp-idf/commit/5c5c7c4449eac9301511f2b6612aec9093479706. Feel free to reopen if the issue still happens. Thanks.

closed 2020-10-12 by Alvin1Zhang