espressif/esp-idf #5134: Bluetooth a2dp_source program works with Echo Dot on ESP-IDF v3.2.2 but not with v4.0

tonystuart, 2020-04-16:
## Environment

- Development Kit: HiLetgo / NodeMCU ESP32
- Module or chip used:  ESP32-WROOM-32 / ESP32D0WDQ6
- IDF versions:
  - v4.2-dev-1126-gd85d3d969
  - v4.0
  - v3.2.2

- Build System: Both Make and CMake / idf.py
- Compiler versions:
  - xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0 (for v4.2)
  - xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0 (for v4.0)
  - xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0 (for v3.2.2)

- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

The Bluetooth a2dp_source example connects successfully to Echo Dot with ESP-IDF v3.2.2, but fails to do so with v4.0 or v4.2 (current master branch).

For more information, see https://esp32.com/viewtopic.php?f=13&t=14888

I recently tried to simplify the a2dp_source example to see if I could figure out more about the cause of the problem. I have attached this stripped-down program, as well as log output from a typical successful run on esp-idf v3.2.2 and a typical failing run on esp-idf v4.0.

I can spend some more time looking into this issue, but I'm running out of things to try. My next idea is to start comparing the differences in Bluetooth source between v3.2.2 and v4.0, but I think that may be a major effort. I've tried enabling the verbose mode for all the Bluetooth logging, but aside from some general patterns about what works and what doesn't I can't really spot the problem. Is there any way to see just what's "going over the wire" between the ESP32 and the Echo Dot, so I can compare the difference between a working environment and a failing one?

### Expected Behavior

To pair the a2dp_source with an Echo Dot as a sink and use it as a Bluetooth speaker that I can send audio data to.

### Actual Behavior

It pairs successfully, but the MEDIA_CTRL_START command fails after about 15 seconds (in later versions of ESP-IDF).

### Steps to reproduce

It is possible to reproduce this problem using the ESP-IDF a2dp_source example as well as the  stripped-down version I've attached.

1. Go into the Alexa app and disconnect and forget the a2dp_source to return to a known starting condition. This may require unplugging / replugging the Echo Dot while the ESP32 is powered off.

2. For the a2dp_source, replace ESP_SPEAKER with the name of the Echo Dot. For my stripped-down version, fill in the six byte Bluetooth Device Address (BDA).

3. Compile and flash the standard a2dp_source example, or the version I've attached. Tell the Echo Dot to pair.

4. With ESP-IDF v3.2.2 it pairs successfully, including disconnecting and reconnecting. On ESP-IDF v4.0 (or v4.2 current master) it pairs but it fails to start the audio stream.

### Code to reproduce this issue

This is the basic logic. See attachment for the delay function and a2dp_callback and event formatting functions. It's also possible to recreate the problem using the standard a2dp_source example.

```cpp
#define $ ESP_ERROR_CHECK
void app_main()
{
    esp_log_level_set("phy_init", ESP_LOG_INFO);

    nvs_flash_init();
    $(nvs_flash_init());

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    $(esp_bt_controller_init(&bt_cfg));
    $(esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT));
    $(esp_bluedroid_init());
    $(esp_bluedroid_enable());

    $(esp_bt_dev_set_device_name("ESP32"));
    $(esp_a2d_register_callback(a2dp_callback));
    $(esp_a2d_source_register_data_callback(data_callback));
    $(esp_a2d_source_init());

#if ESP_IDF_VERSION_MAJOR > 3
    esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
#else
    esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE);
#endif

    esp_bd_addr_t echo_dot;
    echo_dot[0] = 0x0c;
    echo_dot[1] = 0xee;
    echo_dot[2] = 0x99;
    echo_dot[3] = 0x26;
    echo_dot[4] = 0x2d;
    echo_dot[5] = 0x81;

    delay(30, "Please place the speaker in pairing mode");
    $(esp_a2d_source_connect(echo_dot));
    delay(10, "Please wait for device to connect");
    $(esp_a2d_media_ctrl(ESP_A2D_MEDIA_CTRL_CHECK_SRC_RDY));
    delay(5, "Please wait for response from check-source-ready");
    $(esp_a2d_media_ctrl(ESP_A2D_MEDIA_CTRL_START));
    delay(10, "Please wait for data to stream");
    $(esp_a2d_media_ctrl(ESP_A2D_MEDIA_CTRL_CHECK_SRC_RDY));
    delay(5, "Please wait for response from check-source-ready");
    $(esp_a2d_media_ctrl(ESP_A2D_MEDIA_CTRL_STOP));
    $(esp_a2d_source_disconnect(echo_dot));
    delay(10, "Please wait for restart");
    esp_restart();
}
```

## Debug Logs

One difference in the successful version is that MEDIA CTRL ACK event occurs very shortly after starting the audio:

```
W (46932) MAIN: a2dp_callback event=ESP_A2D_MEDIA_CTRL_ACK_EVT
W (46932) MAIN: a2dp_callback cmd=ESP_A2D_MEDIA_CTRL_START
W (46942) MAIN: a2dp_callback status=ESP_A2D_MEDIA_CTRL_ACK_SUCCESS
W (46942) MAIN: a2dp_callback event=ESP_A2D_AUDIO_STATE_EVT
W (46952) MAIN: a2dp_callback state=ESP_A2D_AUDIO_STATE_STARTED
W (46962) MAIN: a2dp_callback remote_bda=0xc:0xee:0x99:0x26:0x2d:0x81
```

Whereas in the failing version, it is delayed for about 15 seconds 

```
W (62012) MAIN: a2dp_callback event=ESP_A2D_MEDIA_CTRL_ACK_EVT
W (62022) MAIN: a2dp_callback cmd=ESP_A2D_MEDIA_CTRL_START
W (62022) MAIN: a2dp_callback status=ESP_A2D_MEDIA_CTRL_ACK_FAILURE
```
## Other items if possible

* The Echo Dot can act as both a Bluetooth source (i.e. to an external speaker) as well as sink (i.e. from a laptop or phone, or an ESP32 source). I realize that the a2dp_source example is not expected to have this dual mode capability, but I wonder if the presence of it on the Echo Dot causes trouble for the ESP32 to connect as a source. I imagine that the Echo Dot figures out it is supposed to work as a sink when pairing with the ESP32 because of the Class of Device bits the ESP32 sends. Has anyone tested a2dp_source with devices that have this dual capability?
* As mentioned in the referenced forum post, I can connect without any issues to the Echo Dot from a laptop or phone. The Echo Dot is a current (v3) model.
* I get a sense from forum posts that others have had similar trouble sending data from an ESP source to Bluetooth speakers. One solution is to try other Bluetooth speakers until you find one that works, but it's hard to develop a product based on the ESP32 without confidence that it will work with a broad range of customer's Bluetooth speakers. It would be great to get this working. Let me know how I can help.

[attachments.zip](https://github.com/espressif/esp-idf/files/4484690/attachments.zip)

tonystuart, 2020-04-16:
Just verified that the simplified a2dp_source example also works with ESP-IDF v3.3.2 (as well as v3.2.2) so the changes that causes it to no longer work was introduced between v3.3.2 and v4.0.

tonystuart, 2020-04-17:
I'm attempting to do a git bisect to see if I can figure out which change causes it to no longer work.

However, it's hard to figure out the toolchain requirements (e.g. compiler and python script versions) for each of the intermediate commits.

I've tried various combinations of my existing v3.2.2, v3.3.2, v4.0 and v4.2 toolchains, but the make process complains about mismatches and unexpected version numbers (e.g. IDF_TOOLS_PATH must go with IDF_PATH, which must meet the Python requirements).

The brute force approach would be to install the toolchain for each of the 10-ish bisect points, but I wanted to make sure I wasn't overlooking an easier way?

tonystuart, 2020-04-17:
I made some progress with the git bisect and I'm down to about 30 commits. It looks like the program stops connecting to the Echo Dot somewhere between commit ab672a5ee (344 commits beyond v4.0-dev) and commit g603d29336 (375 commits beyond v4.0-dev).

tonystuart, 2020-04-20:
The problem commit is 603d293. The program connects to the Echo Dot in commit 8b2128ce0 (372 commits beyond v4.0-dev) and fails to connect in commit 603d293 (375 commits beyond v4.0-dev). The three commits between them were all part of branch that was merged into v4.0-dev on Apr 15, 2019. This branch contains the AVRCP Target APIs and included a lot of modifications (32 changed files with 2,445 additions and 468 deletions).

tonystuart, 2020-04-21:
Commit 603d293 adds AVRCP target support. Amongst other functions, this allows a remote A2DP sink (like the Echo Dot), acting as an AVRCP controller, to remotely control an A2DP source (like an ESP32), acting as an AVRCP target. An example of this would be to set the volume at the ESP32 source using the volume control buttons on the Echo Dot sink.

The ESP-IDF a2dp_sink example illustrates use of the AVRCP target support (i.e. the other way around), specifically to allow the A2DP sink to be controlled as an AVRCP target. In conjunction with this, the ESP-IDF a2dp_source example illustrates how to implement an AVRCP controller, but it also seems to have some AVRCP target code in it that looks like it is non-working (see https://esp32.com/viewtopic.php?t=14888).

Although highly speculative, my current hypothesis is that commit 603d293 changed the previous default "no-AVRCP-target" behavior in such a way that during the connection process, the Echo Dot (as an AVRCP controller) detects that the ESP32 A2DP source is an AVRCP target, but doesn't get back what it expects.

I assume that use of the AVRCP feature is optional and that an A2DP source application should work without using AVRCP. However to test this, I added a simple AVRCP target, and found that it was also not able to play to the Echo Dot.

```
$(esp_avrc_tg_init());
$(esp_avrc_tg_register_callback(avrcp_tg_callback));
esp_avrc_rn_evt_cap_mask_t evt_set = {0};
esp_avrc_rn_evt_bit_mask_operation(ESP_AVRC_BIT_MASK_OP_SET, &evt_set, ESP_AVRC_RN_VOLUME_CHANGE);
$(esp_avrc_tg_set_rn_evt_cap(&evt_set));
```

So to net things out: ESP-IDF Bluetooth A2DP source support stopped working for some classes of Bluetooth devices (like the Echo Dot), at commit 603d293,

Wth-Esp, 2020-05-07:
Hi @tonystuart 

Thanks for all your info!
We have an internal test case to test with the A2DP and AVRCP. We will look into the issues.

Thanks
Weitianhua

tonystuart, 2020-05-08:
Thanks! Let me know if you need any more info.

Wth-Esp, 2020-10-29:
@tonystuart Please update the latest branch to see if the issue still there.

Wth-Esp, 2021-01-21:
Hi @tonystuart 

I agree with your opinion that A2DP can be used without AVRCP and this is what we have done on:

release/v4.2 commit: 84cda60478d7a657be039339c706f4366ee2e861
release/v4.0 commit: 53c57cf2a3dbdd71957c45002371f1c0d94b9bd7

The strategy now is that you can use A2DP without AVRCP, but if you want to use AVRCP, you must initiate AVRCP before A2DP and must use A2DP at the same time. May this change help you.

Thanks

Alvin1Zhang, 2021-01-21:
Thanks for reporting, feel free to reopen.

closed 2021-01-21 by Alvin1Zhang

xiongweichao, 2021-01-29:
Hi @tonystuart 

We have tested the Echo Dot (3rd Gen), but no such issue happens.

Thanks