espressif/esp-idf #5136: a2dp_sink not working when set to DAC

drpain, 2020-04-16:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): not sure where to get this
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-1126-gd85d3d969
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: MSYS2 mingw32
- Using an IDE?: No
- Power Supply: USB

## Problem Description
I can't get the a2dp_sink to work in DAC mode. As soon as you try so play sound, it crashes. I also tried V3.3.2, but it doesn't include volume control which would be ideal.

//Detailed problem description goes here.

### Expected Behavior
It should play something through my amplifier circuit

### Actual Behavior
Device crashes

### Steps to reproduce
I followed the steps contained in the README, and set it to DAC mode, but as soon as you want to play sound it crashes

### Code to reproduce this issue
Using standard example 

## Debug Logs

```
W (21191) BT_APPL: new conn_srvc id:19, app_id:1
I (21201) BT_AV: A2DP audio state: Started
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4012283e  PS      : 0x00060530  A0      : 0x800d510c  A1      : 0x3ffd26c0
0x4012283e: i2s_write at C:/Users/Rudi/Documents/Github/esp-idf/components/driver/i2s.c:1023 (discriminator 2)

A2      : 0x00000000  A3      : 0x3ffd570c  A4      : 0x00000a00  A5      : 0x3ffd26f4
A6      : 0xffffffff  A7      : 0x00000000  A8      : 0x3ffcc098  A9      : 0x3ffd2670
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000002
A14     : 0xffffffff  A15     : 0x000006fe  SAR     : 0x00000000  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000018  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x4012283b:0x3ffd26c0 0x400d5109:0x3ffd26f0 0x40091725:0x3ffd2720
0x4012283b: i2s_write at C:/Users/Rudi/Documents/Github/esp-idf/components/driver/i2s.c:1023 (discriminator 2)

0x400d5109: bt_i2s_task_handler at c:\users\rudi\documents\github\esp-idf\examples\bluetooth\bluedroid\classic_bt\a2dp_sink\build/../m

ain/bt_app_core.c:128

0x40091725: vPortTaskWrapper at C:/Users/Rudi/Documents/Github/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: fc39360f66176ae6

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7188
load:0x40078000,len:13696
load:0x40080400,len:4000
0x40080400: _init at ??:?
```

drpain, 2020-04-16:
@blueMoodBHD Any chance you can have a look as the last to work on this section?

koobest, 2020-04-17:
Hi, 
Can you check the reuturn value of `i2s_driver_install`  ( https://github.com/espressif/esp-idf/blob/7d0f1536e468736766b0977f2902e1321426a1ed/examples/bluetooth/bluedroid/classic_bt/a2dp_sink/main/main.c#L72)?

drpain, 2020-04-17:
@koobest My windows setup stopped working, so I can't get you that info. As a side note I was able to get this to work as expected using the Linux (Docker) setup. Which doesn't make sense to me. It should have both been off release4.0

Not sure where to go from here. Since it's working through the docker setup

linuxhuskarl, 2020-05-08:
Hi, I am having exactly the same problem. Can I ask to reopen this issue?

- Development Kit: Wemos LOLIN32
- Kit version: v1.0.0
- Module or chip used: ESP32-WROOM-32
- IDF version: v4.3 current master (synced today)
- Build System: idf.py
- Compiler version: don't know
- Operating System: Kubuntu 20.04 
- Using an IDE?: Yes, VSCode with Espressif IDF v0.2.1
- Power Supply: USB

Log during boot shows this error:
```
E (609) I2S: /home/zeffire/esp-idf/components/driver/i2s.c:838 (i2s_param_config):I2S communication format invalid.
E (609) I2S: I2S param configure error
```
After connecting to an android another error comes up:
```
I (1329) BT_AV: event: 10
E (510199) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (510199) BT_AV: A2DP connection state: Connecting, [18:01:f1:4d:6f:63]
I (510229) BT_AV: A2DP audio stream configuration, codec type 0
E (510229) I2S: Not initialized yet
I (510229) BT_AV: Configure audio player 11-15-2-35
I (510229) BT_AV: Audio player configured, sample rate=48000
W (510259) BT_APPL: new conn_srvc id:19, app_id:0
I (510259) BT_AV: A2DP connection state: Connected, [18:01:f1:4d:6f:63]
```
Finally, when I try to play any audio i get exactly the same error as @drpain in Debug Logs.

@koobest `i2s_driver_install(0, &i2s_config, 0, NULL);` returns code 258 (0x102)

koobest, 2020-05-09:
hi, @linuxhuskarl 
You can set the `communication_format`  to ` I2S_COMM_FORMAT_PCM`
We have a MR to fix this issue, will be merged soon.

Wth-Esp, 2020-05-09:
Hi, @koobest 

Would you please add the link or make a patch for @drpain  and @linuxhuskarl to test first?

Thanks
Weitianhua

linuxhuskarl, 2020-05-09:
@koobest I can confirm that this solution worked - no errors whatsoever. Thanks!

closed 2020-06-12 by igrr