espressif/esp-idf #5137: ESP32-CAM Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment).

sky5media, 2020-04-16:
## Environment

- Development Board:      ESP32-CAM
- Module or chip used:  ESP32-S
- IDF version: v4.0
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         [USB]

## Problem Description

I am trying to build latest micropython for ESP32-CAM. Board has 8M PSRAM chip (IPUS IPS1704L-SQ)

When trying to allocate 160 KB buffer with:
`heap_caps_calloc(_fb2->size, 1, MALLOC_CAP_SPIRAM)`
I am getting Guru Meditation Error
`Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment). Exception was unhandled.`

I tried both CONFIG_SPIRAM_USE_CAPS_ALLOC=y and CONFIG_SPIRAM_USE_MALLOC=y
same result

Some debugging info:

```
heap_caps_get_free_size(MALLOC_CAP_DEFAULT) -> -1431534259 (negative)
heap_caps_get_free_size(MALLOC_CAP_INTERNAL) -> 172184
heap_caps_get_free_size(MALLOC_CAP_SPIRAM) -> -1431655771 (negative)
heap_caps_get_free_size(MALLOC_CAP_8BIT) -> -1431501531 (negative)
heap_caps_get_free_size(MALLOC_CAP_32BIT) -> -1431483587 (negative)
heap_caps_get_free_size(MALLOC_CAP_DMA) -> 154240

esp_get_free_heap_size() -> 2863433037
esp_spiram_get_size() -> 8388608
esp_himem_get_free_size() -> 4325376
esp_himem_get_phys_size() -> 4325376
esp_himem_reserved_area_size() -> 131072
```

When dumping or trying to get info about heap with MALLOC_CAP_SPIRAM getting same Guru Meditation Error after which device restarts

Same behavior on 2 different boards of the same type

Board images:
[image]
[image]

### Expected Behavior

Buffer should be allocated

### Actual Behavior

`Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment). Exception was unhandled.`

### Code to reproduce this issue

```

    ESP_LOGI(TAG, "Allocating %u frame buffers (%d KB total)", count, (s_state->fb_size * count) / 1024);

    camera_fb_int_t * _fb = NULL, * _fb1 = NULL, * _fb2 = NULL;
    for(size_t i = 0; i < count; i++) {
        _fb2 = (camera_fb_int_t *)malloc(sizeof(camera_fb_int_t));
        if(!_fb2) {
            goto fail;
        }
        memset(_fb2, 0, sizeof(camera_fb_int_t));
        _fb2->size = s_state->fb_size;
        _fb2->buf = (uint8_t*) calloc(_fb2->size, 1);
        if(!_fb2->buf) {
            ESP_LOGI(TAG, "Allocating %d KB frame buffer in PSRAM", s_state->fb_size/1024);
            _fb2->buf = (uint8_t*) heap_caps_calloc(_fb2->size, 1, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
        } else {
            ESP_LOGI(TAG, "Allocating %d KB frame buffer in OnBoard RAM", s_state->fb_size/1024);
        }
        if(!_fb2->buf) {
            free(_fb2);
            ESP_LOGE(TAG, "Allocating %d KB frame buffer Failed", s_state->fb_size/1024);
            goto fail;
        }
      
```

## Debug Logs

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5200
ho 0 tail 12 room 4
load:0x40078000,len:14032
load:0x40080400,len:3800
entry 0x4008063c
I (579) psram: This chip is ESP32-D0WD
I (579) spiram: Found 64MBit SPI RAM device
I (579) spiram: SPI RAM mode: flash 40m sram 40m
I (582) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (589) cpu_start: Pro cpu up.
I (593) cpu_start: Application information:
I (598) cpu_start: Compile time:     Apr 16 2020 21:10:15
I (604) cpu_start: ELF file SHA256:  0000000000000000...
I (610) cpu_start: ESP-IDF:          v4.0
I (614) cpu_start: Starting app cpu, entry point is 0x40083484
I (606) cpu_start: App cpu up.
I (1463) spiram: SPI SRAM memory test OK
I (1464) heap_init: Initializing. RAM available for dynamic allocation:
I (1464) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1470) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1476) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1482) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (1489) heap_init: At 3FFCCD18 len 000132E8 (76 KiB): DRAM
I (1495) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1501) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1508) heap_init: At 4009B9C4 len 0000463C (17 KiB): IRAM
I (1514) cpu_start: Pro cpu start user code
I (1519) spiram: Adding pool of 3968K of external SPI memory to heap allocator
I (1551) esp_himem: Initialized. Using last 4 32KB address blocks for bank switching on 4224 KB of physical memory.
I (1552) spi_flash: detected chip: generic
I (1556) spi_flash: flash io: dio
I (1561) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1571) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
MicroPython v1.12-263-g8d34344dc-dirty on 2020-04-16; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> import camera
>>> camera.init()
I (1060751) sccb: pin_sda 26 pin_scl 27

I (1060751) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1060881) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1060881) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1060891) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1060901) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1060911) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1060921) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1060931) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1060941) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1060951) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1060961) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1060971) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1060981) camera: Allocating 1 frame buffers (160 KB total)

Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment). Exception was unhandled.
Core 1 register dump:
PC      : 0x40094aee  PS      : 0x00060133  A0      : 0x80096a82  A1      : 0x3ffd8820
A2      : 0x66555559  A3      : 0x00060520  A4      : 0x00000001  A5      : 0x0000abab
A6      : 0xb33fffff  A7      : 0x00000001  A8      : 0x0000cdcd  A9      : 0x3ffd8920
A10     : 0x0000abab  A11     : 0x00000000  A12     : 0x00000003  A13     : 0x3ffc8534
A14     : 0x3ffba7ec  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000009
EXCVADDR: 0x66555559  LBEG    : 0x400973ad  LEND    : 0x400973bd  LCOUNT  : 0xfffffff8

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40094aeb:0x3ffd8820 0x40096a7f:0x3ffd8850 0x40096ab4:0x3ffd8870 0x40085c64:0x3ffd88a0 0x40085c9a:0x3ffd88c0 0x40098a92:0x3ffd88e0 0x40098ab4:0x3ffd8900 0x4016ae0d:0x3ffd8920 0x4016b285:0x3ffd89d0 0x400fbb07:0x3ffd8a00 0x400e3d6b:0x3ffd8a20 0x400dfdd1:0x3ffd8a40 0x400dff0a:0x3ffd8a60 0x400ee063:0x3ffd8a80 0x400e3f40:0x3ffd8b20 0x400dfdd1:0x3ffd8b70 0x400dfdfa:0x3ffd8b90 0x40106468:0x3ffd8bb0 0x4010668a:0x3ffd8c40 0x400f7224:0x3ffd8c80 0x40093df9:0x3ffd8cb0

Rebooting...

```

## Other items if possible

sdkconfig:

```
CONFIG_ESP32_SPIRAM_SUPPORT=y
CONFIG_SPIRAM_BOOT_INIT=y

CONFIG_SPIRAM_USE_MEMMAP=
CONFIG_SPIRAM_USE_CAPS_ALLOC=
CONFIG_SPIRAM_USE_MALLOC=y

CONFIG_SPIRAM_TYPE_ESPPSRAM64=y
CONFIG_SPIRAM_SIZE=-1
CONFIG_SPIRAM_SPEED_40M=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y
CONFIG_SPIRAM_BANKSWITCH_ENABLE=y
CONFIG_SPIRAM_BANKSWITCH_RESERVE=4
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=

CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y 
CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0=n
CONFIG_ESP32_XTAL_FREQ_AUTO=y 

# v3.3-only (renamed in 4.0)
CONFIG_SPIRAM_SUPPORT=y

# FreeRTOS
CONFIG_SUPPORT_STATIC_ALLOCATION=y
CONFIG_ENABLE_STATIC_TASK_CLEAN_UP_HOOK=y

# UDP
CONFIG_PPP_SUPPORT=y
CONFIG_PPP_PAP_SUPPORT=y
CONFIG_PPP_CHAP_SUPPORT=y 
```

I tried either of CONFIG_SPIRAM_USE_MEMMAP CONFIG_SPIRAM_USE_CAPS_ALLOC CONFIG_SPIRAM_USE_MALLOC
But none of the setting increase heap size

uraich, 2021-07-25:
Same problem here. Did you find a solution in the meantime?

lucaskdc, 2023-03-04:
Hello there, I have found a similar problem using many heap_caps_* functions when `CONFIG_SPIRAM_USE_CAPS_ALLOC=y` after calling esp_himem_map(...) and esp_himem_unmap(...).

Probably this is related to this issue also, but I am using currently a ESP32 WROVER
When i produce reproductible evidences i will create a pull request with the bugfix then you can test it.

It seems esp_psram_init() miscalculates the correct heap blocks when himem is enabled and reserves some memory.

AxelLin, 2023-10-19:
@lucaskdc Which esp-idf version are you using?