espressif/esp-idf #5138: "esp_ble_gattc_close" disconnects from all clients

fbgencer, 2020-04-16:
If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v3]
- Module or chip used:  [ESP32-WROOM-322]
- IDF version :
    // v4.2-dev
- Build System:  [idf.py]
- Compiler version:
    // 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
I have two devices which I can connect them succesfully via `esp_ble_gattc_app_register` and `esp_ble_gattc_open` functions. Also when callback returns conn_id and gattc_if values, clients update their conn_id and gattc_if. 

For disconnection I was using `esp_ble_gattc_close` which takes two inputs gatt_if and conn_id, which succesfully disconnects from one device. However when I have two connections, the esp_ble_gattc_close request returns me the following warning on my terminal :
```
W (5743) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
W (5753) BT_APPL: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0016
```

First of all, clients have conn_id: 0 and conn_id:1 not 3 and 4 as printed in the warning. Printed gattc_if values match with previosly returned gattc_if (they were taken when `ESP_GATTC_REG_EVT` event is invoked). Also reason is also checkhed which is `ESP_GATT_CONN_TERMINATE_LOCAL_HOST`, and true because I requested this event. 

### Expected Behavior
Disconnection should occur for a requested client only, esp should not close all connections.

### Actual Behavior
ESP Disconnects from all clients instead of the expected client.

WCCWCC, 2020-08-18:
Hi @fbgencer ,

esp_ble_gattc_close, from your description, there are certain problems with the behavior of this api.

In order to solve this problem faster, can you provide the code to use this API esp_ble_gattc_close?

I will also reproduce this problem locally.

zhp0406, 2023-11-17:
@fbgencer 
Has the issue been solved?