espressif/esp-idf #5139: idf.py encrypted-flash monitor problem with spiffs

AxelLin, 2020-04-17:
## Environment

- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.2-dev-1126-gd85d3d969ff4
- Build System: idf.py  
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Using an IDE?: NO
- Power Supply: USB

## Problem Description

I'm following the flash-encryption document to download the firmware. (in developement mode for now).

1. enable flash encryption and do idf.py flash monitor
   It looks fine and flash encryption is enabled.

2. According to the docuemtn, I can use below commands to update modified plaintext application image.
   $ idf.py encrypted-app-flash monitor
   or
   $ idf.py encrypted-flash monitor

   The encrypted-app-flash works fine but my application uses a spiffs partition and I want to update the spiffs image as well.
   I have below line in my CMakeLists.txt
   spiffs_create_partition_image(res ${RES_SRC_DIR} FLASH_IN_PROJECT)
   i.e. When flash encryption is disabled, idf.py flash monitor will update both application and spiffs images.

   When I use encrypted-flash, I would expect both application firmware and spiffs images will be updated.
   However, in my test, after encrypted-flash my application fails to mount the spiffs.
   It seems something wrong with encrypted-flash for spiffs.
   (I'm wondering if the encrypted-flash also encrypt the spiffs image? the spiffs should be not encrypted.)

### Expected Behavior
   The application should be able to mount the spiffs image after encrypted-flash.

AxelLin, 2020-04-17:
BTW, I'm confused by the description of "encrypted-flash".

By default only Bootloader, Partition Table and All “app” type partitions will be encrypted.

But the document says:
If *all* partitions needs to be updated in encrypted format, it can be done as
$ idf.py encrypted-flash monitor

So I'm not sure if the "all partitions" in encrypted-flash means really *all* partitons or
only Bootloader, Partition Table and All “app” type partitions.

AxelLin, 2020-04-29:
@Alvin1Zhang 

No response so far, any one can help?
It's either a bug or I misunderstand the meaning of "idf.py encrypted-flash".
Is "idf.py encrypted-flash" encrypted all partitions or only the partitions marked as encrypted?

When I have spiffs_create_partition_image(res ${RES_SRC_DIR} FLASH_IN_PROJECT) in my CMakeLists.txt.
"idf.py flash" will update both application firmware and spiffs image.
I thought "idf.py encrypted-flash" can automatically recognize spiffs does not need encryption.
However, it seems not true.

I need to manually use esptool.py without "--encrypt" parameter to update my spiffs.bin again.

projectgus, 2020-04-29:
Hi @AxelLin,

Sorry noone got back to you earlier.

As you've noticed, this is a current limitation of the `encrypted-flash` target - it flashes all binaries encrypted. We plan to update the build system to automatically know files should be encrypted, but this is not currently supported.

Currently you can work around it by running targets to specifically flash different things, for example:

```
idf.py -p PORT res-flash encrypted-app-flash monitor
```

Note: using `res-flash` here because the `spiffs_create_partition_image` name is `res`.

Unlike `encrypted-flash`, this will run esptool.py multiple times (currently there's no way to tell esptool to flash some files encrypted and some unencrypted), but it will build and flash each part in sequence.

You can add `encrypted-bootloader-flash` or `encrypted-partition_table-flash` as well if you need them. Use `idf.py help` to see the full list of build & flash targets for your project.

AxelLin, 2020-04-29:
Hi @projectgus ,

> Hi @AxelLin,
> 
> Sorry noone got back to you earlier.
> 
> As you've noticed, this is a current limitation of the `encrypted-flash` target - it flashes all binaries encrypted. We plan to update the build system to automatically know which files should be encrypted, but this is not currently supported.
> 
> Currently you can work around it by running targets to specifically flash different things, for example:
> 
> ```
> idf.py -p PORT res-flash encrypted-app-flash monitor
> ```
> 
> Note: using `res-flash` here because the `spiffs_create_partition_image` name is `res`.

Ok, above command works.

> Unlike `encrypted-flash`, this will run esptool.py multiple times (currently there's no way to tell esptool to flash some files encrypted and some unencrypted), but it will build and flash each part [...]

Running idf.py --help does not show `encrypted-bootloader-flash` or `encrypted-partition_table-flash`. (v4.2-dev-1266-g2dc4bfe7822f)
I got:
Commands:
  all                    Aliases: build. Build the project.
  app                    Build only the app.
  app-flash              Flash the app only.
  bootloader             Build only bootloader.
  bootloader-flash       Flash bootloader only.
  clean                  Delete build output files from the build directory.
  confserver             Run JSON configuration server.
  efuse_common_table     Genereate C-source for IDF's eFuse fields.
  efuse_custom_table     Genereate C-source for user's eFuse fields.
  encrypted-app-flash    Flash the encrypted app only.
  encrypted-flash        Flash the encrypted project.
  erase_flash            Erase entire flash chip.
  erase_otadata          Erase otadata partition.
  flash                  Flash the project.
  fullclean              Delete the entire build directory contents.
  gdb                    Run the GDB.
  gdbgui                 GDB UI in default browser.
  gdbtui                 GDB TUI mode.
  menuconfig             Run "menuconfig" project configuration tool.
  monitor                Display serial output.
  openocd                Run openocd from current path
  partition_table        Build only partition table.
  partition_table-flash  Flash partition table only.
  post_debug             Utility target to read the output of async debug action and stop them.
  read_otadata           Read otadata partition.
  reconfigure            Re-run CMake.
  set-target             Set the chip target to build.
  show_efuse_table       Print eFuse table.
  size                   Print basic size information about the app.
  size-components        Print per-component size information.
  size-files             Print per-source-file size information.

projectgus, 2020-04-29:
> Running idf.py --help does not show encrypted-bootloader-flash or encrypted-partition_table-flash. (v4.2-dev-1266-g2dc4bfe7822f)

It's a bit confusing, there are two commands: running `idf.py --help` prints the list of "high level" commands, which you can see above. Running `idf.py help` should print every build target that's supported by the build system.

Will see about changing this second target name to something else so it's not as confusing.

AxelLin, 2020-06-18:
Hi @projectgus 

I'm testing flash encryption now.

When I update firmware with "idf.py encrypted-app-flash monitor", it shows below messages:
I'm wondering if you hay any idea about this issue:

I (523) cpu_start: ESP-IDF:          v4.2-dev-1905-g625bd5eb1806
I (530) cpu_start: Starting app cpu, entry point is 0x4008195c
0x4008195c: call_start_cpu1 at /home/axel/esp/esp-idf/components/esp32/cpu_start.c:286

D (536) efuse: coding scheme 0
/home/axel/esp/esp-idf/components/freertos/queue.c:1447 (xQueueGenericReceive)- assert failed!
Re-enable cpu cache.

abort() was called at PC 0x4008ae80 on core 0
0x4008ae80: xQueueGenericReceive at /home/axel/esp/esp-idf/components/freertos/queue.c:1448 (discriminator 3)

Backtrace:0x4008784a:0x3ffe39e0 0x40087e2d:0x3ffe3a00 0x4008eb3e:0x3ffe3a20 0x4008ae80:0x3ffe3a90 0x4008bb53:0x3ffe3ad0 0x4019c3bd:0x3ffe3af0 0x4008bae9:0x3ffe3b20 0x400ef64e:0x3ffe3b70 0x400ef4f4:0x3ffe3bc0 0x400ef516:0x3ffe3be0 0x400ef845:0x3ffe3c10 0x400ef887:0x3ffe3c30 0x40081aea:0x3ffe3c50 0x40079584:0x3ffe3c90 |<-CORRUPTED
0x4008784a: panic_abort at /home/axel/esp/esp-idf/components/esp_system/panic.c:341

0x40087e2d: esp_system_abort at /home/axel/esp/esp-idf/components/esp_system/system_api.c:100

0x4008eb3e: abort at /home/axel/esp/esp-idf/components/newlib/abort.c:46

0x4008ae80: xQueueGenericReceive at /home/axel/esp/esp-idf/components/freertos/queue.c:1448 (discriminator 3)

0x4008bb53: esp_log_impl_lock_timeout at /home/axel/esp/esp-idf/components/log/log_freertos.c:45

0x4019c3bd: esp_log_writev at /home/axel/esp/esp-idf/components/log/log.c:170

0x4008bae9: esp_log_write at /home/axel/esp/esp-idf/components/log/log.c:199

0x400ef64e: esp_efuse_utility_process at /home/axel/esp/esp-idf/components/efuse/src/esp_efuse_utility.c:71 (discriminator 6)

0x400ef4f4: esp_efuse_read_field_blob at /home/axel/esp/esp-idf/components/efuse/src/esp_efuse_api.c:51

0x400ef516: esp_efuse_read_field_bit at /home/axel/esp/esp-idf/components/efuse/src/esp_efuse_api.c:60

0x400ef845: esp_get_flash_encryption_mode at /home/axel/esp/esp-idf/components/bootloader_support/src/flash_encrypt.c:97

0x400ef887: esp_flash_encryption_init_checks at /home/axel/esp/esp-idf/components/bootloader_support/src/flash_encrypt.c:62

0x40081aea: call_start_cpu0 at /home/axel/esp/esp-idf/components/esp32/cpu_start.c:223 (discriminator 1)

ELF file SHA256: 15590bb5b49b6ca4

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6020
load:0x40078000,len:13524
ho 0 tail 12 room 4
load:0x40080400,len:3768
0x40080400: _init at ??:?

entry 0x40080648

KonstantinKondrashov, 2020-06-18:
Hi @AxelLin!
It has fixed but not put in GitHub yet (a6ac0e8a).  ESP_LOG[DEBUG] in efuse component leads to the issue, you can use a patch in https://github.com/espressif/esp-idf/issues/5343 until the fix will not be synced with GitHub repo.
Thanks.

closed 2020-08-04 by AxelLin