espressif/esp-idf #5140: bt spp connection shift system time

moggiozzi, 2020-04-17:
## Environment
- Development Kit:      ESP32 DEVKITV1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
v4.2-dev-792-g6330b3345
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows 7
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
I run bt_spp_acceptor with little code to read/write system time. After connect to device by BT (SPP) system time shifted. If I set system time with 2020 year then time still little shift of about 10 minutes (see test2 output).

### Expected Behavior
System time not shifted by establish BT connection

### Actual Behavior
System time shifted after establish BT connection

### Steps to reproduce
run sketch code and connect by BT

### Code to reproduce this issue

```cpp
void test1(void *arg){
	struct timeval tv;
	while(1){
		gettimeofday(&tv, NULL);
		ESP_LOGI("TEST", "Get time: sec = %u", (uint32_t)tv.tv_sec);
		vTaskDelay(pdMS_TO_TICKS(5000));
	}
}
void test2(void *arg){
//	struct tm tm;
//	tm.tm_year = 2020 - 1900;
//	tm.tm_mon = 3;//0..11
//	tm.tm_mday = 17;
//	tm.tm_hour = 9;
//	tm.tm_min = 19;
//	tm.tm_sec = 21;
//	time_t t = mktime(&tm);
//	//Set time: Fri Apr 17 09:19:21 2020 // sec = 1587115161
	struct timeval tv = { .tv_sec = 1587115161, .tv_usec = 0 };
	ESP_LOGI("TEST", "Set time: sec = %u", (uint32_t)tv.tv_sec);
	settimeofday(&tv, NULL);
	while(1){
		gettimeofday(&tv, NULL);
		ESP_LOGI("TEST", "Get time: sec = %u", (uint32_t)tv.tv_sec);
		vTaskDelay(pdMS_TO_TICKS(5000));
	}
}
void run_test(void){
    xTaskCreate(test1, "test_task", 2*1024, NULL, 10, NULL);
}
void app_main(void)
{
...
  run_test();
}
```
## Debug Logs

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[... 97 lines cut ...]
I (491) spi_flash: flash io: dio
W (491) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (501) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (532) BTDM_INIT: BT controller compile version [ab2cbc7]
I (532) system_api: Base MAC address is not set
I (532) system_api: read default base MAC address from EFUSE
I (632) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1212) SPP_ACCEPTOR_DEMO: ESP_SPP_INIT_EVT
I (1232) SPP_ACCEPTOR_DEMO: event: 10
I (1232) SPP_ACCEPTOR_DEMO: ESP_SPP_START_EVT
I (1232) TEST: Set time: sec = 1587115161
I (1232) TEST: Get time: sec = 1587115161
I (6242) TEST: Get time: sec = 1587115166
I (11242) TEST: Get time: sec = 1587115171
ASSERT_WARN(1 9), in lc_task.c at line 5054W (11442) BT_APPL: new conn_srvc id:26, app_id:255
I (11452) SPP_ACCEPTOR_DEMO: ESP_SPP_SRV_OPEN_EVT
ASSERT_WARN(1 9), in lc_task.c at line 5054ASSERT_WARN(1 9), in lc_task.c at line 5054ASSERT_WARN(1 9), in lc_task.c at line 5054I (16242) TEST: Get time: sec = 1587116059
ASSERT_WARN(1 9), in lc_task.c at line 5054I (21242) TEST: Get time: sec = 1587116064
W (21642) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Closed (res: 19)
I (21642) SPP_ACCEPTOR_DEMO: ESP_SPP_CLOSE_EVT
W (24142) BT_RFCOMM: rfc_find_lcid_mcb LCID reused LCID:0x41 current:0x0
W (24142) BT_RFCOMM: RFCOMM_DisconnectInd LCID:0x41
I (26242) TEST: Get time: sec = 1587116069
```

Wth-Esp, 2020-05-07:
Hi, @moggiozzi 

This issue may be caused by the `gettimeofday` function which is used in the demo to calculate transportation speed.

Please apply this patch to your IDF. I add your test code above.

[time_fix_tempo.zip](https://github.com/espressif/esp-idf/files/4592775/time_fix_tempo.zip)

Thanks
Weitianhua

Wth-Esp, 2020-05-07:
I tried it and it works. Here is the log:
```
I (521290) TEST: Get time: sec = 520
I (521790) SPP_ACCEPTOR_DEMO: speed(512849793 ~ 515850652): 1.720787 kbit/s
I (524840) SPP_ACCEPTOR_DEMO: speed(515850652 ~ 518854929): 1.768918 kbit/s
I (526290) TEST: Get time: sec = 525
I (527880) SPP_ACCEPTOR_DEMO: speed(518854929 ~ 521855597): 1.781603 kbit/s
I (530920) SPP_ACCEPTOR_DEMO: speed(521855597 ~ 524857446): 1.796733 kbit/s
I (531290) TEST: Get time: sec = 530
I (533970) SPP_ACCEPTOR_DEMO: speed(524857446 ~ 527860068): 1.798908 kbit/s
I (536290) TEST: Get time: sec = 535
I (537010) SPP_ACCEPTOR_DEMO: speed(527860068 ~ 530861442): 1.736325 kbit/s
I (540050) SPP_ACCEPTOR_DEMO: speed(530861442 ~ 533863576): 1.783371 kbit/s
I (541290) TEST: Get time: sec = 540
I (543100) SPP_ACCEPTOR_DEMO: speed(533863576 ~ 536863647): 1.789878 kbit/s
I (546160) SPP_ACCEPTOR_DEMO: speed(536863647 ~ 539883171): 1.731134 kbit/s
I (546290) TEST: Get time: sec = 545
I (549200) SPP_ACCEPTOR_DEMO: speed(539883171 ~ 542883446): 1.752799 kbit/s
I (551290) TEST: Get time: sec = 550
I (552240) SPP_ACCEPTOR_DEMO: speed(542883446 ~ 545886679): 1.740524 kbit/s
I (555290) SPP_ACCEPTOR_DEMO: speed(545886679 ~ 548887686): 1.786680 kbit/s
I (556290) TEST: Get time: sec = 555
I (558330) SPP_ACCEPTOR_DEMO: speed(548887686 ~ 551888491): 1.797358 kbit/s
I (561290) TEST: Get time: sec = 560
I (561370) SPP_ACCEPTOR_DEMO: speed(551888491 ~ 554890665): 1.722672 kbit/s
I (564420) SPP_ACCEPTOR_DEMO: speed(554890665 ~ 557891618): 1.752403 kbit/s
I (566290) TEST: Get time: sec = 565
I (567460) SPP_ACCEPTOR_DEMO: speed(557891618 ~ 560891744): 1.787205 kbit/s
I (570500) SPP_ACCEPTOR_DEMO: speed(560891744 ~ 563894285): 1.777854 kbit/s
I (571290) TEST: Get time: sec = 570
I (573550) SPP_ACCEPTOR_DEMO: speed(563894285 ~ 566895496): 1.731141 kbit/s
I (576290) TEST: Get time: sec = 575
I (576590) SPP_ACCEPTOR_DEMO: speed(566895496 ~ 569899190): 1.795629 kbit/s
I (579640) SPP_ACCEPTOR_DEMO: speed(569899190 ~ 572901568): 1.801692 kbit/s
```

moggiozzi, 2020-05-07:
I think that problem is NOT in user code.
Let me remind you that time shift occurs in the moment of BT connection establishing (or after using settimeofday), not during data transfer.
Log out from project with your time_fix_tempo.patch (still with a time shift):
```
I (1222) TEST: Get time: sec = 0
I (6232) TEST: Get time: sec = 5
W (10072) BT_APPL: new conn_srvc id:26, app_id:255
I (10082) SPP_ACCEPTOR_DEMO: ESP_SPP_SRV_OPEN_EVT
I (11232) TEST: Get time: sec = 10
ASSERT_WARN(1 9), in lc_task.c at line 5054ASSERT_WARN(1 9), in lc_task.c at line 5054I (16232) TEST
: Get time: sec = 4154501480
ASSERT_WARN(1 9), in lc_task.c at line 5054I (17632) SPP_ACCEPTOR_DEMO: speed(10098969 ~ 25758): 0.0
00000 kbit/s
ASSERT_WARN(1 9), in lc_task.c at line 5054I (21232) TEST: Get time: sec = 4154501485
ASSERT_WARN(1 9), in lc_task.c at line 5054ASSERT_WARN(1 9), in lc_task.c at line 5054I (26232) TEST
: Get time: sec = 4154501490
```
Maybe it related with "ASSERT_WARN(1 9), in lc_task.c at line 5054"?

Wth-Esp, 2020-05-08:
Hi @moggiozzi 

I use the bt_spp_acceptor demo and do the change, as the patch attached, the system's time does not change while the procedure of connection initiation. 

And no log like `ASSERT_WARN(1,9) in lc_task.c at line 5054` happens.

Maybe you could provide me your source code to debug?

Thanks
Weitianhua

moggiozzi, 2020-05-08:
I add my code to: https://github.com/moggiozzi/bt_spp_acceptor_timeShift
Maybe
1) problem related with PC BT driver?
[image]
2) Or config -> ESP32-specific -> Timers used for gettimeofday function

Wth-Esp, 2020-05-09:
Hi, @moggiozzi 

Maybe it's the driver's bug.

I test your demo on ESP32-WROVER-B under Linux with another WROVER-B running bt_spp_initiator.

**The issue does not reproduce.**

Here is the log:
```
I (485) spi_flash: detected chip: generic
I (486) spi_flash: flash io: dio
W (486) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (496) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (530) BTDM_INIT: BT controller compile version [d9cc6b1]
I (530) system_api: Base MAC address is not set
I (530) system_api: read default base MAC address from EFUSE
I (650) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1210) SPP_ACCEPTOR_DEMO: ESP_SPP_INIT_EVT
I (1220) SPP_ACCEPTOR_DEMO: event: 10
I (1230) SPP_ACCEPTOR_DEMO: ESP_SPP_START_EVT
I (1230) TEST: Get time: sec = 0
I (2260) SPP_ACCEPTOR_DEMO: ESP_BT_GAP_CFM_REQ_EVT Please compare the numeric value: 934855
I (2340) SPP_ACCEPTOR_DEMO: authentication success: ESP_SPP_INITIATOR
I (2340) SPP_ACCEPTOR_DEMO: 30 ae a4 cb b5 2e 
W (2450) BT_APPL: new conn_srvc id:26, app_id:255
I (2450) SPP_ACCEPTOR_DEMO: ESP_SPP_SRV_OPEN_EVT
I (5470) SPP_ACCEPTOR_DEMO: speed(2475531 ~ 5479185): 1.700729 kbit/s
I (6230) TEST: Get time: sec = 5
I (8500) SPP_ACCEPTOR_DEMO: speed(5479185 ~ 8493036): 1.768555 kbit/s
I (11230) TEST: Get time: sec = 10
I (11520) SPP_ACCEPTOR_DEMO: speed(8493036 ~ 11493154): 1.734412 kbit/s
I (14540) SPP_ACCEPTOR_DEMO: speed(11493154 ~ 14493540): 1.773852 kbit/s
I (16230) TEST: Get time: sec = 15
I (17570) SPP_ACCEPTOR_DEMO: speed(14493540 ~ 17505885): 1.814135 kbit/s
I (20600) SPP_ACCEPTOR_DEMO: speed(17505885 ~ 20507191): 1.781225 kbit/s
I (21230) TEST: Get time: sec = 20
I (23620) SPP_ACCEPTOR_DEMO: speed(20507191 ~ 23507883): 1.834377 kbit/s
I (26230) TEST: Get time: sec = 25
I (26650) SPP_ACCEPTOR_DEMO: speed(23507883 ~ 26509903): 1.846756 kbit/s
I (29680) SPP_ACCEPTOR_DEMO: speed(26509903 ~ 29510761): 1.818440 kbit/s
I (31230) TEST: Get time: sec = 30
```
Thanks
Weitianhua

moggiozzi, 2020-05-09:
Hi, @Wth-Esp.
When "ASSERT_WARN(1 9), in lc_task.c at line 5054" occurs (what is its cause)?
The source of lc_task.c is not available.

Wth-Esp, 2020-05-11:
Hi, @moggiozzi 

[bt_spp_acceptor_demo.zip](https://github.com/espressif/esp-idf/files/4607387/bt_spp_acceptor_demo.zip)

Thanks
Weitianhua

moggiozzi, 2020-05-11:
> 
> 
> Hi, @moggiozzi
> 
> [bt_spp_acceptor_demo.zip](https://github.com/espressif/esp-idf/files/4607387/bt_spp_acceptor_demo.zip)
> 
> Thanks
> Weitianhua

I run your binary. The time shift still present.

moggiozzi, 2020-05-11:
1) Time shift not occurs if set "Component config -> ESP32-specific -> Timers used for gettimeofday function -> High-resolution timer" (but this version reset system time after each restart)
2) I would like to understand the reason of "ASSERT_WARN(1 9), in lc_task.c at line 5054"

Wth-Esp, 2020-05-11:
Hi，@moggiozzi 

1. I use the default SDK config of SPP Acceptor demo, time shifts like yours not present.
2. If the connection state does not correspond to the `LMP opcode` then the ASSERT_WARN appears.

Thanks
Weitianhua

moggiozzi, 2020-05-11:
I am sorry for offtopic, but how ASSERT_WARN messages can be turned off?
I try set:
Compiler options -> Assertion level -> Disabled
Component config -> Bluetooth -> Bluedroid Options -> BT DEBUG LOG LEVEL to ERROR

Wth-Esp, 2020-05-12:
Hi @moggiozzi 

It's impossible to disable this ASSERT_WARN unless I make a new lib for you. But I don't think it's necessary, because ASSERT_WARN is an important way to diagnose problems.

I cannot reproduce your issue, the only difference between us is the environment, so please double check your driver version of something else or so some test in Linux (Unbuntu) to confirm the issue.

Thanks
Weitianhua

moggiozzi, 2020-05-12:
I have a project which used this uart and extra output on it is not desirable. The other chip pins used for other peripherals.

moggiozzi, 2020-05-15:
I test project on Win10 with other BT adapter and problem not occurs.
[image]

adamkovari, 2020-11-24:
Hi @moggiozzi , can you solved this issue?
thanks

closed 2020-11-25 by Alvin1Zhang