espressif/esp-idf #5141: Can't find configuration option in sdkconfig.h

romocitto88, 2020-04-17:
## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32D
- Operating System:     Windows
- Using an IDE?: PlatformIO

## Problem Description

I'm tring to use the ble usart example, and I want to save power consumption. 
From Espressif API is possible to set CONFIG_BTDM_CONTROLLER_MODEM_SLEEP but I can't found any define in the file sdkconfig.h. Someone can explaine me how to do?

Thank you.

dobairoland, 2020-04-17:
The configuration value was recently renamed to CONFIG_BTDM_MODEM_SLEEP.

Alvin1Zhang, 2020-04-20:
@dobairoland Thanks for the notes. @romocitto88 Thanks for reporting, would you please help share if your issue has been resolved? Thanks.

Alvin1Zhang, 2020-04-26:
@romocitto88 Thanks for reporting, would you please help share if your issue has been resolved? Thanks.

romocitto88, 2020-04-30:
Thanks @dobairoland 
At the end I use directly CONFIG_BTDM_MODEM_SLEEP_MODE_ORIG or CONFIG_BTDM_MODEM_SLEEP_MODE_EVED to do my test.

Alvin1Zhang, 2020-05-06:
@romocitto88 Glad that you found the solution, will close the ticket. Thanks.

closed 2020-05-06 by Alvin1Zhang