espressif/esp-idf #5142: Linker Script Generation api guide has typo in Makefile example

clbr, 2020-04-17:
## Problem Description

linker-script-generation.rst has the following in the Make section of "Creating and Specifying a Linker Fragment File".

``COMPONENT_ADD_LDFRAGMENTS += "my_linker_fragment_file.lf"``

Note the quotes around the filename. This is a bug, if the user copy-pastes that as is, make will error out, ``No rule to make target .../"my_linker_fragment_file.lf"`` - it expects the quotes to be part of the file name.

To fix this, remove the quotes from that line.
``COMPONENT_ADD_LDFRAGMENTS += my_linker_fragment_file.lf``

### Expected Behavior

Doc includes a copy-pastable line that works.

### Actual Behavior

Doc includes buggy example.

### Steps to reproduce

1. Copy-paste the doc line into a new esp-idf project
2. make

Alvin1Zhang, 2020-04-20:
@clbr Thanks for reporting, we will look into.

renzbagaporo, 2020-04-28:
Thanks for bringing this to our attention. A fix has been submitted internally.

closed 2020-05-05 by espressif-bot