espressif/esp-idf #5144: Softap not working with IDF from Branch/Release 4.0

MHAN0, 2020-04-17:
Hello,

i have last time updated the IDF to Version Branch/Release 4.0. Since that, softap is not working or needs too long time (several minutes)

Before i update, i have used IDF Version 3.1.2. To understand what's different, i checked first the sdk config and see the existing configurated parameters are equal. Hardware and User-Software are the same.

With 3.1.2: After Reset the Board, softap is immediatly available and my tablet is connected really fast.
So by a reset, it needs 2-3 seconds for reconnect. That's perfekt.

With Version 4.0: After Reset the Boardd softap is not available, he needs several minutes. 
To check it is my code, I have taken the softap example from idf 4.0, same problem.

What is here different, that softap work with a older Version IDF 3.1.2 perfectly but with the Release IDF 4.0 not. Same Hardware same Software same SDK Config. What is here different?

Can you help me. Thanks.

Here my the SDK config and Output from Terminal for both versions.
 

#######################SDK CONFIG  V3.1.2###############################
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=
CONFIG_BOOTLOADER_APP_TEST=

#
# Security features
#
CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT=
CONFIG_SECURE_BOOT_ENABLED=
CONFIG_FLASH_ENCRYPTION_ENABLED=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="COM3"
CONFIG_ESPTOOLPY_BAUD_115200B=
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=y
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=921600
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=y
CONFIG_FLASHMODE_DOUT=
CONFIG_ESPTOOLPY_FLASHMODE="dio"
[... 1323 lines cut ...]
I (500) wifi:wifi driver task: 3ffc369c, prio:23, stack:3584, core=0
I (500) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (504) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (524) wifi:wifi firmware version: 101cee8
I (525) wifi:config NVS flash: enabled
I (525) wifi:config nano formating: disabled
I (525) wifi:Init dynamic tx buffer num: 32
I (529) wifi:Init data frame dynamic rx buffer num: 32
I (534) wifi:Init management frame dynamic rx buffer num: 32
I (539) wifi:Init management short buffer num: 32
I (544) wifi:Init static rx buffer size: 1600
I (548) wifi:Init static rx buffer num: 20
I (552) wifi:Init dynamic rx buffer num: 32
I (1228) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 1, 0
I (1229) wifi:mode : softAP (30:ae:a4:56:f0:ad)
I (1231) wifi:Total power save buffer number: 16
I (1232) wifi:Init max length of beacon: 752/752
I (1237) wifi:Init max length of beacon: 752/752
I (1242) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0

################ After one minute ##################

I (83486) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (83488) wifi:station: a8:34:6a:5f:d9:e2 join, AID=1, bgn, 20
I (83601) tcpip_adapter: softAP assign IP to station,IP is: 192.168.10.2

MHAN0, 2020-04-17:
Update:

It seems it is a Android 9.0 Problem. With Android 7 and  IOS 13 works. What's the reason??

MHAN0, 2020-04-18:
Update next:

Android 9 has no problem with IDF 3.1.2. Only with IDF 4.0

I have enabled the wifi debug and see after reset the following info several times

E (1687) wifi:no bcast bss, send deauth: reason not authed(6)

Is that the reason??

Alvin1Zhang, 2020-04-20:
@MHAN0 Thanks for reporting and sharing the follow up details, we will look into, thanks.

espxiehang, 2020-05-25:
@MHAN0 I did not reproduce the problem.
Can you find an ESP32 as a station and provide us with the log of the process of connecting the Station to stofap? If possible, also provide us with packet capture.

liuzfesp, 2020-07-07:
Hi @MHAN0 could you provide following info:

- What's the max_connection of your wifi_ap_config_t? Or it's the default value?
- How many stations are connected to ESP32 soft-AP when the problem happens?

Alvin1Zhang, 2020-08-18:
@MHAN0 Thanks for reporting and sorry for the slow turnaround, would you please help share if any updates for this issue? Thanks.

liuzfesp, 2020-09-14:
Hi @MHAN0 any update about this issue?

closed 2020-11-02 by Alvin1Zhang