espressif/esp-idf #5146: PCNT EXAMPLE, Maybe an error?

pikolo84, 2020-04-17:
I had have to change PCNT_EVT_THRES_0 for PCNT_STATUS_THRES0 (and succesive) for make it work. It's an error in the example?, investigating:

PCNT.status_unit[PCNT_UNIT_X].val; returns a uint32_t as this:
    union {
        struct {
            uint32_t cnt_mode:2;                    /*0: positive value to zero; 1: negative value to zero; 2: counter value negative ; 3: counter value positive*/
            uint32_t thres1_lat:1;                  /* counter value equals to thresh1*/
            uint32_t thres0_lat:1;                  /* counter value equals to thresh0*/
            uint32_t l_lim_lat:1;                   /* counter value reaches h_lim*/
            uint32_t h_lim_lat:1;                   /* counter value reaches l_lim*/
            uint32_t zero_lat:1;                    /* counter value equals zero*/
            uint32_t reserved7:25;
        };
        uint32_t val;
    } status_unit[8];

#define PCNT_STATUS_THRES0    BIT(3) in pcnt_reg.h
PCNT_EVT_THRES_0 = 2 in pcnt.h

Can someone clarify for me?

https://github.com/espressif/esp-idf/blob/d85d3d969ff4b42e2616fd40973d637ff337fae6/examples/peripherals/pcnt/main/pcnt_example_main.c#L194

Thank you.

Alvin1Zhang, 2020-04-20:
@pikolo84 Thanks for reporting, we will look into.

koobest, 2020-05-12:
Hi, @pikolo84 

if  `PCNT_EVT_THRES_0 = 2`, it should be a bug, but I saw the definition in` pcnt_types.h` (https://github.com/espressif/esp-idf/blob/d85d3d969ff4b42e2616fd40973d637ff337fae6/components/soc/include/hal/pcnt_types.h#L84)    is:
```
typedef enum {
    PCNT_EVT_THRES_1 = BIT(2),           /*!< PCNT watch point event: threshold1 value event */
    PCNT_EVT_THRES_0 = BIT(3),           /*!< PCNT watch point event: threshold0 value event */
    PCNT_EVT_L_LIM = BIT(4),             /*!< PCNT watch point event: Minimum counter value */
    PCNT_EVT_H_LIM = BIT(5),             /*!< PCNT watch point event: Maximum counter value */
    PCNT_EVT_ZERO = BIT(6),              /*!< PCNT watch point event: counter value zero event */
    PCNT_EVT_MAX
} pcnt_evt_typ
```
can you update your IDF  to check it?

thanks !!

Alvin1Zhang, 2020-05-18:
@pikolo84 Thanks for reporting. Will close due to short of updates. Feel free to reopen with more details, thanks.

closed 2020-05-18 by Alvin1Zhang