espressif/esp-idf #5149: GATT_Server and GATT_Client at same time

fzboffice, 2020-04-18:
i use the GATT_Server and GATT_Client at same time, it works, but I have the problem that if the GATT_Client connects/disconnects to a Server the GATT_Server part also receive the ESP_GATTS_CONNECT_EVT and ESP_GATTS_DISCONNECT_EVT events. And also vice versa if the GATT_Server get a connection/disconnection the GATT_Client part get the ESP_GATTC_CONNECT_EVT and ESP_GATTC_DISCONNECT_EVT event. How can I avoid this ?

NailAgliev, 2020-09-12:
I have the same problem. Any solution so far?

xiewenxiang, 2020-09-16:
In the parameters of the connection event, there is a link_role param,  if the connection is initiated by ESP32 , the link_role is master, if the connection is initiated by remote device, link role will be slave. 

```c
    struct gattc_connect_evt_param {
        uint16_t conn_id;               /*!< Connection id */
        uint8_t link_role;              /*!< Link role : master role = 0  ; slave role = 1*/
        esp_bd_addr_t remote_bda;       /*!< Remote bluetooth device address */
        esp_gatt_conn_params_t conn_params; /*!< current connection parameters */
    } connect;                          /*!< Gatt client callback param of ESP_GATTC_CONNECT_EVT */

    struct gatts_connect_evt_param {
        uint16_t conn_id;               /*!< Connection id */
        uint8_t link_role;              /*!< Link role : master role = 0  ; slave role = 1*/
        esp_bd_addr_t remote_bda;       /*!< Remote bluetooth device address */
        esp_gatt_conn_params_t conn_params; /*!< current Connection parameters */
    } connect;                          /*!< Gatt server callback param of ESP_GATTS_CONNECT_EVT */
```

fzboffice, 2020-09-17:
> In the parameters of the connection event, there is a link_role param, if the connection is initiated by ESP32 , the link_role is master, if the connection is initiated by remote device, link role [...]

感谢您的更新

closed 2020-09-17 by fzboffice