espressif/esp-idf #5151: Keep having to run . ./export.sh

sambilbow, 2020-04-18:
System: macOS Catalina 10.15.1
Shell: zsh

Issue: I get this:

`zsh: command not found: idf.py`

Unless I `cd` to `~/esp/esp-idf` and `. ./export.sh` every new terminal session.

My `~/.zprofile` contains `export IDF_PATH=~/esp/esp-idf`

igrr, 2020-04-18:
@sambilbow This looks like expected behavior: export.sh needs to be sourced into every terminal session where you intend to use IDF. One option to make this easier is to create an alias in your .zprofile file, e.g. `alias get_idf=". $HOME/esp/esp-idf/export.sh"`. Then you can type get_idf in the new terminal session instead of typing the full path.

sambilbow, 2020-04-19:
Brilliant, thanks Ivan

closed 2020-04-19 by sambilbow