espressif/esp-idf #5152: esp-idf ULP examples don't compile/assemble under esp-idf 4.0

stevewms, 2020-04-19:
## Environment

- Development Kit:     ESP32 DevKitC (irrelevant)
- Kit version v2 (irrelevant)
- Module or chip used:  (irrelevant)
- IDF version 4.0 - vanilla install per https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/ with assembler as installed by esp-idf-tools-setup-2.2.exe
- Build System:   CMake - idf.py 
- Compiler xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Assembler 2.28.51.20170517
- Operating System:     Windows 10 - latest update, but similar seen on previous Win10 versions
- (Windows only) Plain Command Prompt (cmd.exe)
- Using an IDE?: No - using cmd.exe
- Power Supply:         Irrelevant

## Problem Description
The two example ULP projects, .../examples/system/ulp and .../examples/system/ulp_adc do not compile, and fail with the same error, with build hanging at step "[819/830] Performing build step of 'ulp_main'".  There is a 'ulp_main' file in .../build/esp-idf/main/ulp_main file in each of the project's build directories, so it's unclear why the build system should hang, but it does.

To test the general install, I built both 'get-started' projects, 'blink' and 'hello_world' without error using the same 'idf.py -p COM3 build' as I used to try to build the ULP examples unsuccessfully.

### Expected Behavior
All the esp-idf examples should build without error.
I especially want to use the ULP examples for clues as to how to set up my own project's CMakeLists.text files.  I can't trust the examples if they don't build without error.

### Actual Behavior
Build failure at step [819/830]

### Steps to reproduce

1. install esp-idf on a Windows 10 system according the the Get Started instructions at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/

2. Start up an esp-idf command prompt 
(C:\WINDOWS\system32\cmd.exe /k ""C:\Users\Steve\.espressif\idf_cmd_init.bat" "C:\Users\Steve\AppData\Local\Programs\Python\Python37\" "C:\Program Files\Git\cmd\""

(The above is the invocation of cmd.exe for an esp-idf command prompt)
The cmd location is c:\users\<user name>\desktop\esp-idf

3. cd examples\system\ulp

4. idf.py makemenuconfig (accept defaults)

5. idf.py -p <your port, mine is COM3> build

6. wait for hang at build step [819/830]

### Code to reproduce this issue
irrelevant

igrr, 2020-04-19:
Hi @stevewms, this seems to be the same underlying issue as https://github.com/espressif/esp-idf/issues/4914. The exact incorrect behavior depends on file associations. On release/v4.0 branch this issue was fixed in commit a8fe083, the fix will be part of the next bugfix release (v4.0.1).

stevewms, 2020-04-19:
igrr,
Thanks for the update.  I have two questions?
1.  Is there a work-around that I can use now?
2. When is bugfix release 4.0.1 expected to drop and how will I know when it hits?  How would I go about updating my current 4.0 install?
Thanks again.
stevewms

igrr, 2020-04-19:
The simplest solution is to update your copy of IDF to the latest release/v4.0 branch:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#updating-to-a-release-branch

The same page contains instructions for updating to a tagged release.

Alternatively, if you want to stay as close to v4.0 release as possible, you can cherry-pick just the bugfix commit a8fe083 on top of v4.0 (`git checkout v4.0; git cherry-pick a8fe083`)

stevewms, 2020-04-20:
The link you pointed me to, https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#updating-to-a-release-branch
does not have instructions for upgrading to v4.0.1, which is what I would prefer to do.  The instructions are oriented to linux systems, and I'm running cmd.exe on a Windows 10 system.  When I tried to adapt the instructions at the link you gave me, I got the following result:

C:\Users\Steve\Desktop\esp-idf>cd %IDF_PATH%

C:\Users\Steve\Desktop\esp-idf>git fetch

C:\Users\Steve\Desktop\esp-idf>git checkout release/v4.0
error: Your local changes to the following files would be overwritten by checkout:
        components/partition_table/gen_esp32part.py
        components/partition_table/parttool.py
        tools/ci/build_examples_cmake.sh
        tools/ci/test_build_system.sh
        tools/ci/test_build_system_cmake.sh
        tools/cmake/convert_to_cmake.py
        tools/idf.py
        tools/idf_monitor.py
        tools/idf_tools.py
        tools/kconfig_new/confgen.py
Please commit your changes or stash them before you switch branches.
Aborting

C:\Users\Steve\Desktop\esp-idf>git pull
You are not currently on a branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.

Would you please provide instructions that work with Win 10 cmd.exe.  You might also consider updating the documentation.
Thanks.
stevewms

igrr, 2020-04-20:
Hi @stevewms, the commands you are running look correct, and git has failed to check out the release/v4.0 branch because of local changes in some of the files. You can run `git diff` to check what the changes are. If you don't wish to keep them, you may run `git stash` or `gut reset --hard v4.0` to stash/discard the changes. Then continue with git checkout, git pull, etc.

stevewms, 2020-04-20:
If one (me, for instance) just re-installs v4.0 from scratch, will that re-install pick up the bug fixes in 4.0.1?  This is not clear from the documentation.

igrr, 2020-04-20:
No, installing v4.0 release version will install exactly that version. When v4.0.1 is released, it will appear as a new option in the installer, probably replacing v4.0.

stevewms, 2020-04-21:
Any idea when 4.0.1 will be released?

igrr, 2020-04-21:
At the moment the release testing is in progress, and the release will happen in 2 weeks if no blocking issues are found. If you use the current release/v4.0 branch, it should be very close to what will be eventually released as v4.0.1.

stevewms, 2020-04-21:
Thanks for your responsiveness and transparency on this issue, as well as your help as I try to come up to speed on all this.  I really appreciate it.

stevewms, 2020-04-24:
igrr,  I am having problems following your instructions on getting esp-idf 4.0.1 onto my Windows 10 system.  The directions you gave along with the instructions at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#updating-to-a-release-branch
just don't work on my system using cmd.exe.
I even tried loading up WSL and Ubuntu, and the directions to load esp-idf in that environment don't work either.

Would it be possible for you to provide a set of "cook book" instructions for getting v4.0.1 onto a Win10 system using cmd.exe, or point me to where such a set of steps exist?
Thanks.

igrr, 2020-04-25:
> just don't work on my system using cmd.exe.

Could you please post the output you are seeing?

The steps you did in this comment: https://github.com/espressif/esp-idf/issues/5152#issuecomment-616322356 look correct, except that `git checkout` command fails, as some files are modified. I can't say why they are modified (whether you have modified them yourself, or not), but if you didn't intend to modify them, you can run:

```
cd %IDF_PATH%
git fetch origin release/v4.0
git reset --hard release/v4.0
```

If you did intend to modify these files, you can stash these changes (`git stash`), then switch to `release/v4.0` branch as indicated above, then restore the changes (`git stash pop`).

Finally, if all the above fails: run the IDF Tools Installer again, and ask it to download `release/v4.0` branch for you. This will give you a new copy of IDF with that branch checked out.

stevewms, 2020-04-27:
I downloaded a fresh copy of release/v4.0, and the problem was not solved.  I installed using 
esp-idf-tools-setup-2.3.exe.
It gave me the option of downloading the latest, which turned out to be v4.2beta.  I chose 4.0, which, as I said, fails in building the example ULP program in the way seen previously.
Would 4.2beta have the bug fix?  I believe I read that 4.2 supports ESP32-S2.  Does it also support ESP32?  If so, does one need to do anything special to tell it to compile for ESP32 and assemble for the ESP32 ULP?

stevewms, 2020-05-07:
As I had plenty of other work to do that I could better spend the time on, rather than try to trouble shoot my inability to bring the v4.0.1 bug fix into my 4.0 installation, given my lack of experience with git and not having cookbook instructions, I decided to wait the two weeks you estimated it would take to integrate the 4.0.1 stuff into the 4.0 stable release.  Today, a bit more than 2 weeks after your estimate, I re-installed stable esp-idf v4.0 and again the ULP example in the idf failed to build with the same error under Windows cmd.exe, failing at step [825/830],  with the error message being:

> ../main/ulp_example_main.c:20:10: fatal error: ulp_main.h: No such file or directory
>  #include "ulp_main.h"
>           ^~~~~~~~~~~~
> compilation terminated.
> [825/830] Building ASM object esp-idf/main/CMakeFiles/__idf_main.dir/__/__/ulp_main.bin.S.obj
> ninja: build stopped: subcommand failed.
> ninja failed with exit code 1

Do you have another estimate as to when this bug fix will be rolled into v4.0?  Would it be in v4.1-beta.  Is there another version of the idf I can try?

Thanks.

igrr, 2020-05-07:
Sorry, could you clarify, which version are you selecting in the IDF Tools Installer?
There should be the following options in the installer:
```
v4.0
v3.3.2
release/v4.0
release/v3.3
master
```

Please try again, selecting the `release/v4.0` option.

>  integrate the 4.0.1 stuff into the 4.0 stable release

Sorry for not making this clear. Release v4.0 has been released, and it is basically set in stone. Any bug fixes to it will be part of v4.0.1, v4.0.2, etc releases. At the moment `release/v4.0` branch is very close to what `v4.0.1` release will be, hence my recommendation to install `release/v4.0`.

stevewms, 2020-05-08:
Thanks for the quick response.  I installed, as you suggested, "release/v4.0" and that fixed the problem.  I appreciate your help on this.  I'm going to mark this issue closed.

wevets.

closed 2020-05-08 by stevewms