espressif/esp-idf #5153: CPU exception, Guru Meditation Error: Core  0 panic'ed

stdenits, 2020-04-19:
## Environment

- Development Kit: [none]
- Kit version: [none]
- Module or chip used:  [ESP32-PICO-D4]
- IDF version: v4
- Build System: [???]
- Compiler version: [???]
- Operating System:     [Windows]
- (Windows only) environment type: [???].
- Using an IDE?: [Yes (VSCode + ESP-IDF Extension)]
- Power Supply:  [external 3.3V]

## Problem Description

ESP32 panic handler send message: "Guru Meditation Error: Core 0 panic'ed (IllegalInstruction). Exception was unhandled.".

I'm having some trouble implementing a SPI communication between the ESP32 (SLAVE) and the EFM32GG11 MCU (MASTER).
Basically, what I'm trying to do is reciveing / sending a byte from the ESP32 to the EFM32 with ESD-IDS SPI Reciver example. Just to test SPI ability.

ESP32 module is: LilyGO TTGO-Micro32-V2.0, https://github.com/LilyGO/TTGO-Micro32-V2.0

### Steps to reproduce

On the ESP32 I'm running the SPI_Slave (Receiver) example with a few changes:

`
#include <stdio.h>
#include <stdint.h>
#include <stddef.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"

#include "lwip/sockets.h"
#include "lwip/dns.h"
#include "lwip/netdb.h"
#include "lwip/igmp.h"

#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "soc/rtc_periph.h"
#include "esp32/rom/cache.h"
#include "driver/spi_slave.h"
#include "esp_log.h"
#include "esp_spi_flash.h"
#include "driver/gpio.h"

/*
It uses the standard SPI pins (MISO, MOSI, SCLK, CS) to
transmit data over in a full-duplex fashion, that is, while the master puts data on the MOSI pin, the slave puts its own
data on the MISO pin.

This example uses one extra pin: GPIO_HANDSHAKE is used as a handshake pin. After a transmission has been set up and we're
ready to send/receive data, this code uses a callback to set the handshake pin high. The sender will detect this and start
sending a transaction. As soon as the transaction is done, the line gets set low again.
*/

/*
Pins in use. The SPI Master can use the GPIO mux, so feel free to change these if needed.
*/
#define GPIO_HANDSHAKE 8
#define GPIO_MOSI 23
#define GPIO_MISO 19
[... 205 lines cut ...]

Core 0 register dump:
PC      : 0x400d4589  PS      : 0x00060430  A0      : 0x800d3bc7  A1      : 0x3ffb4d50
0x400d4589: gpio_config at C:/Users/Nox/esp-idf/components/driver/gpio.c:297

A2      : 0x3ffb4de8  A3      : 0x00000008  A4      : 0x3ff49068  A5      : 0x00000008
A6      : 0x00000000  A7      : 0x00000100  A8      : 0x00000002  A9      : 0xfffffdff
A10     : 0x00000000  A11     : 0x00060023  A12     : 0x00060020  A13     : 0x0000000c
A14     : 0x00060420  A15     : 0x3ffaffe8  SAR     : 0x00000008  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

ELF file SHA256: 88022220a20a2a22a8888a2228a8a200a802aa22a888208aaa0202aa082a80a0

Backtrace: 0x400d4586:0x3ffb4d50 0x400d3bc4:0x3ffb4db0 0x400d0cd6:0x3ffb4f40 0x400875ed:0x3ffb4f60
0x400d4586: gpio_config at C:/Users/Nox/esp-idf/components/driver/gpio.c:294

0x400d3bc4: app_main at C:\Users\Nox\esp\receiver\build/../main/app_main.c:90

0x400d0cd6: main_task at C:/Users/Nox/esp-idf/components/esp32/cpu_start.c:554

0x400875ed: vPortTaskWrapper at C:/Users/Nox/esp-idf/components/freertos/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57
`

igrr, 2020-04-19:
@stdenits Please check PICO-D4 datasheet:

> Pins IO16, IO17, CMD, CLK, SD0 and SD1 are used for connecting the embedded flash, and are not recommended for other uses. For details, please see Section 6 Schematics.

You are using SD1 (GPIO8) for GPIO_HANDSHAKE.

stdenits, 2020-04-20:
@igrr Thank you. Now it running without panic.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6860
load:0x40078000,len:14076
ho 0 tail 12 room 4
load:0x40080400,len:4304
entry 0x400806e8
I (73) boot: Chip Revision: 1
I (74) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot: ESP-IDF v4.0-dirty 2nd stage bootloader
I (42) boot: compile time 14:52:37
I (42) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 4MB
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x06c54 ( 27732) map
I (122) esp_image: segment 1: paddr=0x00016c7c vaddr=0x3ffb0000 size=0x02108 (  8456) load
I (126) esp_image: segment 2: paddr=0x00018d8c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/Users/Nox/esp-idf/components/freertos/xtensa_vectors.S:1778

I (131) esp_image: segment 3: paddr=0x00019194 vaddr=0x40080400 size=0x06e7c ( 28284) load
I (151) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x14ab4 ( 84660) map
0x400d0018: _stext at ??:?

I (182) esp_image: segment 5: paddr=0x00034ad4 vaddr=0x4008727c size=0x02ca0 ( 11424) load
0x4008727c: multi_heap_assert at C:/Users/Nox/esp-idf/components/heap/multi_heap_platform.h:52
 (inlined by) multi_heap_free_impl at C:/Users/Nox/esp-idf/components/heap/multi_heap.c:479

I (193) boot: Loaded app from partition at offset 0x10000
I (193) boot: Disabling RNG early entropy source...
I (194) cpu_start: Pro cpu up.
I (197) cpu_start: Application information:
I (202) cpu_start: Project name:     spi-slave-receiver
I (208) cpu_start: App version:      1
I (212) cpu_start: Compile time:     Apr 19 2020 14:52:02
I (218) cpu_start: ELF file SHA256:  5c8c0fb8251871ea...
I (224) cpu_start: ESP-IDF:          v4.0-dirty
I (230) cpu_start: Starting app cpu, entry point is 0x40081010
0x40081010: call_start_cpu1 at C:/Users/Nox/esp-idf/components/esp32/cpu_start.c:272

I (0) cpu_start: App cpu up.
I (240) heap_init: Initializing. RAM available for dynamic allocation:
I (247) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (253) heap_init: At 3FFB3120 len 0002CEE0 (179 KiB): DRAM
I (259) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (266) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (272) heap_init: At 40089F1C len 000160E4 (88 KiB): IRAM
I (278) cpu_start: Pro cpu start user code
I (296) spi_flash: detected chip: gd
I (297) spi_flash: flash io: dio
I (297) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (305) gpio: GPIO_PIN mask error
```

But now i get "GPIO_PIN mask error". What causes this?

igrr, 2020-04-20:
From driver/gpio.c:
```c++
    if (pGPIOConfig->pin_bit_mask == 0 || pGPIOConfig->pin_bit_mask >= (((uint64_t) 1) << GPIO_PIN_COUNT)) {
        ESP_LOGE(GPIO_TAG, "GPIO_PIN mask error ");
        return ESP_ERR_INVALID_ARG;
    }
```

It seems to happen if pin_bit_mask is 0 or if the bit mask includes a pin number higher than the max number of GPIOs. What is the new value of `GPIO_HANDSHAKE` which you are using?

stdenits, 2020-04-20:
>What is the new value of GPIO_HANDSHAKE which you are using?
It's GPIO 32.

igrr, 2020-04-20:
```
pin_bit_mask = (1 << GPIO_HANDSHAKE)
```

Ok, keep in mind that 1 << 32 == 0. You need to use `1ULL << GPIO_HANDSHAKE` instead.

```
WRITE_PERI_REG(GPIO_OUT_W1TS_REG, (1 << GPIO_HANDSHAKE));
```
This part will also not work with GPIOs 32 and up. You need to change this to GPIO_OUT1_W1TS, and shift by (gpio - 32).

stdenits, 2020-04-20:
Probably it works.

Could you describe sender-reciver algoritm with handshake? Not sure if i understood it correct.
If you have the willing and time, of course.

Need to fix my mistakes to get it somehow work.

Thank you.

closed 2020-04-21 by stdenits