espressif/esp-idf #5154: UART 1 pin assignment not working

TJ-C, 2020-04-19:
## Environment

- Development Kit:  None
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used: TTGO T-Beam ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
- IDF version v4.2-dev-1126-gd85d3d969 when not using PlatformIO. When compiling under PlatformIO using Core 4.3.1 Home 3.1.1
- Build System:  idf.py
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: Yes VS Code
- Power Supply:  USB

## Problem Description

//Detailed problem description goes here.
Using this example [nmea0183_parser](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/nmea0183_parser) setting TX pin to 12 and RX pin to 15 results in no data. I believe there is a bug in your code that does not setup the uart 1 correctly. I have written and tried numerous examples using the Arduino framework and uart 1 receives gps data as expected.### Expected Behavior

### Actual Behavior
no gps data
### Steps to reproduce
use example in link above. You **must not** use default pins for UART 1

// If possible, attach a picture of your setup/wiring here.
[TTGO T-Beam pinouts and schematics](https://tinymicros.com/wiki/TTGO_T-Beam)

## Debug Logs

N/A

## Other items if possible

suda-morris, 2020-04-20:
Hi @TJ-C I ran the example as you did (i.e. connect IO15 to GPS's TXD) but the example still works. 

I guess what you mean `results in no data` is just nothing print out to the command line except `nmea_parser: NMEA Parser init OK`, right? If so, I suggest you to run the example's menuconfig, under the `NMEA Statement Support`, only selects `NMEA_STATEMENT_GGA`, in this way, you don't have to parse all the GPS statements, in case your GPS doesn't support some of statement that enabled by default in this example.

TJ-C, 2020-04-20:
Thanks for the feedback, as you guessed I do get: ` I (309) nmea_parser: NMEA Parser init OK` and nothing more.

Using Arduino to fetch raw data to show I am receiving the correct NEMA strings:
`
// The serial connection to the GPS device
HardwareSerial hs(1); 

// Pins and baudrate for gps serial port
static const int RXPin = 15, TXPin = 12;
static const uint32_t GPSBaud = 9600;

void setup(){
  //initialize Serial Monitor and uart1
  Serial.begin(115200);
  hs.begin(GPSBaud, SERIAL_8N1, TXPin, RXPin);
}

void loop(){
  while (hs.available() > 0){
    // get the byte data from the GPS
    byte gpsData = hs.read();
    Serial.write(gpsData);
  }
}

$GNGGA,104838.000,5119.6263,N,00103.3953,W,1,07,1.4,96.4,M,0.0,M,,*5C
$GNGLL,5119.6263,N,00103.3953,W,104838.000,A,A*50
$GPGSA,A,3,19,29,24,12,02,25,,,,,,,2.2,1.4,1.7*32
$BDGSA,A,3,13,,,,,,,,,,,,2.2,1.4,1.7*22
$GPGSV,3,1,10,02,22,095,19,06,21,063,,12,70,062,09,14,37,301,*76
$GPGSV,3,2,10,19,11,037,16,24,37,131,25,25,73,265,16,29,26,189,19*76
$GPGSV,3,3,10,31,12,300,,32,46,269,*74
$BDGSV,1,1,04,08,24,043,15,11,,,20,12,49,086,,13,26,070,23*53
$GNRMC,104838.000,A,5119.6263,N,00103.3953,W,0.00,336.60,200420,,,A*63
$GNVTG,336.60,T,,M,0.00,N,0.00,K,A*23
$GNZDA,104838.000,20,04,2020,00,00*48
$GPTXT,01,01,01,ANTENNA OK*35
`

Would you mind showing me precisely what you changed in your code?

These are the bits of code I changed:

From nema.parser.h
`typedef struct {
    struct {
        uart_port_t uart_port;        /*!< UART port number */
        **uint32_t tx_pin;			  /*!< UART Tx Pin number */
        uint32_t rx_pin;              /*!< UART Rx Pin number */**
        uint32_t baud_rate;           /*!< UART baud rate */
        uart_word_length_t data_bits; /*!< UART data bits length */
        uart_parity_t parity;         /*!< UART parity */
        uart_stop_bits_t stop_bits;   /*!< UART stop bits length */
        uint32_t event_queue_size;    /*!< UART event queue size */
    } uart;                           /*!< UART specific configuration */
} nmea_parser_config_t;

#define NMEA_PARSER_CONFIG_DEFAULT()       \
    {                                      \
        .uart = {                          \
            .uart_port = UART_NUM_1,       \
	    **.tx_pin = 12,                   \
            .rx_pin = 15,                   \**
            .baud_rate = 9600,             \
            .data_bits = UART_DATA_8_BITS, \
            .parity = UART_PARITY_DISABLE, \
            .stop_bits = UART_STOP_BITS_1, \
            .event_queue_size = 16         \
        }                                  \
    }
`
From nema_parser_example.main.c 
`
static void gps_event_handler(void *event_handler_arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    **ESP_LOGI(TAG, "Entered event handler");**
	gps_t *gps = NULL;
    switch (event_id) {
`

From nema_parser.c
`
if (uart_set_pin(esp_gps->uart_port, **config->uart.tx_pin, config->uart.rx_pin,**
                     UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE) != ESP_OK) {
        ESP_LOGE(GPS_TAG, "config uart gpio failed");
        goto err_uart_config;
    }
`

Did I miss something or did something incorrectly?

suda-morris, 2020-04-20:
I changed no code, but doing some configuration in the menuconfig.

Please try to reduce the statements to parse in this example. FYI: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/nmea0183_parser#configure-the-project

And please note the limitation of this example: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/nmea0183_parser#limitation

From the output from your GPS, I suggest you to only select `GGA` and `RMC` statement in the menuconfig.

TJ-C, 2020-04-20:
I noticed in v4.2 the HAL layer has been added for uart, as the HAL layer underpins the Arduino code, I was wondering if there was any documentation for the HAL layer available?

TJ-C, 2020-04-22:
The code is not defective, just the operator. :(

Closing.

closed 2020-04-22 by TJ-C