espressif/esp-idf #5156: uart_wait_tx_done(UART_NUM_2, portMAX_DELAY) hangs forever

JosuGZ, 2020-04-20:
## Environment

- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.2.3-264-g0c7abb897
  - **Note**: I've seen the issue happening on an older version, In this version, I have only triggered it with the method described below, I currently waiting for it to happen spontaneously but it may take hours.
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- (Windows only) environment type: MSYS2 mingw32

## Problem Description

After a few hours working, a call to `uart_wait_tx_done(UART_NUM_2, portMAX_DELAY)` never returns.

### Expected Behavior

It should return once the tx has been completed.

### Actual Behavior

The function call never returns.

### Steps to reproduce

I'm issuing AT commands, there is a moment when it stops working. I can't reproduce it manually.

If I add a delay between returning due to an empty buffer and enabling the interrupt, I can make it happen fast, I think the TX ends in that precise instant and there is never an interrupt:

```c
/**************************************************************/
esp_err_t uart_wait_tx_done(uart_port_t uart_num, TickType_t ticks_to_wait)
{
    UART_CHECK((uart_num < UART_NUM_MAX), "uart_num error", ESP_FAIL);
    UART_CHECK((p_uart_obj[uart_num]), "uart driver error", ESP_FAIL);
    BaseType_t res;
    portTickType ticks_start = xTaskGetTickCount();
    //Take tx_mux
    res = xSemaphoreTake(p_uart_obj[uart_num]->tx_mux, (portTickType)ticks_to_wait);
    if(res == pdFALSE) {
        return ESP_ERR_TIMEOUT;
    }
    xSemaphoreTake(p_uart_obj[uart_num]->tx_done_sem, 0);
    typeof(UART0.status) status = UART[uart_num]->status;
    //Wait txfifo_cnt = 0, and the transmitter state machine is in idle state.
    if(status.txfifo_cnt == 0 && status.st_utx_out == 0) {
        xSemaphoreGive(p_uart_obj[uart_num]->tx_mux);
        return ESP_OK;
    }
    vTaskDelay(1000 / portTICK_PERIOD_MS); // <<<<---------------------------- I suspect the problem happens here, just before enabling the interrupt
    uart_enable_intr_mask(uart_num, UART_TX_DONE_INT_ENA_M);

    TickType_t ticks_end = xTaskGetTickCount();
    if (ticks_end - ticks_start > ticks_to_wait) {
        ticks_to_wait = 0;
    } else {
        ticks_to_wait = ticks_to_wait - (ticks_end - ticks_start);
    }
    //take 2nd tx_done_sem, wait given from ISR
    res = xSemaphoreTake(p_uart_obj[uart_num]->tx_done_sem, (portTickType)ticks_to_wait);
    if(res == pdFALSE) {
        uart_disable_intr_mask(uart_num, UART_TX_DONE_INT_ENA_M);
        xSemaphoreGive(p_uart_obj[uart_num]->tx_mux);
        return ESP_ERR_TIMEOUT;
    }
    xSemaphoreGive(p_uart_obj[uart_num]->tx_mux);
    return ESP_OK;
}
```

## SDK Config
[... 554 lines cut ...]
CONFIG_SPIFFS_USE_MTIME=y

#
# Debug Configuration
#
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# TCP/IP Adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_TCPIP_LWIP=y

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y

```

JosuGZ, 2020-04-21:
I've added the IDF version where I saw the problem for the last time (`v3.2.3-147-g287bee03b`). I haven't seen any commit that may fix this.

JosuGZ, 2020-04-22:
After 40 hours I can confirm It has happened on the latest IDF version.

JosuGZ, 2020-04-23:
I think something like this solves the issue, but I'm not sure about the implications:

```diff
diff --git a/components/driver/uart.c b/components/driver/uart.c
index f5e76644d..5c0a678c0 100644
--- a/components/driver/uart.c
+++ b/components/driver/uart.c
@@ -1032,12 +1032,13 @@ esp_err_t uart_wait_tx_done(uart_port_t uart_num, TickType_t ticks_to_wait)
     }
     xSemaphoreTake(p_uart_obj[uart_num]->tx_done_sem, 0);
     typeof(UART0.status) status = UART[uart_num]->status;
+    uart_enable_intr_mask(uart_num, UART_TX_DONE_INT_ENA_M);
     //Wait txfifo_cnt = 0, and the transmitter state machine is in idle state.
     if(status.txfifo_cnt == 0 && status.st_utx_out == 0) {
         xSemaphoreGive(p_uart_obj[uart_num]->tx_mux);
+        uart_disable_intr_mask(uart_num, UART_TX_DONE_INT_ENA_M);
         return ESP_OK;
     }
-    uart_enable_intr_mask(uart_num, UART_TX_DONE_INT_ENA_M);
 
     TickType_t ticks_end = xTaskGetTickCount();
     if (ticks_end - ticks_start > ticks_to_wait) {
```

JosuGZ, 2020-05-11:
I have written this code to work around the problem, and count the number of times the problem happens:

```c
  esp_err_t r;
  do {
    r = uart_wait_tx_done(SIM800_UART, 1000 / portTICK_PERIOD_MS);
    if (r == ESP_ERR_TIMEOUT) tx_done_errors++;
  } while (r != ESP_OK);
```

`tx_done_errors` gets incremented over time (2 times in three days) and I never write more than 1KB so 1 second at 115200 bauds it is more than enough time to not reach a timeout.

koobest, 2020-05-12:
Hi, @JosuGZ 
Thank you for spending so much time to help debug this issue. Can you provide sample code to reproduce this problem?

thanks !!

JosuGZ, 2020-05-12:
It is hard since it is very random.

I think it could be reproduced by initializing UART 2  at 115200 (it will probably happen with other configurations though) and constantly writing nonstop; while at the same time having other tasks perform some random work.

Ethan-CYQ, 2020-08-18:
Hi, @JosuGZ 
I think the problem is caused by ` vTaskDelay()` in the `uart_wait_tx_done` function. 

The `vTaskDelay()` is above of the `uart_enable_intr_mask()`, so the `UART_TX_DONE_INT` interrupt was disable and could not give the `tx_done_sem` while the program was runing delay function.

You can refer to the `uart_rx_intr_handler_default` ISR in the `uart.c`:
```
if(uart_intr_status & UART_TX_DONE_INT_ST_M) {
    uart_disable_intr_mask(uart_num, UART_TX_DONE_INT_ENA_M);
    uart_clear_intr_status(uart_num, UART_TX_DONE_INT_CLR_M);
    // If RS485 half duplex mode is enable then reset FIFO and 
    // reset RTS pin to start receiver driver
    if (UART_IS_MODE_SET(uart_num, UART_MODE_RS485_HALF_DUPLEX)) {
        UART_ENTER_CRITICAL_ISR(&uart_spinlock[uart_num]);
        uart_reset_rx_fifo(uart_num); // Allows to avoid hardware issue with the RXFIFO reset
        uart_reg->conf0.sw_rts = 1;
        UART_EXIT_CRITICAL_ISR(&uart_spinlock[uart_num]);
    }
    xSemaphoreGiveFromISR(p_uart_obj[uart_num]->tx_done_sem, &HPTaskAwoken);
}
```
You can see that `tx_done_sem` will be gave in ISR and taken in `uart_wait_tx_done()` while the `UART_TX_DONE_INT` interrupt is enable.

If you put the `vTaskDelay()` behind the `uart_enable_intr_mask()`, the `uart_wait_tx_done` will return correct value.

JosuGZ, 2020-08-18:
Yes, the vTaskDelay is there to make the gap where the problem occurs longer, maximizing the chance that it happens.

The problem still occurs without it.

I'm not sure whether that brief period is, in fact, the cause though, as I think that the patch I provided didn't work.

Ethan-CYQ, 2020-08-19:
Ok, I will try to reproduced the problem.

Ethan-CYQ, 2020-08-19:
Hi, @JosuGZ 

Sorry I have not reproduced your problem. I ran the uart_write constantly with UART2 at  115200 baud rate, but it still ran well.

JosuGZ, 2020-08-19:
I had the idea of having a loop writing to the UART and calling uart_wait_tx_done, like I guess you have done, but also with some tasks doing some work and randomly waiting, for example something like this, with a few tasks:

```
long long counter = 0;
while (true) {
  counter++;
  if (counter % (rand() % 1000)) {
    vTaskDelay((rand() % 1000) / portTickPeriodMS);
  }
}
```

The idea is to add some entropy to see if the problem arise. I might try it someday but I leave it here just in case you want to try.

It is probably hard to reproduce though.

Ethan-CYQ, 2020-08-19:
There was a task that gets random number and prints the `tx_done_error` when I ran the loop of UART write:
```
static void random_task(void *param)
{
    while (1) {
        printf("I get random number: %2d, tx_done_errors: %2d\n", GET_RANDOM_NUM, tx_done_errors);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}
```
But I will try your idea later.

Ethan-CYQ, 2020-08-20:
Hi, @JosuGZ 
I had run the uart_write with randomly delay task. But it still ran well. And the waveform of the UART GPIO was correct too.

lexxvir, 2020-12-25:
Hi there,

I encountered into this issue too.
I use UART2 on 921600. On 115200 this issue doesn't occur for me.

Unfortunately, it occurs only on my real application, I can't reproduce it on simple demo.

In my application I found that it occurs 100% when app writes to the flash.
The behavior looks that:
1. my app gets a command from UART2
2. my app send an ACK response to UART2 (by another task)
3. my app writes to the flash

Steps 2 & 3 execute simultaneously, each on separate task.
Task from step 2 hangs forever in `uart_wait_tx_done`.

JosuGZ, 2021-01-11:
I've found a way of reliably triggering this by bussywaiting (on v3.2.2-407-g0c7abb8978):

```
#include <stdio.h>
#include <string.h>
#include <driver/uart.h>

#define UART (uart_port_t)UART_NUM_2

static void init_uart() {
  uart_config_t uart_config = {0};
  uart_config.baud_rate = 115200;
  uart_config.data_bits = UART_DATA_8_BITS;
  uart_config.parity    = UART_PARITY_DISABLE;
  uart_config.stop_bits = UART_STOP_BITS_1;
  uart_config.flow_ctrl = UART_HW_FLOWCTRL_DISABLE;
  uart_config.use_ref_tick = true; // Needed for power management

  ESP_ERROR_CHECK(uart_param_config(UART, &uart_config));
  ESP_ERROR_CHECK(uart_driver_install(UART, 2*1024, 0, 0, NULL, 0));
  ESP_ERROR_CHECK(uart_set_pin(UART, 27, 26, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));
}

const char *TEST_MSG = "Hello World!";
volatile int64_t max_bussy = 0;
volatile int64_t bussy;
void test() {
  for (;;) {
    bussy = max_bussy;
    uart_write_bytes(UART, TEST_MSG, strlen(TEST_MSG));
    while (bussy--) {};
    uart_wait_tx_done(UART, portMAX_DELAY);
    printf("Max bussy %lld\n", max_bussy);
    max_bussy++;
  }
}

void app_main() {
  init_uart();
  test();
}
```

With my current configuration, it stops at about 4900 (With the CPU at 80Mhz it stops at about 2400).

[image]

I have updated the issue.

JosuGZ, 2021-01-11:
This should shed a lot of light.

The image shows the time bussywaiting before calling tx_done, along with the bits of the interrupt that are set (I'm using an extern variable for this). In purple, we can see exactly when it fails: about 1021us. Then the function starts working again, but with no interrupt (tx_done sees the FIFO is empty so it just returns).

As I suspected, the problem happens exactly at the precise moment when the transmission ends so there is no interruption, but for some reason, the FIFO is not empty yet.

The interrupt handler seems complex and I'm not familiar with it so I doubt I will fix it although I might keep looking.
[image]

JosuGZ, 2021-01-12:
> I think something like this solves the issue, but I'm not sure about the implications:
> 
> ```diff
> diff --git a/components/driver/uart.c b/components/driver/uart.c
> index f5e76644d..464509432 100644
> --- a/components/driver/uart.c
> +++ b/components/driver/uart.c
> @@ -1032,12 +1032,13 @@ esp_err_t uart_wait_tx_done(uart_port_t uart_num, TickType_t ticks_to_wait)
>      }
>      xSemaphoreTake(p_uart_obj[uart_num]->tx_done_sem, 0);
>      typeof(UART0.status) status = UART[uart_num]->status;
> +    uart_enable_intr_mask(uart_num, UART_TX_DONE_INT_ENA_M);
>      //Wait txfifo_cnt = 0, and the transmitter state machine is in idle state.
>      if(status.txfifo_cnt == 0 && status.st_utx_out == 0) {
> +        uart_disable_intr_mask(uart_num, UART_TX_DONE_INT_ENA_M);
>          xSemaphoreGive(p_uart_obj[uart_num]->tx_mux);
>          return ESP_OK;
>      }
> -    uart_enable_intr_mask(uart_num, UART_TX_DONE_INT_ENA_M);
>  
>      TickType_t ticks_end = xTaskGetTickCount();
>      if (ticks_end - ticks_start > ticks_to_wait) {
> ```

The problem with this original fix is this line before enabling the interrupt: `typeof(UART0.status) status = UART[uart_num]->status;`. It copies the struct, so enabling the interrupt should happen one line early.

ginkgm, 2021-07-27:
Hi all,
    We have a related commit may fix this issue: c801b3a18209ae707d3f8224b0734315cdc99db8.

    Could you please have a try and see whether it fix your issue? I have looked into the code, and now all the disable/clearing logic are in the ISR, so that we can easily ensure every time we clear the interrupt, a semaphore will be sent. 

The task only enable the interrupt when needed, and waiting for the semaphore. This way we reduce the complexity, and the risk of mis-clearing/disabling the interrupt.

@JosuGZ , @koobest , @Ethan-CYQ , @lexxvir

JosuGZ, 2021-07-27:
I can't test your commit right now but you have [this](https://github.com/espressif/esp-idf/issues/5156#issuecomment-757979976) `app_main()` test program to reproduce the issue.

The problem is not that the semaphore is not being sent, the problem happens due to returning before enabling the interrupt: I think that in line 1037 the UART is idle, but the interrupt can happen inside the if statement. That is why in my fix I enable the interrupt mask before checking if the uart is idle, and then I disable it before returning.

[image]

JosuGZ, 2021-07-27:
Sorry, my previous comment is wrong, what I want to point out is the following:

Lets have PC as program context, and IC as interrupt context.

PC (line 1037): uart_hal_is_tx_idle? -> No -> we dont return
IC: Interrupt happens here
PC: uart_hal_ena_intr_mask -> There won't be an interruption.

ginkgm, 2021-07-28:
@JosuGZ 

- If the interrupt status is cleared, the semaphore must be sent 
- If the interrupt status hasn't been cleared, once the interrupt is enabled, the interrupt should trigger. It's not necessary for the interrupt to be enabled before the status is active.

btw, the is_tx_idle is kind of optimization for calling wait_tx_done several times without sending more data.

JosuGZ, 2021-07-28:
> If the interrupt status hasn't been cleared

There is no interrupt if it is not enabled, which is true until line 1042 in your code. So if uart_hal_is_tx_idle returns false, there is a gap where the transmission can end and the interrupt won't be fired.

At least that is what caused the problem I fixed. That is why I enable the interrupt before checking `uart_hal_is_tx_idle`, and then I disable it again before returning if it is indeed idle,

As you can see here, when I bussywait less than 1020us before calling tx_done, the interrupt fires and returns. If I wait for more than 1021us, the uart is already idle so it returns. But there is a gap (the purple one) where the uart is not idle but the interrupt never fires so the function returns timeout (purple == timeout) (or never returns if timeout is portMAX_DELAY).

[image]

ginkgm, 2021-07-28:
Hi @JosuGZ 

Is this testing result based on the latest code, or the commit before https://github.com/espressif/esp-idf/commit/c801b3a18209ae707d3f8224b0734315cdc99db8?

acf-bwl, 2021-07-28:
Hello,

We've just tested our software with the latest master, as well as master from about a week ago from before the patch was committed. The issue appears to be fixed following the introduction of the patch. Thanks very much for this work!

-Alan

JosuGZ, 2021-07-29:
Ok, so I have tested c801b3a18209ae707d3f8224b0734315cdc99db8 and I can't reproduce it, so it seems that it has been fixed. The previous commit still triggers the issue.

I'm not sure whether this removes the problem or just makes the gap where the problem happens small enough, as now there is a critical zone immediately.

ginkgm, 2021-12-13:
Thanks for your resopnd!

The issue was caused by the disabling of the interrupt under parallel execution. I think it fixes the root of the issue. If you are still able to reproduce the issue after the patch, please open it again.

closed 2021-12-13 by ginkgm