espressif/esp-idf #5158: esp_spp_write() too slow, any faster alternative?

francis2tm, 2020-04-20:
**Is your feature request related to a problem? Please describe.**

I'm using esp_spp_write() to transmit data over classic bluetooth. My specifications require to send 8bytes every 1ms, but the esp_spp_write() takes about 2.5ms to send 8bytes. I noticed that even if I send 100bytes instead of 8 bytes at once, esp_spp_write() also takes about 2.5ms, which made me surprised becaused I thought the duration of transmisstion would be linear with the amount of bytes to be sent.

**Describe the solution you'd like**

I would like a lightweight bt_spp_write function to send data rapidly.

**Describe alternatives you've considered**
Is there any solution to my problem right now?
It looks like the esp_spp_write() waits for something before sending the data
**Additional context**

Add any other context or screenshots about the feature request here.

Alvin1Zhang, 2020-04-21:
@francis2tm Thanks for reporting, we will evaluate.

Wth-Esp, 2020-05-19:
Hi @francis2tm 

Could you please describe how the 2.5ms is measured?

Thanks
Weitianhua

Alvin1Zhang, 2020-08-18:
@francis2tm Thanks for reporting, would you please help share if any updates? Thanks.

Wth-Esp, 2020-10-29:
Hi @francis2tm 

Please update your IDF and have a try. We have pushed a related commit to improve this delay.

Thanks

francis2tm, 2020-10-29:
> Hi @francis2tm
> 
> Please update your IDF and have a try. We have pushed a related commit to improve this delay.
> 
> Thanks

Thanks, I will

closed 2022-01-17 by francis2tm