espressif/esp-idf #5159: esp_timer_get_time() is inaccurate if called in short sequence

maikeriva, 2020-04-20:
## Environment

- Development Kit: custom board
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.1-beta1-112-g4fe18a925
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?:  VSCode

## Problem Description
Calling esp_timer_get_time() in short sequence (20ms or less) may occasionally return the same value twice, or generally values too close to each other.

### Expected Behavior
The returned time should be accurate

### Actual Behavior
See problem description

### Steps to reproduce
Run the following code in a test application. On my setup it fails about 30% of times.
```cpp
TEST_CASE("esp_timer_get_time() burst", "timer")
{
        int64_t t[100] = {};
        for (uint8_t i=0;i<100;i++) {
                t[i] = esp_timer_get_time();
                vTaskDelay(pdMS_TO_TICKS(10));
        }
        for (uint8_t i=1;i<100;i++) {
                ESP_LOGI(TAG,"%lld,%lld",t[i-1],t[i]);
                TEST_ASSERT_EQUAL(0,t[i-1]/1000==t[i]/1000);
        }
}
```
Please let me know if you need more details

KonstantinKondrashov, 2021-03-22:
Hi @maikeriva!
I see this is the old report. I tried to run this code (using v4.1-beta1-112-g4fe18a925 ) but it does not fail. It works well. I suppose there is some special to reproduce it. Could you share the sdkconfig file?
Thanks.

KonstantinKondrashov, 2021-03-23:
Hi @maikeriva! This case is not related to `esp_timer`. I did some experiments with the Tick rate (Hz) and I see that this behavior happens when CONFIG_FREERTOS_HZ = 100. 

```c
vTaskDelay(pdMS_TO_TICKS(10));
...
#define pdMS_TO_TICKS( xTimeInMs ) ( ( ( TickType_t ) ( xTimeInMs ) * configTICK_RATE_HZ ) / ( TickType_t ) 1000 )
```

`pdMS_TO_TICKS(10)` gives 1. `1 tick = 10ms. `
100Hz Tick rate does not provide a good resolution for your goal. For vTaskDelay it is possible to get not accurate delay (even less than 1ms).

closed 2021-03-23 by Alvin1Zhang