espressif/esp-idf #5160: 不初始化网络，直接进行socket会导致系统panic

newstarqu, 2020-04-21:
我在应用esp32的时候，系统出现panic
具体的应用流程如下：
1、创建task，获取ntp服务器的时间
2、初始化WiFi连接

这样开机的时候出现panic，位置在获取ntp服务器时间中调用socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP)的地方，请问这个是我的流程有问题还是系统有bug？

以下是panic日志：
/mnt/f/work/esp/esp32/esp-idf/components/freertos/queue.c:1447 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40095255 on core 1
0x40095255: xQueueGenericReceive at /mnt/f/work/esp/esp32/esp-idf/components/freertos/queue.c:1447 (discriminator 1)

ELF file SHA256: d8242db0210ddd29cc75b0ebcd0e947e43053708d5f0675465f1e8da180b3f9b

Backtrace: 0x40091ac9:0x3ffc74c0 0x40091e51:0x3ffc74e0 0x40095255:0x3ffc7500 0x40144fd9:0x3ffc7540 0x40145302:0x3ffc7560 0x4013340c:0x3ffc7580 0x40133478:0x3ffc75a0
 0x4014693d:0x3ffc75c0 0x4014558f:0x3ffc75e0 0x401328f1:0x3ffc7630 0x400d4b77:0x3ffc7650 0x400d4f78:0x3ffc7760 0x400954ed:0x3ffc77b0
0x40091ac9: invoke_abort at /mnt/f/work/esp/esp32/esp-idf/components/esp32/panic.c:157

0x40091e51: abort at /mnt/f/work/esp/esp32/esp-idf/components/esp32/panic.c:174

0x40095255: xQueueGenericReceive at /mnt/f/work/esp/esp32/esp-idf/components/freertos/queue.c:1447 (discriminator 1)

0x40144fd9: sys_mutex_lock at /mnt/f/work/esp/esp32/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:81

0x40145302: sys_arch_protect at /mnt/f/work/esp/esp32/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:467

0x4013340c: do_memp_malloc_pool at /mnt/f/work/esp/esp32/esp-idf/components/lwip/lwip/src/core/memp.c:255

0x40133478: memp_malloc at /mnt/f/work/esp/esp32/esp-idf/components/lwip/lwip/src/core/memp.c:350 (discriminator 2)

0x4014693d: netconn_alloc at /mnt/f/work/esp/esp32/esp-idf/components/lwip/lwip/src/api/api_msg.c:734

0x4014558f: netconn_new_with_proto_and_callback at /mnt/f/work/esp/esp32/esp-idf/components/lwip/lwip/src/api/api_lib.c:155

0x401328f1: lwip_socket at /mnt/f/work/esp/esp32/esp-idf/components/lwip/lwip/src/api/sockets.c:1706 (discriminator 12)

0x400d4b77: socket at /mnt/f/work/esp/esp32/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:656
 (inlined by) get_time_from_ntp at /mnt/f/work/esp/my_project/01_eb/program/source/main/sync_time.c:81

0x400d4f78: task_sync_time at /mnt/f/work/esp/my_project/01_eb/program/source/main/sync_time.c:214

0x400954ed: vPortTaskWrapper at /mnt/f/work/esp/esp32/esp-idf/components/freertos/port.c:143

Alvin1Zhang, 2020-04-24:
@newstarqu Thanks for the report. We will look into. @liuzfesp

freakyxue, 2020-04-24:
hi @newstarqu 
从assert来看 应该是lwip_init 没有执行 导致 lwip没有初始化.导致assert.
因为lwip初始化的时候 会初始化内存池.mem_init();
在申请套接字时,会从内存池中 申请MEMP_NETCONN 类型的变量,如果内存池没有初始化,直接申请会出问题.

解决方法:在申请套接字时,要初始化网络.tcpip_adapter_init();

newstarqu, 2020-04-24:
好的，我之前把tcpip_adapter_init()放在wifi_smart_config()中了，wifi_smart_config()调用要晚一些，我把tcpip_adapter_init()提出来放到前面试一下。

initcomputer@126.com
 
发件人： freakyxue
发送时间： 2020-04-24 14:44
收件人： espressif/esp-idf
抄送： quxiaolong; Mention
主题： Re: [espressif/esp-idf] 不初始化网络，直接进行socket会导致系统panic (IDFGH-3143) (#5160)
hi @newstarqu
从assert来看 应该是lwip_init 没有执行 导致 lwip没有初始化.导致assert.
因为lwip初始化的时候 会初始化内存池.mem_init();
在申请套接字时,会从内存池中 申请MEMP_NETCONN 类型的变量,如果内存池没有初始化,直接申请会出问题.
解决方法:在申请套接字时,要初始化网络.tcpip_adapter_init();

freakyxue, 2020-04-28:
hi @newstarqu 
调整了tcpip_adapter_init位置 ,问题解决了吗?

freakyxue, 2020-05-15:
hi @newstarqu 
这个问题解决了吗?如果没问题,我将关闭了

newstarqu, 2020-05-15:
不好意思，已经解决了，前段时间一直没有测试

initcomputer@126.com
 
发件人： freakyxue
发送时间： 2020-05-15 16:14
收件人： espressif/esp-idf
抄送： quxiaolong; Mention
主题： Re: [espressif/esp-idf] 不初始化网络，直接进行socket会导致系统panic (IDFGH-3143) (#5160)
hi @newstarqu
这个问题解决了吗?如果没问题,我将关闭了

Alvin1Zhang, 2020-05-15:
@newstarqu Thanks for the updates, glad that your issue is resolved. Feel free to reopen if it still happens. Thanks.

closed 2020-05-15 by Alvin1Zhang