espressif/esp-idf #5162: UART: Unable to TX more than 128 bytes - i need 256

pad52, 2020-04-21:
## Environment
- Module or chip used:  [ESP32-WROOM-32]
- IDF version: v3.3.1-211-g6f9a7264c
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:   macOS
- Using an IDE?: No
- Power Supply:     Battery

## Problem Description
Like in https://github.com/espressif/esp-idf/issues/4833

calling uart_tx_bytes or uart_tx_chars with more than 128 bytes results in sending only the first 128 bytes. 
I see that 128 bytes is Hard-coded, I would like to change it to 256.

koobest, 2020-04-21:
Hi, @pad52 
Can you provide the code to produce this issue?I think your application uses these two APIs by mistake.

thanks !!

pad52, 2020-04-21:
Hello,

This is my function:

`void uart_write(char* data, uint8_t len){
	int size;
	size = uart_tx_chars(uart_num, data, len);
	if(size  > 0){
		ESP_LOGI(TAG, "UART Wrote %d bytes",size);
	}else{
		ESP_LOGI(TAG, "UART Comm Failed");
	}`

and I call it like this
`#define OTA_DATASIZE 248
char outbuf[OTA_DATASIZE];
int len = OTA_DATASIZE;

 [...]
uart_write(&outbuf[0], len);`

and I always get "UART Wrote 128 bytes"

I suspect it's something connected to this: https://www.esp32.com/viewtopic.php?t=8858
and this i read in file: components/soc/soc/esp32/include/soc/uart_reg.h

`/* UART_TX_SIZE : R/W ;bitpos:[10:7] ;default: 4'h1 ; */
/*description: This register is used to configure the amount of mem allocated
 to transmitter's fifo.the default byte num is 128.*/
#define UART_TX_SIZE  0x0000000F
#define UART_TX_SIZE_M  ((UART_TX_SIZE_V)<<(UART_TX_SIZE_S))
#define UART_TX_SIZE_V  0xF
#define UART_TX_SIZE_S  7`

negativekelvin, 2020-04-21:
It's true if you use a function that writes directly to the uart fifo then you can't write more than the size of the fifo.  That's why the uart driver and its functions like uart_write_bytes exist, to overcome those hardware limitations.

pad52, 2020-04-21:
You are right, calling uart_tx_bytes instead uart_tx_chars fixed the problem.
Thank you.

closed 2020-04-21 by pad52

syed-haider5, 2020-04-30:
@pad52 
Hi, can you show me any part of code in which you're writing (via uart_write_byte OR via uart_tx_bytes) to send data of more than 255 bytes over UART. 
Basically I also wanted to send numerical data over UART having value more than 300. (Like 300 Watts, 400 Volts etc)

pad52, 2020-05-01:
Hello,

Look if you need to send an integer of "300" you need less than 16 byte..
In any case use uart_write_bytes(); and follow the docs here https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html