espressif/esp-idf #5164: error ompile  spi_bus_lock.c

wuppergit, 2020-04-21:
Compilation of spi_bus_lock.c throws  error:

error: implicit declaration of function 'xSemaphoreCreateBinaryStatic'; did you mean 'xSemaphoreCreateBinary'? [-Werror=implicit-function-declaration]

     g_spi_lock_main_flash_dev->semphr = xSemaphoreCreateBinaryStatic(&main_flash_semphr);

/esp/esp-idf/components/driver/spi_bus_lock.c:822:39: warning: assignment to 'SemaphoreHandle_t' {aka 'void *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
     g_spi_lock_main_flash_dev->semphr = xSemaphoreCreateBinaryStatic(&main_flash_semphr);

Changing to xSemaphoreCreateBinary doesn't work

AxelLin, 2020-04-21:
This was reported 3 weeks ago: https://github.com/espressif/esp-idf/issues/5046

Alvin1Zhang, 2020-04-22:
@AxelLin Thanks for the notes. @wuppergit Thanks for reporting, since the issue is same with https://github.com/espressif/esp-idf/issues/5046, will close this ticket, please help follow up on https://github.com/espressif/esp-idf/issues/5046. Thanks.

closed 2020-04-22 by Alvin1Zhang

wuppergit, 2020-04-22:
Thank' for the respond. It''s working fine.